kyokenn/rogdrv #1: Please could you assist

CunningSquirrel, 2019-07-09:
Trying to install this as I would love to use the mouse in Ubuntu. But it seems that a module is missing from your github. Whenever I try install using "sudo python3 setup.py install" I get:

Traceback (most recent call last):
  File "setup.py", line 3, in <module>
    from setuptools import setup
ModuleNotFoundError: No module named 'setuptools'

Am I doing something wrong of is the "setuptools" missing? Your help would be Greatly appreciated.

kyokenn, 2019-07-09:
Looks like you missing **python3-setuptools** package. Also make sure that you have **python3-pip** installed too.

CunningSquirrel, 2019-07-11:
Thanks so much, I will install those and check again. My next update might be a few weeks. My laptop stopped charging so I took it in yesterday under warranty. Need to wait for it to return before I can test again. On the plus side the mouse arrived today. I was trying to get the software installed on my system while I waited for the mouse to arrive. 

I'm stuck using it on the company's Windows machine now but looking forward to getting this working on Ubuntu. Thanks so much for taking the time to reply to me and for developing this in the first place. You are a legend, and your work is much appreciated :) 

I will let you know how it goes once I get the laptop back.

CunningSquirrel, 2019-07-31:
Good day, I have installed all the prerequisets as mentioned but it still fails with that setup.py error. Is there not supposed to be a setup.py in your folder that is missing ? If there is any info that you want from me please let me know and I will be happy to screenshot or run anything and post you the results if you have the time to assist me. I would really appreciate the help.

Regards
CS

kyokenn, 2019-07-31:
Make sure that you have installed the required ubuntu packages with:
`sudo apt install python3-setuptools python3-pip `

Then you can install the driver with:
`sudo python3 setup.py install`

Or with some other alternative method:
`sudo pip3 install . `

CunningSquirrel, 2019-07-31:
Install just ran perfectly, thank you so much! Now I just need to figure out how to use it.

running install
running bdist_egg
running egg_info
creating rogdrv.egg-info
writing rogdrv.egg-info/PKG-INFO
writing dependency_links to rogdrv.egg-info/dependency_links.txt
writing entry points to rogdrv.egg-info/entry_points.txt
writing top-level names to rogdrv.egg-info/top_level.txt
writing manifest file 'rogdrv.egg-info/SOURCES.txt'
reading manifest file 'rogdrv.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'rogdrv.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/rogdrv
copying rogdrv/__main__.py -> build/lib/rogdrv
copying rogdrv/defs.py -> build/lib/rogdrv
copying rogdrv/device.py -> build/lib/rogdrv
copying rogdrv/__init__.py -> build/lib/rogdrv
copying rogdrv/gtk3.py -> build/lib/rogdrv
copying rogdrv/colors.py -> build/lib/rogdrv
copying rogdrv/bindings.py -> build/lib/rogdrv
copying rogdrv/gtk3.glade -> build/lib/rogdrv
copying rogdrv/rog-symbolic.symbolic.png -> build/lib/rogdrv
copying rogdrv/rog.png -> build/lib/rogdrv
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/__main__.py -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/defs.py -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/rog-symbolic.symbolic.png -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/device.py -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/__init__.py -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/gtk3.py -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/colors.py -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/gtk3.glade -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/rog.png -> build/bdist.linux-x86_64/egg/rogdrv
copying build/lib/rogdrv/bindings.py -> build/bdist.linux-x86_64/egg/rogdrv
byte-compiling build/bdist.linux-x86_64/egg/rogdrv/__main__.py to __main__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/rogdrv/defs.py to defs.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/rogdrv/device.py to device.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/rogdrv/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/rogdrv/gtk3.py to gtk3.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/rogdrv/colors.py to colors.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/rogdrv/bindings.py to bindings.cpython-36.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying rogdrv.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying rogdrv.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying rogdrv.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying rogdrv.egg-info/entry_points.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying rogdrv.egg-info/not-zip-safe -> build/bdist.linux-x86_64/egg/EGG-INFO
copying rogdrv.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
creating dist
creating 'dist/rogdrv-0.0.4-py3.6.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing rogdrv-0.0.4-py3.6.egg
creating /usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg
Extracting rogdrv-0.0.4-py3.6.egg to /usr/local/lib/python3.6/dist-packages
Adding rogdrv 0.0.4 to easy-install.pth file
Installing rogdrv script to /usr/local/bin
Installing rogdrv-config script to /usr/local/bin

Installed /usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg
Processing dependencies for rogdrv==0.0.4
Finished processing dependencies for rogdrv==0.0.4

kyokenn, 2019-07-31:
Looks good. You can start the driver with:
`rogdrv`

If it works, then you can enable autostart. So it will start automatically after logging in into desktop environment.

CunningSquirrel, 2019-07-31:
Do I just open a terminal and type rogdrv ? I dont need to go to the install dir open terminal from there and then type rogdrv ?

Sorry to be such a pain

CunningSquirrel, 2019-07-31:
If I run it from terminal I get this: 

:~$ rogdrv
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 43, in loop
    e = device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'

Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 11, in <module>
    load_entry_point('rogdrv==0.0.4', 'console_scripts', 'rogdrv')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 52, in rogdrv
    gtk3_main(device)
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/gtk3.py", line 149, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'

kyokenn, 2019-07-31:
The driver can't find the mouse. Do you have the mouse connected? What mouse model you have? Do you installed the device ".rules" files? - https://github.com/kyokenn/rogdrv#installation

CunningSquirrel, 2019-07-31:
The device is installed I have model:L701(mouse). I did run the .rules but let me try running it again

CunningSquirrel, 2019-07-31:
cd to the install dir then ran the commands:

cn@CunningSquirrel:~/rogdrv-master$ sudo cp -f udev/50-rogdrv.rules /etc/udev/rules.d/
[sudo] password for cn:           
cn@CunningSquirrel:~/rogdrv-master$ sudo udevadm control --reload-rules
cn@CunningSquirrel:~/rogdrv-master$ sudo udevadm trigger
cn@CunningSquirrel:~/rogdrv-master$ 

Still seems to be something wrong:

cn@CunningSquirrel:~$ rogdrv
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 43, in loop
    e = device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'

Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 11, in <module>
    load_entry_point('rogdrv==0.0.4', 'console_scripts', 'rogdrv')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 52, in rogdrv
    gtk3_main(device)
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/gtk3.py", line 149, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'

cn@CunningSquirrel:~$ rogdrv-config profile 1
Device not found

Let me try restarting my Laptop

CunningSquirrel, 2019-07-31:
Restarted and still no luck. I'm feeling pretty useless right about now. I'm running Zorin OS 15 which is based on Ubuntu 18.04.2. That shouldn't make a difference though. Going to check in etc/udev/rules.d to see if I can see anything

CunningSquirrel, 2019-07-31:
It definitely copied the rules like you said:

cn@CunningSquirrel:/etc/udev/rules.d$ ls
50-rogdrv.rules   70-snap.core.rules
60-vboxdrv.rules  70-snap.leagueoflegends.rules

I see the file is there "50-rogdrv.rules"

kyokenn, 2019-07-31:
> The device is installed I have model:L701(mouse).

L701? So it's a ROG Spatha. It's not supported.

I can try to add it's ID to the driver. So some features may work, if it's compatible with Pugio.
I need an output of an "lsusb" command for that.

CunningSquirrel, 2019-07-31:
Thank you SO MUCH ! I would really appreciate it. I already ran that command and was about to post it for you :)

:~$ lsusb
Bus 002 Device 003: ID 0451:8140 Texas Instruments, Inc. 
Bus 002 Device 002: ID 0451:8140 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04ca:7070 Lite-On Technology Corp. 
Bus 001 Device 003: ID 0b05:1824 ASUSTek Computer, Inc. 
Bus 001 Device 008: ID 8087:0aaa Intel Corp. 
Bus 001 Device 009: ID 0451:ca01 Texas Instruments, Inc. 
Bus 001 Device 006: ID 0451:8142 Texas Instruments, Inc. TUSB8041 4-Port Hub
Bus 001 Device 013: ID 046d:c229 Logitech, Inc. G19 Gaming Keyboard Macro Interface
Bus 001 Device 007: ID 046d:c228 Logitech, Inc. G19 Gaming Keyboard
Bus 001 Device 004: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Bus 001 Device 002: ID 0451:8142 Texas Instruments, Inc. TUSB8041 4-Port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

CunningSquirrel, 2019-07-31:
Thank you SO MUCH !

kyokenn, 2019-07-31:
I have added Spatha USB ID to the driver. You can update it with `git pull`. Then you have to reinstall it.

The button rebinding may not work, because Spatha has more buttons than Pugio.
The LED configuration could work, because most ROG mices have the same amount of LEDs.
The battery status and wireless settings will not work, because Pugio is wired.

CunningSquirrel, 2019-07-31:
I will test and let you know, thank you :)

CunningSquirrel, 2019-07-31:
I have reinstalled and it no longer says "Device not found" but it still doesn't seem to want to work. Thank you for your assistance though with all this testing.

cn@CunningSquirrel:~$ rogdrv
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 11, in <module>
    load_entry_point('rogdrv==0.0.4', 'console_scripts', 'rogdrv')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 38, in rogdrv
    device = DeviceManager.get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 381, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 78, in __init__
    self._kbd = hidapi.Device(kbd_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 159, in __init__
    raise IOError("Could not open connection to device.")
OSError: Could not open connection to device.

cn@CunningSquirrel:~$ rogdrv-config profile 1
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.4', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 137, in rogdrv_config
    device = DeviceManager.get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 381, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 78, in __init__
    self._kbd = hidapi.Device(kbd_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 159, in __init__
    raise IOError("Could not open connection to device.")
OSError: Could not open connection to device.

kyokenn, 2019-07-31:
It can't open the device. This could be a permissions problem.
Did you copied the new updated ".rules" files?
After you copy them you have to tell the system reload those files with:
```
sudo udevadm control --reload-rules
sudo udevadm trigger
```
or you can just reboot a PC.

If nothing works, you can try to run rogdrv-config with sudo:
```sudo rogdrv-config profile 1```

CunningSquirrel, 2019-07-31:
Unfortunately that did not work. I ran the commands you suggested and even restarted the laptop, still nothing:

cn@CunningSquirrel:~$ sudo rogdrv
[sudo] password for kyle:           
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 11, in <module>
    load_entry_point('rogdrv==0.0.4', 'console_scripts', 'rogdrv')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 38, in rogdrv
    device = DeviceManager.get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 381, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 79, in __init__
    self._ctl = hidapi.Device(ctl_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 156, in __init__
    raise ValueError("Must provide either a DeviceInfo object, 'path' "
ValueError: Must provide either a DeviceInfo object, 'path' or 'vendor_id' and 'product_id'.
Exception ignored in: <bound method Device.__del__ of <hidapi.Device object at 0x7f2c4bc1d278>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/hidapi.py", line 164, in __del__
    if self._device is not None:
AttributeError: 'Device' object has no attribute '_device'

cn@CunningSquirrel:~$ sudo rogdrv-config profile 1
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.4', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/__main__.py", line 137, in rogdrv_config
    device = DeviceManager.get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 381, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.4-py3.6.egg/rogdrv/device.py", line 79, in __init__
    self._ctl = hidapi.Device(ctl_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 156, in __init__
    raise ValueError("Must provide either a DeviceInfo object, 'path' "
ValueError: Must provide either a DeviceInfo object, 'path' or 'vendor_id' and 'product_id'.
Exception ignored in: <bound method Device.__del__ of <hidapi.Device object at 0x7fe4af4a7c88>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/hidapi.py", line 164, in __del__
    if self._device is not None:
AttributeError: 'Device' object has no attribute '_device'

CunningSquirrel, 2019-07-31:
I will be formatting my laptop to format my NVMe drive with Samsung Magic this weekend and then I will do a fresh install and see if anything changes. Thank you for taking the time to try and assist.

yaliv, 2019-08-10:
@kyokenn I can get it working on Manjaro, but only with sudo.
Unlike Deepin OS (Debian based), I can run it without sudo.

Can I change this condition, like setting a python config somewhere to allow running it without sudo?

kyokenn, 2019-08-15:
> Can I change this condition, like setting a python config somewhere to allow running it without sudo?

You need to install the udev .rules files. Those files are used by the system to chmod your /dev/hidraw* files. You can check the current files permissions with: ls -alh /dev/hidraw*

yaliv, 2019-08-17:
@kyokenn yes, I've installed the custom udev rules.

The output of `ls -alh /dev/hidraw*`:

- Mouse unplugged:
```sh
crw------- 1 root root 239, 1 Agu 17 15:59 /dev/hidraw1
```

- Mouse plugged:
```sh
crw-rw-rw- 1 root root 239, 0 Agu 17 17:34 /dev/hidraw0
crw------- 1 root root 239, 1 Agu 17 15:59 /dev/hidraw1
crw-rw-rw- 1 root root 239, 2 Agu 17 17:34 /dev/hidraw2
crw-rw-rw- 1 root root 239, 3 Agu 17 17:34 /dev/hidraw3
```

Running some `rogdrv` commands:
```sh
$ rogdrv-config color logo 50 0 200                                                    
Traceback (most recent call last):
  File "/usr/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.4', 'console_scripts', 'rogdrv-config')()
  File "/usr/lib/python3.7/site-packages/rogdrv-0.0.4-py3.7.egg/rogdrv/__main__.py", line 107, in rogdrv_config
    device = DeviceManager.get_device()
  File "/usr/lib/python3.7/site-packages/rogdrv-0.0.4-py3.7.egg/rogdrv/device.py", line 386, in get_device
    return device_class()
  File "/usr/lib/python3.7/site-packages/rogdrv-0.0.4-py3.7.egg/rogdrv/device.py", line 78, in __init__
    self._kbd = hidapi.Device(kbd_info)
  File "/usr/lib/python3.7/site-packages/hidapi.py", line 159, in __init__
    raise IOError("Could not open connection to device.")
OSError: Could not open connection to device.

$ sudo rogdrv-config color logo 50 0 200                                                    
1 (logo):	[brightness:4] (50, 0, 200)
2 (wheel):	[brightness:0] (0, 0, 0)
3 (bottom):	[brightness:0] (0, 0, 0)
```

kyokenn, 2019-08-21:
Looks like your mouse consists of those device files: hidraw0 hidraw2 hidraw3
The permissions looks fine for them.
I don't know where is the problem. It's something that is related to devices enumeration.
This could be the problem with HID subsystem configuration, but I'm not familiar with your Linux distribution, so I can't help you with it.

MykoIce, 2020-04-14:
Hey, old issue, hi.

I´ve been messing with my Spartha, and I´ve managed to get it loaded. (Apparently, both Controller and Keyboard interfaces on the wireless are 0? I dunno.) But I´m having trouble getting profiles to load, the error being this.

```
einar@CMD:~/Downloads/rogdrv-master$ sudo rogdrv-config profile 1 --debug
[2020-04-14 23:00:04,225] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2020-04-14 23:00:04,227] [DEBUG] 0 devices found
[2020-04-14 23:00:04,227] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2020-04-14 23:00:04,228] [DEBUG] 0 devices found
[2020-04-14 23:00:04,228] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2020-04-14 23:00:04,229] [DEBUG] 0 devices found
[2020-04-14 23:00:04,229] [DEBUG] searching for device StrixEvolve (VendorID: 0x0B05, ProductID: 0x185B)
[2020-04-14 23:00:04,230] [DEBUG] 0 devices found
[2020-04-14 23:00:04,230] [DEBUG] searching for device Spatha (VendorID: 0x0B05, ProductID: 0x181C)
[2020-04-14 23:00:04,231] [DEBUG] 0 devices found
[2020-04-14 23:00:04,231] [DEBUG] searching for device SpathaWireless (VendorID: 0x0B05, ProductID: 0x1824)
[2020-04-14 23:00:04,233] [DEBUG] found 1 subdevices:
[2020-04-14 23:00:04,233] [DEBUG] /dev/hidraw0: ASUS ROG SPATHA Receiver interface 0 [using as keyboard]
[2020-04-14 23:00:04,233] [DEBUG] opening keyboard subdevice
[2020-04-14 23:00:04,233] [DEBUG] opening control subdevice
[2020-04-14 23:00:04,233] [DEBUG] setting profile to 1
[2020-04-14 23:00:04,233] [DEBUG] > 50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2020-04-14 23:00:13,102] [DEBUG] < 01 00 00 00 00 00 00 00 00
[2020-04-14 23:00:13,102] [DEBUG] getting profile
[2020-04-14 23:00:13,102] [DEBUG] > 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2020-04-14 23:00:13,174] [DEBUG] < 00 00 00 00 00 00 00 00 00
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv/__main__.py", line 177, in rogdrv_config
    profile = device.get_profile()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv/device.py", line 371, in get_profile
    return response[10] + 1
IndexError: index out of range
```

kyokenn, 2020-04-22:
This is a problem:
"[2020-04-14 23:00:04,233] [DEBUG] found 1 subdevices:"
There are should be at least 3 separate subdevices: mouse, keyboard, control.

Can you try the mouse in wired mode?

MykoIce, 2020-04-22:
Right, so.
When in wireless, I only get 1 subdevice, the receiver. 
When I plug the mouse in, I only get 2 subdevices, with the second one being used as a keyboard.

I think the issue is that the subdevices I´m getting are mouse and keyboard, but not control, as that´s where the debugger says the error happened. 

[2020-04-22 11:50:30,490] [DEBUG] opening control subdevice
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/__main__.py", line 165, in rogdrv_config
    device = get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 80, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 148, in __init__
    self._ctl = hidapi.Device(ctl_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 156, in __init__
    raise ValueError("Must provide either a DeviceInfo object, 'path' "
ValueError: Must provide either a DeviceInfo object, 'path' or 'vendor_id' and 'product_id'.
Exception ignored in: <bound method Device.__del__ of <hidapi.Device object at 0x7f96807d6400>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/hidapi.py", line 164, in __del__
    if self._device is not None:
AttributeError: 'Device' object has no attribute '_device'

kyokenn, 2020-04-22:
I don't known which subdevice should be used as control or keyboard, so I'm just guessing the subdevice's index. There are no autodetection.

You can try to set different indexes by yourself with something like:
```
    keyboard_interface = 1
    control_interface = 2
```
The index number starts with 0.

Most mices uses second (index 1) subdevice for keyboard and third (index 2) for control, but I have changed it for Strix Carry - https://github.com/kyokenn/rogdrv/blob/master/rogdrv/device.py#L646
You can use it as example.

You can try some combinations for Spatha. There is also separate Device class for Spatha in wireless mode.

MykoIce, 2020-04-22:
I did some data digging, and I found an interesting print from using dmesg. 

```
[52442.497015] usb 1-8: new full-speed USB device number 6 using xhci_hcd
[52442.647549] usb 1-8: New USB device found, idVendor=0b05, idProduct=181c, bcdDevice=42.01
[52442.647555] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[52442.647559] usb 1-8: Product: ROG SPATHA
[52442.647562] usb 1-8: Manufacturer: ASUS
[52442.650369] input: ASUS ROG SPATHA as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:0B05:181C.0008/input/input22
[52442.650875] hid-generic 0003:0B05:181C.0008: input,hidraw6: USB HID v1.11 Mouse [ASUS ROG SPATHA] on usb-0000:00:14.0-8/input0
[52442.652775] input: ASUS ROG SPATHA Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:0B05:181C.0009/input/input23
[52442.709686] input: ASUS ROG SPATHA Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:0B05:181C.0009/input/input24
[52442.710007] input: ASUS ROG SPATHA Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:0B05:181C.0009/input/input26
```

I don´t know if it means anything, I´m just curious if the Control and Mouse should be in different inputs?

kyokenn, 2020-04-22:
So it even recognised the subdevices/interfaces: "Keyboard", "Consumer Control", "Mouse".
Too bad the "hid-generic" linux kernel module is not detected all of them.

This is how it looks with my Strix Carry mouse:
```
[38998.624277] usb 1-1: new full-speed USB device number 13 using xhci_hcd
[38998.775206] usb 1-1: New USB device found, idVendor=0b05, idProduct=18b4, bcdDevice= 1.09
[38998.775208] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[38998.775209] usb 1-1: Product: ROG STRIX CARRY
[38998.775209] usb 1-1: Manufacturer: ASUSTek
[38998.781341] input: ASUSTek ROG STRIX CARRY as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:0B05:18B4.0007/input/input25
[38998.840452] hid-generic 0003:0B05:18B4.0007: input,hidraw0: USB HID v1.11 Mouse [ASUSTek ROG STRIX CARRY] on usb-0000:00:14.0-1/input0
[38998.841331] hid-generic 0003:0B05:18B4.0008: hiddev0,hidraw1: USB HID v1.11 Device [ASUSTek ROG STRIX CARRY] on usb-0000:00:14.0-1/input1
[38998.842342] input: ASUSTek ROG STRIX CARRY as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:0B05:18B4.0009/input/input26
[38998.900443] hid-generic 0003:0B05:18B4.0009: input,hidraw2: USB HID v1.11 Keyboard [ASUSTek ROG STRIX CARRY] on usb-0000:00:14.0-1/input2
[38998.901315] input: ASUSTek ROG STRIX CARRY as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:0B05:18B4.000A/input/input27
[38998.960533] hid-generic 0003:0B05:18B4.000A: input,hidraw3: USB HID v1.11 Device [ASUSTek ROG STRIX CARRY] on usb-0000:00:14.0-1/input3
```

You can compare the outputs. I have 4 subdevices detected by "hid-generic":
- hidraw0 - mouse (not used in driver)
- hidraw1 - control (for changing mouse settings)
- hidraw2 - keyboard (for keyboard bindings)
- hidraw3 - unknown for me

What is your linux distribution, kernel version and xorg version?

MykoIce, 2020-04-22:
I am on the latest Linux Mint, which I believe is Ubuntu 18? 
Kernel is 5.3.0-46-generic
X.Org X Server 1.20.5
X Protocol Version 11, Revision 0

Some choice data from Xorg log.

```
[     6.904] 
X.Org X Server 1.20.5
X Protocol Version 11, Revision 0
[     6.904] Build Operating System: Linux 4.4.0-170-generic x86_64 Ubuntu
[     6.904] Current Operating System: Linux CMD 5.3.0-46-generic #38~18.04.1-Ubuntu SMP Tue Mar 31 04:17:56 UTC 2020 x86_64
[     6.904] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.3.0-46-generic root=UUID=fbd228f0-657a-4920-bbeb-a3f7065cd3fa ro quiet splash vt.handoff=1
[     6.904] Build Date: 18 December 2019  08:15:29AM
[     6.904] xorg-server-hwe-18.04 2:1.20.5+git20191008-0ubuntu1~18.04.1 (For technical support please see http://www.ubuntu.com/support) 
[     6.904] Current version of pixman: 0.34.0
[     6.904] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```

```
[    10.807] (II) config/udev: Adding input device ASUS ROG SPATHA Receiver (/dev/input/event9)
[    10.807] (**) ASUS ROG SPATHA Receiver: Applying InputClass "libinput pointer catchall"
[    10.807] (II) Using input driver 'libinput' for 'ASUS ROG SPATHA Receiver'
[    10.807] (**) ASUS ROG SPATHA Receiver: always reports core events
[    10.807] (**) Option "Device" "/dev/input/event9"
[    10.807] (**) Option "_source" "server/udev"
[    10.867] (II) event9  - ASUS ROG SPATHA Receiver: is tagged by udev as: Mouse
[    10.867] (II) event9  - ASUS ROG SPATHA Receiver: device is a pointer
[    10.867] (II) event9  - ASUS ROG SPATHA Receiver: device removed
[    10.906] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0B05:1824.0005/input/input9/event9"
[    10.906] (II) XINPUT: Adding extended input device "ASUS ROG SPATHA Receiver" (type: MOUSE, id 14)
[    10.906] (**) Option "AccelerationScheme" "none"
[    10.907] (**) ASUS ROG SPATHA Receiver: (accel) selected scheme none/0
[    10.907] (**) ASUS ROG SPATHA Receiver: (accel) acceleration factor: 2.000
[    10.907] (**) ASUS ROG SPATHA Receiver: (accel) acceleration threshold: 4
[    10.967] (II) event9  - ASUS ROG SPATHA Receiver: is tagged by udev as: Mouse
[    10.967] (II) event9  - ASUS ROG SPATHA Receiver: device is a pointer
[    10.967] (II) config/udev: Adding input device ASUS ROG SPATHA Receiver (/dev/input/mouse1)
```

MykoIce, 2020-04-22:
Also did print lsub -v 

```
Bus 001 Device 008: ID 0b05:1824 ASUSTek Computer, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1824 
  bcdDevice           d0.02
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     112
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     191
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

kyokenn, 2020-04-22:
It can't reads the descriptors:
```
         Report Descriptors: 
           ** UNAVAILABLE **
```
Make sure that you're running lsusb with sudo:
sudo lsusb -d 0b05:1824 -v

MykoIce, 2020-04-22:
Report Descriptors still comes up as unavailable, even with sudo.

MykoIce, 2020-04-22:
Right, so. Tried usbhid-dump, and I get this error.

```
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/002/001: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes
```.

But with sudo, usbhid-dump works.

`sudo usbhid-dump -m 0b05:1824
001:008:001:DESCRIPTOR         1587581758.831141
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 08 19 01 29 05 95 05
 75 01 91 02 95 01 75 03 91 01 05 07 19 00 2A FF
 00 15 00 26 FF 00 75 08 95 08 81 00 C0 05 0C 09
 01 A1 01 85 02 19 00 2A 9C 02 15 00 26 9C 02 95
 01 75 10 81 00 C0 05 08 09 4B A1 01 85 03 15 00
 25 00 19 00 29 00 75 08 95 06 81 01 85 04 96 04
 08 B1 01 85 05 95 4C B1 01 85 06 95 45 B1 01 85
 07 95 12 B1 01 85 08 95 14 B1 01 85 0A 96 04 10
 B1 01 85 0C 95 08 B1 01 85 10 95 20 B1 01 C0 05
 01 09 02 A1 01 85 0D 09 01 A1 00 05 01 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0

001:008:000:DESCRIPTOR         1587581758.834189
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01
 06 00 FF 09 40 95 02 75 08 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 05 01 09
 30 09 31 16 00 80 26 FF 7F 75 10 95 02 81 06 05
 0C 0A 38 02 75 08 95 01 15 81 25 7F 81 06 06 00
 FF 09 80 75 08 95 08 15 00 26 FF 00 B1 02 C0 C0
`

MykoIce, 2020-04-22:
And another info dump. 

```
einar@CMD:~/Desktop$ sudo rogdrv-config profile 1 --debug
[sudo] password for einar:         
[2020-04-22 19:00:40,987] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1877)
[2020-04-22 19:00:40,988] [DEBUG] 0 devices found
[2020-04-22 19:00:40,988] [DEBUG] searching for device Gladius2Alt (VendorID: 0x0B05, ProductID: 0x1845)
[2020-04-22 19:00:40,990] [DEBUG] 0 devices found
[2020-04-22 19:00:40,990] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2020-04-22 19:00:40,991] [DEBUG] 0 devices found
[2020-04-22 19:00:40,991] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2020-04-22 19:00:40,992] [DEBUG] 0 devices found
[2020-04-22 19:00:40,992] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2020-04-22 19:00:40,993] [DEBUG] 0 devices found
[2020-04-22 19:00:40,994] [DEBUG] searching for device StrixEvolve (VendorID: 0x0B05, ProductID: 0x185B)
[2020-04-22 19:00:40,995] [DEBUG] 0 devices found
[2020-04-22 19:00:40,995] [DEBUG] searching for device Spatha (VendorID: 0x0B05, ProductID: 0x181C)
[2020-04-22 19:00:40,996] [DEBUG] 0 devices found
[2020-04-22 19:00:40,996] [DEBUG] searching for device SpathaWireless (VendorID: 0x0B05, ProductID: 0x1824)
[2020-04-22 19:00:40,997] [DEBUG] found 1 subdevices:
[2020-04-22 19:00:40,997] [DEBUG] /dev/hidraw1: ASUS ROG SPATHA Receiver interface 0 [using as control]
[2020-04-22 19:00:40,997] [DEBUG] opening keyboard subdevice
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/__main__.py", line 165, in rogdrv_config
    device = get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 80, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 146, in __init__
    self._kbd = hidapi.Device(kbd_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 156, in __init__
    raise ValueError("Must provide either a DeviceInfo object, 'path' "
ValueError: Must provide either a DeviceInfo object, 'path' or 'vendor_id' and 'product_id'.
Exception ignored in: <bound method Device.__del__ of <hidapi.Device object at 0x7fc8b56e90b8>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/hidapi.py", line 164, in __del__
    if self._device is not None:
AttributeError: 'Device' object has no attribute '_device'
einar@CMD:~/Desktop$ sudo rogdrv-config profile 1 --debug
[2020-04-22 19:01:30,279] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1877)
[2020-04-22 19:01:30,281] [DEBUG] 0 devices found
[2020-04-22 19:01:30,281] [DEBUG] searching for device Gladius2Alt (VendorID: 0x0B05, ProductID: 0x1845)
[2020-04-22 19:01:30,282] [DEBUG] 0 devices found
[2020-04-22 19:01:30,282] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2020-04-22 19:01:30,284] [DEBUG] 0 devices found
[2020-04-22 19:01:30,284] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2020-04-22 19:01:30,285] [DEBUG] 0 devices found
[2020-04-22 19:01:30,285] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2020-04-22 19:01:30,287] [DEBUG] 0 devices found
[2020-04-22 19:01:30,287] [DEBUG] searching for device StrixEvolve (VendorID: 0x0B05, ProductID: 0x185B)
[2020-04-22 19:01:30,288] [DEBUG] 0 devices found
[2020-04-22 19:01:30,288] [DEBUG] searching for device Spatha (VendorID: 0x0B05, ProductID: 0x181C)
[2020-04-22 19:01:30,290] [DEBUG] found 2 subdevices:
[2020-04-22 19:01:30,290] [DEBUG] /dev/hidraw6: ASUS ROG SPATHA interface 0
[2020-04-22 19:01:30,290] [DEBUG] /dev/hidraw7: ASUS ROG SPATHA interface 1 [using as keyboard]
[2020-04-22 19:01:30,290] [DEBUG] opening keyboard subdevice
[2020-04-22 19:01:30,290] [DEBUG] opening control subdevice
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/__main__.py", line 165, in rogdrv_config
    device = get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 80, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 148, in __init__
    self._ctl = hidapi.Device(ctl_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 156, in __init__
    raise ValueError("Must provide either a DeviceInfo object, 'path' "
[... 49 lines cut ...]
[2020-04-22 19:02:31,719] [DEBUG] 0 devices found
[2020-04-22 19:02:31,719] [DEBUG] searching for device StrixEvolve (VendorID: 0x0B05, ProductID: 0x185B)
[2020-04-22 19:02:31,721] [DEBUG] 0 devices found
[2020-04-22 19:02:31,721] [DEBUG] searching for device Spatha (VendorID: 0x0B05, ProductID: 0x181C)
[2020-04-22 19:02:31,722] [DEBUG] found 2 subdevices:
[2020-04-22 19:02:31,722] [DEBUG] /dev/hidraw6: ASUS ROG SPATHA interface 0 [using as keyboard]
[2020-04-22 19:02:31,722] [DEBUG] /dev/hidraw7: ASUS ROG SPATHA interface 1 [using as control]
[2020-04-22 19:02:31,722] [DEBUG] opening keyboard subdevice
[... cut ...]
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/__main__.py", line 177, in rogdrv_config
    profile = device.get_profile()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 371, in get_profile
    return response[10] + 1
IndexError: index out of range
einar@CMD:~/Desktop$ 
```

MykoIce, 2020-04-22:
A usbhid dump on Spartha wired.

```
einar@CMD:~/Desktop$ sudo usbhid-dump -m 0b05:181c
001:010:001:DESCRIPTOR         1587582303.298379
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 08 19 01 29 05 75 01
 95 05 91 02 75 03 95 01 91 01 05 07 19 00 2A FF
 00 15 00 26 FF 00 75 08 95 08 81 00 C0 05 0C 09
 01 A1 01 85 02 19 00 2A 9C 02 15 00 26 9C 02 75
 10 95 01 81 00 C0 05 08 09 4B A1 01 85 03 25 FF
 29 FF 75 08 95 06 81 01 85 04 96 04 02 B1 01 85
 05 95 D1 B1 01 85 06 95 27 B1 01 85 07 95 12 B1
 01 85 08 95 14 B1 01 85 0A 96 04 08 B1 01 85 0C
 95 08 B1 01 85 0E 96 04 08 B1 01 85 10 95 20 B1
 01 C0 05 01 09 02 A1 01 85 0D 09 01 A1 00 05 01
 09 30 09 31 16 00 80 26 FF 7F 75 10 95 02 81 06
 C0 C0

001:010:000:DESCRIPTOR         1587582303.299330
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 03 95 01 81 01
 06 00 FF 09 40 15 81 25 7F 75 08 95 02 81 02 05
 01 09 38 95 01 81 06 05 01 09 30 09 31 16 00 80
 26 FF 7F 75 10 95 02 81 06 05 0C 0A 38 02 15 81
 25 7F 75 08 95 01 81 06 06 00 FF 09 80 75 08 95
 08 15 00 26 FF 00 B1 02 C0 C0

```

kyokenn, 2020-04-22:
So in wireless mode it have 1 subdevice, in wired mode - 2 subdevices: mouse and keyboard.
Every time the control subdevice is missing.
Did you have the latest firmware on your mouse?

MykoIce, 2020-04-22:
Yeah. There hasn´t been a new firmware for the mouse in... Over a year?

MykoIce, 2020-04-22:
Oh, hang on. There´s one from January. I´m going to jump on to VM and see if my ROG utility can detect my mouse firmware.

MykoIce, 2020-04-22:
Right, that was a ride. Has to get my laptop to run the firmware updater because there´s no linux variant, then I forgot my login stuff. Anyways, the receiver and mouse are now running the latest firmware, 1.40 and 1,80, respectively. 

But there doesn´t appear to be any difference. 

This one is the receiver.
```
sudo usbhid-dump -m 0b05:1824
001:014:001:DESCRIPTOR         1587588670.878306
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 08 19 01 29 05 95 05
 75 01 91 02 95 01 75 03 91 01 05 07 19 00 2A FF
 00 15 00 26 FF 00 75 08 95 08 81 00 C0 05 0C 09
 01 A1 01 85 02 19 00 2A 9C 02 15 00 26 9C 02 95
 01 75 10 81 00 C0 05 08 09 4B A1 01 85 03 15 00
 25 00 19 00 29 00 75 08 95 06 81 01 85 04 96 04
 08 B1 01 85 05 95 4C B1 01 85 06 95 45 B1 01 85
 07 95 12 B1 01 85 08 95 14 B1 01 85 0A 96 04 10
 B1 01 85 0C 95 08 B1 01 85 10 95 20 B1 01 C0 05
 01 09 02 A1 01 85 0D 09 01 A1 00 05 01 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0

001:014:000:DESCRIPTOR         1587588670.878943
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01
 06 00 FF 09 40 95 02 75 08 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 05 01 09
 30 09 31 16 00 80 26 FF 7F 75 10 95 02 81 06 05
 0C 0A 38 02 75 08 95 01 15 81 25 7F 81 06 06 00
 FF 09 80 75 08 95 08 15 00 26 FF 00 B1 02 C0 C0
```
And the lsusb

```
sudo lsusb -d 0b05:1824 -v

Bus 001 Device 014: ID 0b05:1824 ASUSTek Computer, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1824 
  bcdDevice           d0.02
  iManufacturer           1 ASUS
  iProduct                2 ROG SPATHA Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
[... 152 lines cut ...]
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     194
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
```

kyokenn, 2020-04-22:
Yeah. Still not good.

I think you have to debug the mouse by yourself, because I don't have this type of mouse.
You can do it with any USB debugging software, basically you will need a USB packet sniffer.
You can use Wireshark for it - https://wiki.wireshark.org/CaptureSetup/USB

So you will need to start a USB debugger and play around with filters to exclude unwanted interfaces with packets like mouse movement events or clicks.
Then you'll need to open the ASUS driver app (Armoury), try to switch profiles or change some mouse settings and look for USB packets.
This is how I made this driver.

MykoIce, 2020-04-22:
I can get the data needed, no problem, I just have no idea how to put it into code. But if it´s just replacing the values and numbers in this code, I shooouuld be able to? We´ll see. I´ll keep you updated.

Galaxis78, 2020-05-25:
Hello
Here's my problem:
When inserting my ROG Spatha wired mouse
Lights flash once very quickly when connecting
then nothing.
The one if this detect in VID 12CF and PID 0036.
During the firmware update the mouse is not detected.
I tested all the info I gleaned from the ROG site but nothing helped
I don't have the correct VID and PID.

Is there a way to bluff the VID and PID to reprogram it
if someone can give me a little information
it would be of great help to me.
Internal chip SN32F248FG
ARM Cortex-M0，50MHz (45DMIPS) ， 64KB Flash ROM

kyokenn, 2020-05-25:
Did you broken your mouse?
Unfortunately I haven't reverse engineered the firmware update protocol. So you have to use the official firmware update tools from ASUS.

Galaxis78, 2020-05-26:
Yes my mouse is broken, connected but with a bad vid and pid
i tested all softwares Asus but nothink for this error of vid and Pid.
Thanks anyway

ajk0024, 2021-02-07:
Hi I have managed to install the program but now once i try run rogdrv it outputs this:
`Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/__main__.py", line 24, in <module>
    from .device import EventHandler, get_device
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/device.py", line 22, in <module>
    from . import defs, hid
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/hid.py", line 21, in <module>
    import hid
ModuleNotFoundError: No module named 'hid'
`
I am running ubuntu 20.04 if that helps

kyokenn, 2021-02-08:
> ModuleNotFoundError: No module named 'hid' 

You need to install python3-hid package:

```
sudo apt install python3-hid
```

I also added this on into requirements section of the readme.

ajk0024, 2021-02-08:
Seemed to have gotten rid of the hid problem however now this comes up
`Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/__main__.py", line 46, in loop
    e = device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv')()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/__main__.py", line 55, in rogdrv
    gtk3_main(device)
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/gtk3.py", line 148, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'
`

kyokenn, 2021-02-08:
Can you try running:
```
rogdrv-config --debug dump
```
Just to make sure that driver can detect the mouse.

ajk0024, 2021-02-08:
Wow didn't think you'd reply so quickly thank you
`[2021-02-08 21:09:14,184] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,184] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1877)
[2021-02-08 21:09:14,187] [DEBUG] 0 devices found
[2021-02-08 21:09:14,187] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,187] [DEBUG] searching for device Gladius2Alt (VendorID: 0x0B05, ProductID: 0x1845)
[2021-02-08 21:09:14,187] [DEBUG] 0 devices found
[2021-02-08 21:09:14,187] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,187] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2021-02-08 21:09:14,188] [DEBUG] 0 devices found
[2021-02-08 21:09:14,188] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,188] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2021-02-08 21:09:14,188] [DEBUG] 0 devices found
[2021-02-08 21:09:14,188] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,188] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2021-02-08 21:09:14,188] [DEBUG] 0 devices found
[2021-02-08 21:09:14,188] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,188] [DEBUG] searching for device StrixEvolve (VendorID: 0x0B05, ProductID: 0x185B)
[2021-02-08 21:09:14,188] [DEBUG] 0 devices found
[2021-02-08 21:09:14,188] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,188] [DEBUG] searching for device Spatha (VendorID: 0x0B05, ProductID: 0x181C)
[2021-02-08 21:09:14,188] [DEBUG] 0 devices found
[2021-02-08 21:09:14,188] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,188] [DEBUG] searching for device SpathaWireless (VendorID: 0x0B05, ProductID: 0x1824)
[2021-02-08 21:09:14,188] [DEBUG] found 2 subdevices:
[2021-02-08 21:09:14,188] [DEBUG] 0005:0002:00:   interface 0
[2021-02-08 21:09:14,188] [DEBUG] 0005:0002:01:   interface 1 [using as keyboard]
[2021-02-08 21:09:14,188] [DEBUG] control subdevice not found
[2021-02-08 21:09:14,188] [DEBUG] using HID interface: hidapi
[2021-02-08 21:09:14,188] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2021-02-08 21:09:14,188] [DEBUG] 0 devices found
Device not found
`

I'm using a rog spatha in wireless mode so looks like it successfully found it and my keyboard

kyokenn, 2021-02-09:
I also need output (mouse-related part) of
```sudo dmesg```
after connecting the mouse (you can disconnect and connect it again)

I'm looking for output like this one - https://github.com/kyokenn/rogdrv/issues/1#issuecomment-617908138

ajk0024, 2021-02-09:
`[ 1695.137455] usb 5-3: USB disconnect, device number 2
[ 1727.559073] usb 5-3: new full-speed USB device number 4 using xhci_hcd
[ 1728.175788] usb 5-3: device descriptor read/64, error -71
[ 1728.453443] usb 5-3: New USB device found, idVendor=0b05, idProduct=1824, bcdDevice=d0.02
[ 1728.453446] usb 5-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1728.453447] usb 5-3: Product: ROG SPATHA Receiver
[ 1728.453448] usb 5-3: Manufacturer: ASUS
[ 1728.465650] input: ASUS ROG SPATHA Receiver as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-3/5-3:1.0/0003:0B05:1824.000C/input/input28
[ 1728.465789] hid-generic 0003:0B05:1824.000C: input,hidraw0: USB HID v1.11 Mouse [ASUS ROG SPATHA Receiver] on usb-0000:08:00.3-3/input0
[ 1728.470855] input: ASUS ROG SPATHA Receiver Keyboard as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-3/5-3:1.1/0003:0B05:1824.000D/input/input29
[ 1728.527278] input: ASUS ROG SPATHA Receiver Consumer Control as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-3/5-3:1.1/0003:0B05:1824.000D/input/input30
[ 1728.527358] input: ASUS ROG SPATHA Receiver Mouse as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-3/5-3:1.1/0003:0B05:1824.000D/input/input32
[ 1728.527470] hid-generic 0003:0B05:1824.000D: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [ASUS ROG SPATHA Receiver] on usb-0000:08:00.3-3/input1`

This is what I am assuming you wanted as there was a lot more output

kyokenn, 2021-02-09:
So the hid-generic kernel driver still fails:
```
device descriptor read/64, error -71
```

This device was not detected by hid-generic:
```
input: ASUS ROG SPATHA Receiver Consumer Control as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-3/5-3:1.1/0003:0B05:1824.000D/input/input30
```
But it's required for changing mouse settings.

So you got only 2 HID devices:
```
hid-generic 0003:0B05:1824.000C: input,hidraw0: USB HID v1.11 Mouse [ASUS ROG SPATHA Receiver] on usb-0000:08:00.3-3/input0
hid-generic 0003:0B05:1824.000D: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [ASUS ROG SPATHA Receiver] on usb-0000:08:00.3-3/input1
```

ajk0024, 2021-02-09:
So what would you suggest to do to fix this? Apologies I am very new to linux

kyokenn, 2021-02-09:
As you can see many people tried to get this mouse working on Linux, but no one succeed. So you have to use ASUS Armoury on Windows to change the mouse settings. The settings are saved into the mouse, so this should be fine.

ajk0024, 2021-04-01:
Do you know of any programs for windows that I could use to change my mouse button functions more specifically as all of the options on ASUS Armoury are very limited.

closed 2021-07-01 by kyokenn