kyokenn/rogdrv #3: [Feature request] Please add support for another mouse

paulius-adamonis, 2019-08-25:
Hello, I have an ASUS ROG Strix Evolve mouse that is unsupported. Please, can you add support for it? Mouse id that I got from lsusb: "Bus 001 Device 004: ID 0b05:185b ASUSTek Computer, Inc.".

Kagetsuki, 2019-08-26:
Generally to check if any changes actually work @kyokenn would have to have that specific device. It would be best if you could fork and clone this repository, edit this file: https://github.com/kyokenn/rogdrv/blob/master/udev/50-rogdrv.rules and https://github.com/kyokenn/rogdrv/blob/master/rogdrv/device.py to add your device, test, and if it works commit/push and submit a merge request.

kyokenn, 2019-08-27:
I have added USB ID's of the mouse. Now you can check if it works.

closed 2020-08-18 by kyokenn