kyokenn/rogdrv #4: Error while running rogdrv

drevofil, 2019-10-02:
I bought new ASUS ROG Spatha, and try to use your drivers. After installing from manual i get this error

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/bas/.local/lib/python3.6/site-packages/rogdrv/__main__.py", line 44, in loop
    e = device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'

Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv')()
  File "/home/bas/.local/lib/python3.6/site-packages/rogdrv/__main__.py", line 53, in rogdrv
    gtk3_main(device)
  File "/home/bas/.local/lib/python3.6/site-packages/rogdrv/gtk3.py", line 149, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'

```

and rogdrv-config profile (or dump) says
`Device not found`

I read previous issue, and tried to experiment with device ID in udev conf (i have Bus 001 Device 008: ID 0b05:181c ASUSTek Computer, Inc. ), but no result.

drevofil, 2019-10-02:
Also, OS is KDE Neon 18.04 (Ubuntu fork with KDE, you know)

kyokenn, 2019-10-06:
Does it work with "sudo" ?

`sudo rogdrv-config profile `

drevofil, 2019-10-06:
Nope, sudo doesn't helps. I used windows workstation to bind buttons and save profile. Btw, buttons are recognised as 7 8 9 = + are they the same as keyboard input? On rat9 extra buttons recognised as Mouse 4 and Mouse 5, can reach same result on rog's mouse?

kyokenn, 2019-10-06:
> On rat9 extra buttons recognised as Mouse 4 and Mouse 5, can reach same result on rog's mouse?

Yes. Mouse buttons 4, 5 are also called "browser back, forward". If you bind "back" and "forward" to some of your buttons, they will be recognised as buttons 4 and 5 on Linux.

For the rogdrv-config you can try to mess around with control_interface variable. I don't have this mouse, so I can't go with experimenting by myself. You can try to set control_interface to values 0, 1, 2, 3, 4, like I did for the Strix Carry - https://github.com/kyokenn/rogdrv/blob/2661b3fc03da11d2bf6df8932aae2295483c3644/rogdrv/device.py#L607
into the Spatha device definition - https://github.com/kyokenn/rogdrv/blob/2661b3fc03da11d2bf6df8932aae2295483c3644/rogdrv/device.py#L630

drevofil, 2019-10-06:
I will try, thanks. How to deal with product_id of Spatha device? On wireless its 181c, on cord - 1824

kyokenn, 2019-10-06:
> How to deal with product_id of Spatha device? On wireless its 181c, on cord - 1824

Wow. I didn't know about that. Nice to know. I'm going to put another device definition for the wireless version.

drevofil, 2019-10-06:
Also, i can share you any information or logs of thid mouse to improve your project, and I will get full functionality mouse driver. I can PM you to email in profile.

kyokenn, 2019-10-06:
Yeah, sure. I've made some fixes for the logging, so you can use the "--debug" option for the rogdrv-config.

drevofil, 2019-10-06:
Sorry, I give you wrong information about device_id, on cord = 181c, wireless = 1824

kyokenn, 2019-10-06:
> Sorry, I give you wrong information about device_id, on cord = 181c, wireless = 1824

ok, fixed

closed 2020-08-18 by kyokenn