kyokenn/rogdrv #9: new model 'ROG GLADIUS' 0b05:181a not working

jjfaith, 2020-05-30:
Hi,
I recently got a new 'ROG GLADIUS' mouse and was trying your driver on Linux Mint 19.3.
First thing I found was that I needed to install python3-setuptools perhaps you could add this to requirements.
However I was unable to get the driver to work as the mouse seems to be an unknown model.
```
sudo lsusb -v -s 1:5
Bus 001 Device 005: ID 0b05:181a ASUSTek Computer, Inc.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x181a
  bcdDevice           29.02
  iManufacturer           1 ASUS
  iProduct                2 ROG GLADIUS
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           66
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     112
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     186
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled

sudo usbhid-dump -m0b05:181a
libusb: warning [seek_to_next_config] config length mismatch wTotalLength 66 real 59
libusb: warning [seek_to_next_config] config length mismatch wTotalLength 66 real 59
001:006:001:DESCRIPTOR         1590846424.470023
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 08 19 01 29 05 95 05
 75 01 91 02 95 01 75 03 91 01 05 07 19 00 2A FF
 00 15 00 26 FF 00 75 08 95 08 81 00 C0 05 0C 09
 01 A1 01 85 02 19 00 2A 9C 02 15 00 26 9C 02 95
 01 75 10 81 00 C0 05 08 09 4B A1 01 85 03 15 00
 25 00 19 00 29 00 75 08 95 06 81 01 85 04 96 04
 08 B1 01 85 05 95 43 B1 01 85 06 95 33 B1 01 85
 07 95 12 B1 01 85 0A 96 04 08 B1 01 85 0B 96 84
 01 B1 01 85 0C 95 08 B1 01 C0 05 01 09 02 A1 01
 85 0D 09 01 A1 00 05 01 09 30 09 31 16 00 80 26
 FF 7F 75 10 95 02 81 06 C0 C0

001:006:000:DESCRIPTOR         1590846424.470668
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01
 06 00 FF 09 40 95 02 75 08 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 05 01 09
 30 09 31 16 00 80 26 FF 7F 75 10 95 02 81 06 05
 0C 0A 38 02 75 08 95 01 15 81 25 7F 81 06 06 00
 FF 09 80 75 08 95 08 15 00 26 FF 00 B1 02 C0 C0
```
dmesg output when mouse plugged in:-
```
[14854.940565] usb 1-1.6.1: new full-speed USB device number 6 using ehci-pci
[14855.170379] usb 1-1.6.1: config 1 has an invalid descriptor of length 0, skipping remainder of the config
[14855.170903] usb 1-1.6.1: New USB device found, idVendor=0b05, idProduct=181a, bcdDevice=29.02
[14855.170906] usb 1-1.6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[14855.170908] usb 1-1.6.1: Product: ROG GLADIUS
[14855.170909] usb 1-1.6.1: Manufacturer: ASUS
[14855.172095] input: ASUS ROG GLADIUS as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.0/0003:0B05:181A.0004/input/input15
[14855.228871] hid-generic 0003:0B05:181A.0004: input,hidraw1: USB HID v1.11 Mouse [ASUS ROG GLADIUS] on usb-0000:00:1a.0-1.6.1/input0
[14855.230231] input: ASUS ROG GLADIUS Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.1/0003:0B05:181A.0005/input/input16
[14855.288930] input: ASUS ROG GLADIUS Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.1/0003:0B05:181A.0005/input/input17
[14855.289108] input: ASUS ROG GLADIUS Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.1/0003:0B05:181A.0005/input/input19
[14855.289520] hid-generic 0003:0B05:181A.0005: input,hiddev0,hidraw2: USB HID v1.11 Keyboard [ASUS ROG GLADIUS] on usb-0000:00:1a.0-1.6.1/input1
```
So to me it looks like
  /dev/hidraw1 is the mouse
  /dev/hidraw2 is the keyboard

I added following to 50-rogdrv.rules:-
```
  KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="181a", MODE="0666"
  KERNEL=="hidraw*", SUBSYSTEM=="hidraw", KERNELS=="0003:0b05:181a.*", MODE="0666"
```
```
ls -l /dev/hid*
  crw------- 1 root root 243, 0 May 30 09:53 /dev/hidraw0
  crw-rw-rw- 1 root root 243, 1 May 30 14:00 /dev/hidraw1
  crw-rw-rw- 1 root root 243, 2 May 30 14:00 /dev/hidraw2
```
so modified 50-rogdrv.rules seems to work

I tried adding following to device.py:-
```
class Gladius(Gladius2):
    product_id = 0x181a
    control_interface = 2
    keyboard_interface = 1
```

And tried misc values for control_interface but nothing worked.
Is the problem that there is no /dev/hidraw3 ?

e.g. if set as above when I run, rogdrv-config --debug dpi, I get:-
```
[2020-05-30 14:31:26,955] [DEBUG] searching for device Gladius (VendorID: 0x0B05, ProductID: 0x181A)
[2020-05-30 14:31:26,956] [DEBUG] found 2 subdevices:
[2020-05-30 14:31:26,956] [DEBUG] /dev/hidraw1: ASUS ROG GLADIUS interface 0
[2020-05-30 14:31:26,956] [DEBUG] /dev/hidraw2: ASUS ROG GLADIUS interface 1 [using as keyboard]
[2020-05-30 14:31:26,956] [DEBUG] opening keyboard subdevice
[2020-05-30 14:31:26,956] [DEBUG] opening control subdevice
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/__main__.py", line 202, in rogdrv_config
    device = get_device()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 80, in get_device
    return device_class()
  File "/usr/local/lib/python3.6/dist-packages/rogdrv-0.0.5-py3.6.egg/rogdrv/device.py", line 148, in __init__
    self._ctl = hidapi.Device(ctl_info)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 156, in __init__
    raise ValueError("Must provide either a DeviceInfo object, 'path' "
ValueError: Must provide either a DeviceInfo object, 'path' or 'vendor_id' and 'product_id'.
Exception ignored in: <bound method Device.__del__ of <hidapi.Device object at 0x7f3b1a5cfeb8>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/hidapi.py", line 164, in __del__
    if self._device is not None:
AttributeError: 'Device' object has no attribute '_device'
```

kyokenn, 2020-05-30:
>  Is the problem that there is no /dev/hidraw3 ?
Yes, it is.

It looks like that control device was still detected by usb input driver:
`[14855.288930] input: ASUS ROG GLADIUS Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.1/0003:0B05:181A.0005/input/input17`

But it's unreadable by lsusb and hid. So you have mouse and keyboard only as available descriptors:
```
     Interface Descriptor:
      bInterfaceProtocol      2 Mouse
     Interface Descriptor:
      bInterfaceProtocol      1 Keyboard
```

I think this could be related to this lines:
`[14855.170379] usb 1-1.6.1: config 1 has an invalid descriptor of length 0, skipping remainder of the config`

```
libusb: warning [seek_to_next_config] config length mismatch wTotalLength 66 real 59
libusb: warning [seek_to_next_config] config length mismatch wTotalLength 66 real 59
```

This could be a linux kernel bug or some kind of protocol obfuscation by ASUS.
What linux kernel version are you on?

jjfaith, 2020-05-30:
kernel version:5.3.0-53-generic #47~18.04.1-Ubuntu SMP

jjfaith, 2020-05-30:
Found some more info using udevadm, note that input47 is the Consumer Control input(after mouse reattached a few times)
```
sudo udevadm info --path=/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.1/0003:0B05:181A.0011/input/input47
P: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.1/0003:0B05:181A.0011/input/input47
E: ABS=100000000
E: DEVPATH=/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.1/1-1.6.1:1.1/0003:0B05:181A.0011/input/input47
E: EV=1f
E: ID_BUS=usb
E: ID_FOR_SEAT=input-pci-0000_00_1a_0-usb-0_1_6_1_1_1
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_MODEL=ROG_GLADIUS
E: ID_MODEL_ENC=ROG\x20GLADIUS
E: ID_MODEL_ID=181a
E: ID_PATH=pci-0000:00:1a.0-usb-0:1.6.1:1.1
E: ID_PATH_TAG=pci-0000_00_1a_0-usb-0_1_6_1_1_1
E: ID_REVISION=2902
E: ID_SERIAL=ASUS_ROG_GLADIUS
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030102:030001:
E: ID_USB_INTERFACE_NUM=01
E: ID_VENDOR=ASUS
E: ID_VENDOR_ENC=ASUS
E: ID_VENDOR_ID=0b05
E: KEY=300ff 0 0 483ffff17aff32d bfd4444600000000 1 130ff38b17c000 677bfad9415fed 9ed68000004400 10000002
E: MODALIAS=input:b0003v0B05p181Ae0111-e0,1,2,3,4,k71,72,73,74,77,80,82,83,85,86,87,88,89,8A,8B,8C,8E,90,96,98,9B,9C,9E,9F,A1,A3,A4,A5,A6,A7,A8,A9,AB,AC,AD,AE,B0,B1,B2,B5,B6,CE,CF,D0,D1,D2,D4,D8,D9,DB,DF,E0,E1,E4,E5,E6,E7,E8,E9,EA,EB,F0,F1,F4,100,161,162,166,16A,16E,172,174,176,177,178,179,17A,17B,17C,17D,17F,180,182,183,185,188,189,18C,18D,18E,18F,190,191,192,193,195,197,198,199,19A,19C,1A0,1A1,1A2,1A3,1A4,1A5,1A6,1A7,1A8,1A9,1AA,1AB,1AC,1AD,1AE,1AF,1B0,1B1,1B7,1BA,240,241,242,243,244,245,246,247,250,251,r6,C,a20,m4,lsfw
E: MSC=10
E: NAME="ASUS ROG GLADIUS Consumer Control"
E: PHYS="usb-0000:00:1a.0-1.6.1/input1"
E: PRODUCT=3/b05/181a/111
E: PROP=0
E: REL=1040
E: SUBSYSTEM=input
E: TAGS=:seat:
E: UNIQ=""
E: USEC_INITIALIZED=30982702192
```

kyokenn, 2020-05-30:
Can you try to run lsusb with debugging enabled?
`LIBUSB_DEBUG=9 sudo -E lsusb -v -d 0b05:181a`

jjfaith, 2020-05-30:
```
LIBUSB_DEBUG=9 sudo -E lsusb -v -d 0b05:181a
libusb: debug [libusb_init] created default context
libusb: debug [libusb_init] libusb v1.0.21.11156
libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
libusb: debug [op_init] bulk continuation flag supported
libusb: debug [op_init] zero length packet flag supported
libusb: debug [op_init] sysfs can relate devices
libusb: debug [op_init] sysfs has complete descriptors
libusb: debug [linux_udev_event_thread_main] udev event thread entering.
libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
libusb: debug [linux_get_device_address] scan usb3
libusb: debug [linux_get_device_address] bus=3 dev=1
libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
libusb: debug [linux_get_device_address] getting address for device: 3-4 detached: 0
libusb: debug [linux_get_device_address] scan 3-4
libusb: debug [linux_get_device_address] bus=3 dev=2
libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
libusb: debug [linux_get_parent_info] Dev 0x55deb0d98c10 (3-4) has parent 0x55deb0d95b10 (usb3) port 4
libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
libusb: debug [linux_get_device_address] scan usb4
libusb: debug [linux_get_device_address] bus=4 dev=1
libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
libusb: debug [linux_get_device_address] scan usb1
libusb: debug [linux_get_device_address] bus=1 dev=1
libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
libusb: debug [linux_get_device_address] scan 1-1
libusb: debug [linux_get_device_address] bus=1 dev=2
libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
libusb: debug [linux_get_parent_info] Dev 0x55deb0d96870 (1-1) has parent 0x55deb0d98710 (usb1) port 1
libusb: debug [linux_get_device_address] getting address for device: 1-1.3 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.3
libusb: debug [linux_get_device_address] bus=1 dev=3
libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
libusb: debug [linux_get_parent_info] Dev 0x55deb0d95640 (1-1.3) has parent 0x55deb0d96870 (1-1) port 3
libusb: debug [linux_get_device_address] getting address for device: 1-1.6 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.6
libusb: debug [linux_get_device_address] bus=1 dev=4
libusb: debug [linux_enumerate_device] busnum 1 devaddr 4 session_id 260
libusb: debug [linux_enumerate_device] allocating new device for 1/4 (session 260)
libusb: debug [linux_get_parent_info] Dev 0x55deb0ea8ec0 (1-1.6) has parent 0x55deb0d96870 (1-1) port 6
libusb: debug [linux_get_device_address] getting address for device: 1-1.6.1 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.6.1
libusb: debug [linux_get_device_address] bus=1 dev=9
libusb: debug [linux_enumerate_device] busnum 1 devaddr 9 session_id 265
libusb: debug [linux_enumerate_device] allocating new device for 1/9 (session 265)
libusb: debug [linux_get_parent_info] Dev 0x55deb0d97af0 (1-1.6.1) has parent 0x55deb0ea8ec0 (1-1.6) port 1
libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
libusb: debug [linux_get_device_address] scan usb2
libusb: debug [linux_get_device_address] bus=2 dev=1
libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
libusb: debug [linux_get_device_address] scan 2-1
libusb: debug [linux_get_device_address] bus=2 dev=2
libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
libusb: debug [linux_get_parent_info] Dev 0x55deb0d97ec0 (2-1) has parent 0x55deb0ea37c0 (usb2) port 1
libusb: debug [usbi_add_pollfd] add fd 6 events 1
libusb: debug [usbi_io_init] using timerfd for timeouts
libusb: debug [usbi_add_pollfd] add fd 8 events 1
libusb: debug [libusb_get_device_list] 
[... 187 lines cut ...]
libusb: debug [usbi_handle_transfer_completion] transfer 0x55deb0ea3dc0 has callback 0x7f5797e6eb30
libusb: debug [sync_transfer_cb] actual_length=2
libusb: debug [libusb_free_transfer] transfer 0x55deb0ea3dc0
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
libusb: debug [libusb_close] 
libusb: debug [usbi_remove_pollfd] remove fd 9
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_exit] 
libusb: debug [libusb_exit] destroying default context
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll fds modified, reallocating
libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
libusb: debug [handle_events] poll() returned 0
libusb: debug [usbi_remove_pollfd] remove fd 6
libusb: debug [usbi_remove_pollfd] remove fd 8
libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

kyokenn, 2020-05-30:
Looks like a firmware bug. You can try to update the firmware.

jjfaith, 2020-05-31:
I managed to find an old windows 7 laptop and got the Armoury app installed.
Firmware version is 1.30 which seems to be the latest, from searching the web this appears to be about 4 years old so I don't think an update is likely.
I used the armoury software to turn off the leds which is all I really wanted to do.
Thanks for your help.

closed 2021-07-01 by kyokenn