kyokenn/rogdrv #11: Error on bind button

reijota, 2020-08-18:
When I execute the following command for assign KEY_1 to MOUSE_BACKWARD

`rogdrv-config bind 30 228`

I get the error:

`Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.7/dist-packages/rogdrv-0.0.5-py3.7.egg/rogdrv/__main__.py", line 123, in rogdrv_config
    device.bind(button, action)
  File "/usr/local/lib/python3.7/dist-packages/rogdrv-0.0.5-py3.7.egg/rogdrv/device.py", line 321, in bind
    request[4] = defs.BUTTON_SLOTS[button]
KeyError: 30`

Other commands like rogdrv-config profile 2 works perfect.

kyokenn, 2020-08-18:
The first digit is mouse button you want to bind, the second digit is the key code.
For example, if you want to bind your left mouse button (which is button 1) to MOUSE_BACKWARD, then you should do:
`rogdrv-config bind 1 228`
This is just an example, so don't try it, because you'll want to keep your left mouse button as is!
"KEY_1" is just an keyboard's key code for "1", you have mistaken it for mouse button.

reijota, 2020-08-19:
I use the list of the command `rogdrv-config actions`

`Keyboard actions:
  0 (0x00): UNDEFINED
  4 (0x04): KEY_A
  5 (0x05): KEY_B
  6 (0x06): KEY_C
  7 (0x07): KEY_D
  8 (0x08): KEY_E
  9 (0x09): KEY_F
  10 (0x0A): KEY_G
  11 (0x0B): KEY_H
  12 (0x0C): KEY_I
  13 (0x0D): KEY_J
  14 (0x0E): KEY_K
  15 (0x0F): KEY_L
  16 (0x10): KEY_M
  17 (0x11): KEY_N
  18 (0x12): KEY_O
  19 (0x13): KEY_P
  20 (0x14): KEY_Q
  21 (0x15): KEY_R
  22 (0x16): KEY_S
  23 (0x17): KEY_T
  24 (0x18): KEY_U
  25 (0x19): KEY_V
  26 (0x1A): KEY_W
  27 (0x1B): KEY_X
  28 (0x1C): KEY_Y
  29 (0x1D): KEY_Z
  30 (0x1E): KEY_1
  31 (0x1F): KEY_2
  32 (0x20): KEY_3
  33 (0x21): KEY_4
  34 (0x22): KEY_5
  35 (0x23): KEY_6
  36 (0x24): KEY_7
  37 (0x25): KEY_8
  38 (0x26): KEY_9
  39 (0x27): KEY_0
  40 (0x28): KEY_ENTER
  41 (0x29): KEY_ESC
  42 (0x2A): KEY_BACKSPACE
  43 (0x2B): KEY_TAB
  44 (0x2C): KEY_SPACE
  45 (0x2D): KEY_MINUS
  46 (0x2E): KEY_KPPLUS
  53 (0x35): KEY_GRAVE
  54 (0x36): KEY_EQUAL
  56 (0x38): KEY_SLASH
  58 (0x3A): KEY_F1
  59 (0x3B): KEY_F2
  60 (0x3C): KEY_F3
  61 (0x3D): KEY_F4
  62 (0x3E): KEY_F5
  63 (0x3F): KEY_F6
  64 (0x40): KEY_F7
  65 (0x41): KEY_F8
  66 (0x42): KEY_F9
  67 (0x43): KEY_F10
  68 (0x44): KEY_F11
  69 (0x45): KEY_F12
  74 (0x4A): KEY_HOME
  75 (0x4B): KEY_PAGEUP
  76 (0x4C): KEY_DELETE
  78 (0x4E): KEY_PAGEDOWN
  79 (0x4F): KEY_RIGHT
  80 (0x50): KEY_LEFT
  81 (0x51): KEY_DOWN
  82 (0x52): KEY_UP
  89 (0x59): KEY_KP1
  90 (0x5A): KEY_KP2
  91 (0x5B): KEY_KP3
  92 (0x5C): KEY_KP4
  93 (0x5D): KEY_KP5
  94 (0x5E): KEY_KP6
  95 (0x5F): KEY_KP7
  96 (0x60): KEY_KP8
  97 (0x61): KEY_KP9

Mouse actions:
  240 (0xF0): MOUSE_LEFT
  241 (0xF1): MOUSE_RIGHT
  242 (0xF2): MOUSE_MIDDLE
  228 (0xE4): MOUSE_BACKWARD
  229 (0xE5): MOUSE_FORWARD
  230 (0xE6): MOUSE_DPI
  232 (0xE8): MOUSE_SCROLL_UP
  233 (0xE9): MOUSE_SCROLL_DOWN
  255 (0xFF): MOUSE_DISABLED
  225 (0xE1): MOUSE_BACKWARD_ALT
  226 (0xE2): MOUSE_FORWARD_ALT`

And in `rogdrv-config --help` especifies: 

`rogdrv-config bind [button action]              - bind a button or display bindings
    button: button no. (1-99)
    action: action code (241 or 0xF1 or 0xf1)`

if this is not the list i have to use, what is it? How can I know which are the numbers of the side buttons?

With the command `rogdrv-config bind 228 30` I have the same error

kyokenn, 2020-08-19:
> How can I know which are the numbers of the side buttons?

`rogdrv-config bind` will give you the list of your mouse buttons with numbers and current actions.

It will looks like (this on is from my mouse):
```
BTN | TYPE  | CODE - ACTION
----|-------|--------------
01  | MOUSE | 0xF0 - MOUSE_LEFT
02  | MOUSE | 0xF2 - MOUSE_MIDDLE
03  | MOUSE | 0xF1 - MOUSE_RIGHT
04  | MOUSE | 0xE8 - MOUSE_SCROLL_UP
05  | MOUSE | 0xE9 - MOUSE_SCROLL_DOWN
06  | MOUSE | 0xE6 - MOUSE_DPI
07  | MOUSE | 0xE4 - MOUSE_BACKWARD
08  | MOUSE | 0xE5 - MOUSE_FORWARD
```

reijota, 2020-08-19:
Ok, First I have run again install_udev.sh and now the command work.

`rogdrv-config bind 7 30`
`rogdrv-config bind 8 31`

`BTN | TYPE  | CODE - ACTION
----|-------|--------------
01  | MOUSE | 0xF0 - MOUSE_LEFT
02  | MOUSE | 0xF1 - MOUSE_RIGHT
03  | MOUSE | 0xF2 - MOUSE_MIDDLE
04  | MOUSE | 0xE8 - MOUSE_SCROLL_UP
05  | MOUSE | 0xE9 - MOUSE_SCROLL_DOWN
06  | MOUSE | 0xE6 - MOUSE_DPI
07  | KEYB. | 0x1E - KEY_1
08  | KEYB. | 0x1F - KEY_2`

But When I press the lateral buttons, it dont do nothing, The mouse is the ROG GLADIUS II and the system Ubuntu 19.10, 
May it not be compatible?

kyokenn, 2020-08-19:
Ok, good. You have set your mouse buttons to keyboard's keys "1" and "2".
Now you need to start `rogdrv` and keep it running (you can also add it to autostart) all the time to get those keyboard's keys to work.

reijota, 2020-08-19:
> Ok, good. You have set your mouse buttons to keyboard's keys "1" and "2".

Yes :+1: 

> Now you need to start `rogdrv` and keep it running (you can also add it to autostart) all the time to get those keyboard's keys to work.

Yes, it is in autostart and I restart the computer and the button configuration on restart is still saved perfectly.
but when i press the lateral button still does nothing. :(

closed 2021-07-01 by kyokenn