kyokenn/rogdrv #14: I have trouble....

Borgensgaard2, 2020-11-16:
Hi, I have some difficulty.. I try to change my mouse settings, but I get this:

Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 11, in <module>
    load_entry_point('rogdrv==0.0.5', 'console_scripts', 'rogdrv-config')()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/__main__.py", line 202, in rogdrv_config
    device = get_device()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/device.py", line 79, in get_device
    return device_class()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/device.py", line 148, in __init__
    self._ctl = hid.open_device(control)
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.0.5-py3.8.egg/rogdrv/hid.py", line 44, in open_device
    device.open_path(info['path'])
  File "hid.pyx", line 72, in hid.device.open_path
OSError: open failed

I have tried this, as you suggested..

You need r/w permissions for /dev/hidrawX file of your mouse.

You can solve this by installing a custom udev rules:

But I get this:

sudo cp -f udev/50-rogdrv.rules /etc/udev/rules.d/
[sudo] password for bruger:     
cp: cannot stat 'udev/50-rogdrv.rules': No such file or directory

What am I doing wrong?

kyokenn, 2020-11-16:
You need to run "cp" command from the directory where you did cloned/downloaded this repository. Make sure that you have "udev" directory and file "50-rogdrv.rules" inside it.

RyabykinIlya, 2020-12-31:
@kyokenn i have the same issue:
`Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 8, in <module>
    sys.exit(rogdrv())
  File "/usr/local/lib/python3.8/dist-packages/rogdrv/__main__.py", line 41, in rogdrv
    device = get_device()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv/device.py", line 79, in get_device
    return device_class()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv/device.py", line 148, in __init__
    self._ctl = hid.open_device(control)
  File "/usr/local/lib/python3.8/dist-packages/rogdrv/hid.py", line 44, in open_device
    device.open_path(info['path'])
  File "hid.pyx", line 74, in hid.device.open_path
OSError: open failed`

but i copied 50-rogdrv.rules successfully
i can debug by myself, but may be you have some quick advice?

kyokenn, 2021-01-11:
> but i copied 50-rogdrv.rules successfully
> i can debug by myself, but may be you have some quick advice?

Check permissions of your mouse HID devices files:
`
ls -lah /dev/hidraw*
`

They should be rw-rw-rw

closed 2021-07-01 by kyokenn