kyokenn/rogdrv #15: Chakram / Chakram Core Support

nimr0d, 2020-11-28:
Is there intention to support these mice?

kyokenn, 2020-11-28:
I don't have this mouse, so it's impossible for me. Someone should use an USB debugger software together with ASUS Armoury and dump USB packets from this mouse.

nimr0d, 2020-12-01:
I can do that if you tell me how.

kyokenn, 2020-12-01:
I need to install some USB debugger/sniffer software, for example WireShark https://www.wireshark.org/
If you don't see USB capturing feature, then you need to install some extra stuff from https://wiki.wireshark.org/CaptureSetup/USB

Then you need to do something in Armoury, like switching profiles and find the related USB packets (request and response) in WireShark which does that. There will be a lot of packets, which aren't useful, like mouse movement events. This is how it looks like for my mouse (check the screenshots): https://github.com/kyokenn/rogdrv/issues/12#issuecomment-697171950

closed 2021-07-01 by kyokenn