kyokenn/rogdrv #19: Setting DPI on Gladius 2 Origin Pink actually sets it as double the asked DPI

emithethief, 2021-07-17:
After setting the DPI to 800 with rogdrv-config, I compared with ROG Armoury the DPI, and it reported as 1600, setting it as 800 on ROG Armoury will make it report as 400 by rogdrv-config, etc. Also, the ROG Armoury reported DPI seems to be the actual correct one, comparing to another mouse (not Asus) I have which is on 800 DPI.
Sorry for reporting about a device not yet supported. Product ID just in case: 18CD

kyokenn, 2021-07-18:
Does it affects the both DPI presets?
(all Gladius II -based mices have two DPI presets)

>Sorry for reporting about a device not yet supported.

It's fine. Actually the new device would never be supported unless someone reports about it.

emithethief, 2021-07-18:
I confirm both DPI presets are affected. Also it might affect all Gladius II Origin and not only the pink one, checking with an owner of a base Origin one could help.
Here is some demonstrations if you need:

`emilie@pop-os:~$ rogdrv-config dpi
DPI Preset 1: 400
DPI Preset 2: 400

# Setting both presets to different DPIs
emilie@pop-os:~$ rogdrv-config dpi 800 2
DPI Preset 1: 400
DPI Preset 2: 800
emilie@pop-os:~$ rogdrv-config dpi 200 1
DPI Preset 1: 200
DPI Preset 2: 800

# Preset 1 reports as 400 DPI in ROG Armoury
# Preset 2 reports as 1600 DPI in ROG Armoury

# After changing DPI from ROG Armoury on Windows to 800 on both profiles
emilie@pop-os:~$ rogdrv-config dpi
DPI Preset 1: 400
DPI Preset 2: 400
`

kyokenn, 2021-07-18:
> Here is some demonstrations if you need:

Can you run all the same with "--debug" option like "rogdrv-config dpi --debug" ? You can skip the device searching output.

kyokenn, 2021-07-18:
>Also it might affect all Gladius II Origin and not only the pink one

Yeah. I doubt the Pink one have some different hardware than generic Origin.

emithethief, 2021-07-18:
> Can you run all the same with "--debug"

Hope that's what you were looking for but here it is

```bash
emilie@pop-os:~$ rogdrv-config dpi --debug 200 1
[2021-07-18 10:26:12,484] [DEBUG] setting DPI to 200 for preset 1
[2021-07-18 10:26:12,484] [DEBUG] > 51 31 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:12,486] [DEBUG] < 51 31 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:12,486] [DEBUG] saving setting
[2021-07-18 10:26:12,486] [DEBUG] > 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:12,505] [DEBUG] < 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:12,505] [DEBUG] getting DPI and polling rate
[2021-07-18 10:26:12,505] [DEBUG] Number of dpi presets: 2
[2021-07-18 10:26:12,505] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:12,507] [DEBUG] < 12 04 00 00 03 00 07 00 03 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:12,507] [DEBUG] dpi_offset = 8
DPI Preset 1: 200
DPI Preset 2: 400

emilie@pop-os:~$ rogdrv-config dpi --debug 800 2
[2021-07-18 10:26:17,696] [DEBUG] setting DPI to 800 for preset 2
[2021-07-18 10:26:17,696] [DEBUG] > 51 31 01 00 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:17,697] [DEBUG] < 51 31 01 00 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:17,697] [DEBUG] saving setting
[2021-07-18 10:26:17,697] [DEBUG] > 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:17,716] [DEBUG] < 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:17,716] [DEBUG] getting DPI and polling rate
[2021-07-18 10:26:17,716] [DEBUG] Number of dpi presets: 2
[2021-07-18 10:26:17,716] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:17,718] [DEBUG] < 12 04 00 00 03 00 0F 00 03 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2021-07-18 10:26:17,718] [DEBUG] dpi_offset = 8
DPI Preset 1: 200
DPI Preset 2: 800
```

kyokenn, 2021-07-18:
I have added Origin Pink and patched DPI settings. You can test it and make sure that you have selected the same profile when comparing the settings with the Armoury.

emithethief, 2021-07-18:
The latest commits makes it impossible to access any setting of the mouse as far as I can tell (can't access dpi settings, polling rate, etc). Perhaps I am missing a dependency ?

```
emilie@pop-os:~/AppImages/rogdrv-master$ rogdrv-config --debug dpi
[2021-07-18 22:49:04,195] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2021-07-18 22:49:04,197] [DEBUG] 0 devices found
[2021-07-18 22:49:04,197] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2021-07-18 22:49:04,197] [DEBUG] 0 devices found
[2021-07-18 22:49:04,198] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2021-07-18 22:49:04,199] [DEBUG] found 5 subdevices:
[2021-07-18 22:49:04,199] [DEBUG] /dev/hidraw1: ASUSTeK ROG Gladius II Origin PNK LTD  interface 0
[2021-07-18 22:49:04,199] [DEBUG] /dev/hidraw1: ASUSTeK ROG Gladius II Origin PNK LTD  interface 0
[2021-07-18 22:49:04,199] [DEBUG] /dev/hidraw2: ASUSTeK ROG Gladius II Origin PNK LTD  interface 1 [using as keyboard]
[2021-07-18 22:49:04,199] [DEBUG] /dev/hidraw2: ASUSTeK ROG Gladius II Origin PNK LTD  interface 1 [using as keyboard]
[2021-07-18 22:49:04,199] [DEBUG] /dev/hidraw3: ASUSTeK ROG Gladius II Origin PNK LTD  interface 2 [using as control]
[2021-07-18 22:49:04,199] [DEBUG] opening control subdevice
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 8, in <module>
    sys.exit(rogdrv_config())
  File "/usr/local/lib/python3.9/dist-packages/rog/__main__.py", line 202, in rogdrv_config
    device = get_device()
  File "/usr/local/lib/python3.9/dist-packages/rog/device.py", line 77, in get_device
    return device_class()
  File "/usr/local/lib/python3.9/dist-packages/rog/device.py", line 141, in __init__
    self._ctl.open()
  File "/usr/local/lib/python3.9/dist-packages/rog/hid.py", line 49, in open
    self._device = hid.device()
AttributeError: module 'hid' has no attribute 'device'
```

kyokenn, 2021-07-18:
Made some HID API related changes. The driver should be compatible with both python3-hid and python3-hidapi. You can try to install one the them and remove the other one. Which Linux distribution and version are you using?

emithethief, 2021-07-19:
I'm using Pop!_OS 21.0 (Ubuntu-based). I had both python3-hid and python3-hidapi installed, tried running the latest source code with both installed, then one or the other installed.

**NOTE: Skip both two command outputs as they are most likely NOT the actual issue**
Doing `rogdrv-config dpi`
With python3-hid, I had the following output:
```
Traceback (most recent call last):
  File "/home/emilie/AppImages/rogdrv-master/./rogdrv-config", line 23, in <module>
    rogdrv_config()
  File "/home/emilie/AppImages/rogdrv-master/rog/__main__.py", line 202, in rogdrv_config
    device = get_device()
  File "/home/emilie/AppImages/rogdrv-master/rog/device.py", line 77, in get_device
    return device_class()
  File "/home/emilie/AppImages/rogdrv-master/rog/device.py", line 141, in __init__
    self._ctl.open()
  File "/home/emilie/AppImages/rogdrv-master/rog/hid.py", line 50, in open
    self._device = hid.device()
AttributeError: module 'hid' has no attribute 'device'
```

With python3-hidapi, I had the following output:
```
Traceback (most recent call last):
  File "/home/emilie/AppImages/rogdrv-master/./rogdrv-config", line 23, in <module>
    rogdrv_config()
  File "/home/emilie/AppImages/rogdrv-master/rog/__main__.py", line 202, in rogdrv_config
    device = get_device()
  File "/home/emilie/AppImages/rogdrv-master/rog/device.py", line 77, in get_device
    return device_class()
  File "/home/emilie/AppImages/rogdrv-master/rog/device.py", line 141, in __init__
    self._ctl.open()
  File "/home/emilie/AppImages/rogdrv-master/rog/hid.py", line 64, in open
    self._device = hidapi.Device(path=self['path'])
  File "/usr/lib/python3/dist-packages/hidapi.py", line 159, in __init__
    raise IOError("Could not open connection to device.")
OSError: Could not open connection to device.
```
With both installed it defaults to hidapi and gives the same output thereof.

**install_udev issues**
Also, running `sudo udevadm trigger` after the install_udev script now sometimes outputs to me `Failed to write 'change' to '/sys/devices/virtual/input/input24/uevent': Cannot allocate memory`
I was suspecting issues with the install_udev script or the udevadm commands to run ~~but after rebooting it seems to actually work, it didn't before rebooting and would fail to find the devices for some reasons according to rogdrv-config using `--debug`. To be honest I am not sure what caused this issue but right now it seems to be fixed and working~~. Also if the udev rules are not installed yet, this issue doesn't seem to happen. **I think you should make sure that the udev rules file doesn't actually exist before writing to it, and if it does, remove it before installing a new one**, because I think this causes issues, it should be working after that.

**Actual issue**
The actual issue might have been in finding the device and not actually in the hid/hidapi calls. This might be related to the udev rules issues I reported above.
Right now `rogdrv-config dpi` reports the correct DPI and sets it correctly, on both presets, it matches what ROG Armoury reports.

In the current state **things seems to be working**. To ensure things are actually working I completely uninstalled everything including the udev rules, rebooted then redid the whole install process and it seems to be working.

The only issue I can find right now is that when the mouse has just been plugged-in, if you try to access DPI settings (and maybe probably other settings then), it reports DPIs of 100 in the first seconds of being plugged in, then correctly reports the DPI. This isn't really a big issue but it might be confusing at first.

emithethief, 2021-07-19:
New report: the device seems to be found for some time, then it stops being found and outputs the same things I just sent... Unplugging and re-plugging it fixes it, but I guess it will still get "lost" at some point again.

kyokenn, 2021-07-19:
> AttributeError: module 'hid' has no attribute 'device'

I have figured this out. There are acturally 3 HID-related packages. 2 of them are called "hid" which is confusing. I made the driver compatible with all 3 packages: python3-hid, python3-hidapi, hid (from PyPi).

I have removed "hid" (from PyPi) from requirements so it should never be installed with driver along. I suggest to uninstall it if you still have it installed with "sudo pip3 uninstall hid".

>OSError: Could not open connection to device.

This one could be permission problem if you don't have any problems running it with sudo. That's why I'm installing udev rules, so you don't need sudo.

> Failed to write 'change' to '/sys/devices/virtual/input/input24/uevent': Cannot allocate memory

Never had this one.

> I think you should make sure that the udev rules file doesn't actually exist before writing to it, and if it does, remove it before installing a new one

Ok. I'll try it.

>Right now rogdrv-config dpi reports the correct DPI and sets it correctly, on both presets, it matches what ROG Armoury reports.

Good.

>The only issue I can find right now is that when the mouse has just been plugged-in, if you try to access DPI settings (and maybe probably other settings then), it reports DPIs of 100 in the first seconds of being plugged in, then correctly reports the DPI.

Yeah, it's a known issue. It's a kinda rare one.

>New report: the device seems to be found for some time, then it stops being found and outputs the same things I just sent... Unplugging and re-plugging it fixes it, but I guess it will still get "lost" at some point again.

Never had this. Can you check the number of /dev/hidraw* files (before and after) and "dmesg" output when device is lost?

emithethief, 2021-07-21:
After using it for a few days, the device doesn't seem to get "lost" anymore (I think it was related to that hid package from PyPi that is now removed). The only issue that seems to appear is that everytime I poll for the DPI for the first time after the mouse gets initiated or stuff (like after booting, plugging it in, etc), it reports both DPIs as 100, only for the first poll. This issue seems to also affect response time, polling rate and snapping, I haven't tried the other customisable options as I do not use them, but I guess every first poll after the device's initialisation will return the lowest/first value possible for it. It seems to only affect get calls, as setting the DPI at a target will actually set it correctly even though it reports as 100 if it's the very first get call.
> Yeah, it's a known issue. It's a kinda rare one.
Not so rare actually.
Apart from that I think I can assume this device is supported with configurations for DPI, response time, polling rate and angle snapping. Other options including lighting and button bindings were not tested.

closed 2021-10-31 by kyokenn