kyokenn/rogdrv #22: Seemingly random device errors?

slacking247, 2021-11-03:
Likes to break occasionally on Zorin OS 16 (Ubuntu 20.04-based) with ROG Pugio. Tray menu settings stop working, can't quit with the "Quit" button. Killing the process and rebooting seems to fix the issue but perhaps there is a more elegant way to do this or to avoid this entirely?

Restarting the driver through terminal without rebooting doesn't fix the issue, the process gets stuck in the terminal and closing the window stops the driver with it.
```Traceback (most recent call last):
File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/ui/handler.py", line 106, in on_dpi_choice
dpis, _, _, _ = self._device.get_dpi_rate_response_snapping()
File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/device/base.py", line 425, in get_dpi_rate_response_snapping
response = self.query(bytes(request))
File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/device/base.py", line 213, in query
raise DeviceError()
rog.device.base.DeviceError
Traceback (most recent call last):
File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/ui/handler.py", line 82, in on_profile_choice
profile, _, _ = self._device.get_profile_version()
File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/device/base.py", line 374, in get_profile_version
 response = self.query(bytes(request))
File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/device/base.py", line 213, in query
raise DeviceError()
rog.device.base.DeviceError
```

closed 2021-12-14 by slacking247

kyokenn, 2021-12-14:
I think it's legit. Why did you closed it? The mouse could go into sleep or somehow got into invalid state, but the program is not reopening the device and just crashing. It was kind of bad design from the beginning, so it would require a huge amount of work.

slacking247, 2021-12-14:
Oh. I did not intend to do this.

I think I did open the issue too soon and without my own research, I still don't know what exactly causes it (I still suspect dual-booting but I honestly don't know, it doesn't happen all the time) 
I also have not encountered the issue for at least 3 weeks now and just kind of assumed it went away after a system update of some kind.