kyokenn/rogdrv #23: Python 2.7 only

ThomasSeeker, 2021-11-09:
Hello, 

I would like to buy the ROG Gladius II mouse and use your driver.
But at the moment I still use an old Linux with python 2.7 only.

Until I upgrade my system and use your driver will my Linux at least recognize all the 6 buttons?
And will there be a way to switch off the LED's completely (maybe via plugging into Windows briefly)?

I would be grateful, if you could help me with these 2 questions.

Best greetings from Germany!

kyokenn, 2021-11-10:
> Until I upgrade my system and use your driver will my Linux at least recognize all the 6 buttons?

If you bind generic "back" and "forward" events on mouse buttons or just keep defaults they would work. They will be detected as "mouse4" and "mouse5" in games.
Setting-related events like profile and DPI switching would work too, because they are handled by the firmware inside the mouse.
But if you bind keyboard events on mouse buttons they would not work without a driver.

> And will there be a way to switch off the LED's completely (maybe via plugging into Windows briefly)?

Yes. You can boot into windows (even on virtual machine) and change the settings with ROG Armory. All the mouse settings are saved into mouse flash memory.

ThomasSeeker, 2021-11-10:
Dear Kyoken, thank you very much!
This helps a lot!

:-)

closed 2022-09-01 by kyokenn