kyokenn/rogdrv #24: Strix Evolve support

Peteris-Zvejnieks, 2021-11-15:
Just bought Strix Evolve hoping that it would turn out to be supported, unless I messed up the install, it is not.

Tried both of my USB 3.0 ports, the mouse can not be found.

Attempting to launch:
`$ rogdrv`
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/zvejni95/Downloads/rogdrv-master/rog/__main__.py", line 65, in run
    e = self._device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 33, in <module>
    sys.exit(load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv')())
  File "/home/zvejni95/Downloads/rogdrv-master/rog/__main__.py", line 393, in rogdrv
    gtk3_main(app.device)
  File "/home/zvejni95/Downloads/rogdrv-master/rog/ui/__init__.py", line 49, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'
```

Debug:
`$ rogdrv --debug`
```
[2021-11-15 16:01:09,338] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2021-11-15 16:01:09,380] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,382] [DEBUG] 0 devices found
[2021-11-15 16:01:09,383] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2021-11-15 16:01:09,383] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,384] [DEBUG] 0 devices found
[2021-11-15 16:01:09,384] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2021-11-15 16:01:09,384] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,386] [DEBUG] 0 devices found
[2021-11-15 16:01:09,386] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2021-11-15 16:01:09,386] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,387] [DEBUG] 0 devices found
[2021-11-15 16:01:09,387] [DEBUG] searching for device PugioGladiusII (VendorID: 0x0B05, ProductID: 0x1851)
[2021-11-15 16:01:09,387] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,388] [DEBUG] 0 devices found
[2021-11-15 16:01:09,388] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2021-11-15 16:01:09,388] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,389] [DEBUG] 0 devices found
[2021-11-15 16:01:09,389] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2021-11-15 16:01:09,389] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,389] [DEBUG] 0 devices found
[2021-11-15 16:01:09,389] [DEBUG] searching for device StrixImpactII (VendorID: 0x0B05, ProductID: 0x1947)
[2021-11-15 16:01:09,390] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,390] [DEBUG] 0 devices found
[2021-11-15 16:01:09,390] [DEBUG] searching for device StrixImpactIIWireless (VendorID: 0x0B05, ProductID: 0x1949)
[2021-11-15 16:01:09,390] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,391] [DEBUG] 0 devices found
[2021-11-15 16:01:09,391] [DEBUG] searching for device StrixEvolve (VendorID: 0x0B05, ProductID: 0x185B)
[2021-11-15 16:01:09,391] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,392] [DEBUG] found 2 subdevices:
[2021-11-15 16:01:09,392] [DEBUG] /dev/hidraw2: ASUS Evolve interface 0
[2021-11-15 16:01:09,392] [DEBUG] /dev/hidraw3: ASUS Evolve interface 1 [using as keyboard]
[2021-11-15 16:01:09,393] [DEBUG] control subdevice not found
[2021-11-15 16:01:09,393] [DEBUG] searching for device Spatha (VendorID: 0x0B05, ProductID: 0x181C)
[2021-11-15 16:01:09,393] [DEBUG] getting list of devicesdetected using "hidapi-cffi"
[2021-11-15 16:01:09,393] [DEBUG] 0 devices found
[2021-11-15 16:01:09,393] [DEBUG] searching for device SpathaWireless (VendorID: 0x0B05, ProductID: 0x1824)
[2021-11-15 16:01:09,393] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,394] [DEBUG] 0 devices found
[2021-11-15 16:01:09,394] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2021-11-15 16:01:09,394] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,395] [DEBUG] 0 devices found
[2021-11-15 16:01:09,395] [DEBUG] searching for device KerisWireless (VendorID: 0x0B05, ProductID: 0x1960)
[2021-11-15 16:01:09,395] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,396] [DEBUG] 0 devices found
[2021-11-15 16:01:09,396] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x195E)
[2021-11-15 16:01:09,396] [DEBUG] getting list of devices using "hidapi-cffi"
[2021-11-15 16:01:09,397] [DEBUG] 0 devices found
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/zvejni95/Downloads/rogdrv-master/rog/__main__.py", line 65, in run
    e = self._device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 33, in <module>
    sys.exit(load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv')())
  File "/home/zvejni95/Downloads/rogdrv-master/rog/__main__.py", line 393, in rogdrv
    gtk3_main(app.device)
  File "/home/zvejni95/Downloads/rogdrv-master/rog/ui/__init__.py", line 49, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'
```

kyokenn, 2021-11-15:
What is your kernel version?
Can you show me "dmesg" output after connecting the mouse to the USB?

Peteris-Zvejnieks, 2021-11-15:
[dmesg.log](https://github.com/kyokenn/rogdrv/files/7539801/dmesg.log)

kyokenn, 2021-11-15:
So you got mouse interface detected by "hid-generic" driver:
```
[    2.153897] hid-generic 0003:0B05:185B.0003: input,hidraw2: USB HID v1.11 Mouse [ASUS Evolve] on usb-0000:00:14.0-8/input0
```
and also keyboard interface:
```
[    2.214695] hid-generic 0003:0B05:185B.0004: input,hiddev1,hidraw3: USB HID v1.11 Keyboard [ASUS Evolve] on usb-0000:00:14.0-8/input1
```
But "ASUS Evolve Consumer Control" is not detected by "hid-generic".

Can you get output of:
```LIBUSB_DEBUG=9 sudo -E lsusb -v -d 0b05:185b```

Peteris-Zvejnieks, 2021-11-15:
[lsusb.log](https://github.com/kyokenn/rogdrv/files/7540813/lsusb.log)

kyokenn, 2021-11-15:
So descriptor have only mouse and keyboard interfaces and no consumer control. It's the same as in Spatha and old Gladius. Looks like the all old ASUS mice from Gladius generation have this problem. They are incompatible with linux HID drivers.

Unfortunately I can't help you with this mouse. It may need some heavy changes to the linux kernel, but I'm not a kernel developer.

Thanks for the report. I'm going to make a list of mice which are confirmed as incompatible.

Peteris-Zvejnieks, 2021-11-15:
That is unfortunate! 
Either way, thanks for looking in to it so quickly. This was a very pleasant experience!

closed 2022-09-01 by kyokenn