kyokenn/rogdrv #26: Help for Strix Carry

Graval504, 2021-12-03:
I recently installled majaro linux on my notebook with using my new mouse,
ROG Strix Carry with mouse reciever.( so i am not an expert on manjaro linux, both with ubuntu)
to use rogdrv, i installed python,python-evdev,python-hid,cython-hidapi with yay.
(actually cython-hidapi is installed after the last command of this text)
i tried to find gir-appindicator3 on yay,
but i couldn't find them.(only found something different appindicator and just python-hid.)
actually i can't remember about the appindicator i installed :( 

on this situation, i ran rogdrv on terminal, and got
[graval504@graval504-14u530gh50k ~]$ rogdrv-config dpi --debug
[2021-12-03 11:28:16,663] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2021-12-03 11:28:16,663] [DEBUG] 0 devices found
[2021-12-03 11:28:16,663] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2021-12-03 11:28:16,663] [DEBUG] 0 devices found
[2021-12-03 11:28:16,663] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2021-12-03 11:28:16,663] [DEBUG] 0 devices found
[2021-12-03 11:28:16,663] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2021-12-03 11:28:16,664] [DEBUG] 0 devices found
[2021-12-03 11:28:16,664] [DEBUG] searching for device PugioGladiusII (VendorID: 0x0B05, ProductID: 0x1851)
[2021-12-03 11:28:16,664] [DEBUG] 0 devices found
[2021-12-03 11:28:16,664] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2021-12-03 11:28:16,664] [DEBUG] 0 devices found
[2021-12-03 11:28:16,664] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2021-12-03 11:28:16,665] [DEBUG] 0 devices found
[2021-12-03 11:28:16,665] [DEBUG] searching for device StrixImpactIIWirelessWired (VendorID: 0x0B05, ProductID: 0x1947)
[2021-12-03 11:28:16,665] [DEBUG] 0 devices found
[2021-12-03 11:28:16,665] [DEBUG] searching for device StrixImpactIIWireless (VendorID: 0x0B05, ProductID: 0x1949)
[2021-12-03 11:28:16,665] [DEBUG] 0 devices found
[2021-12-03 11:28:16,665] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2021-12-03 11:28:16,665] [DEBUG] 0 devices found
[2021-12-03 11:28:16,666] [DEBUG] searching for device KerisWireless (VendorID: 0x0B05, ProductID: 0x1960)
[2021-12-03 11:28:16,666] [DEBUG] 0 devices found
[2021-12-03 11:28:16,666] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x195E)
[2021-12-03 11:28:16,666] [DEBUG] 0 devices found
Device not found

(actually i losted log about running rogdrv sry)
and tried LIBUSB_DEBUG=9 sudo -E lsusb -v -d 0b05:0x18B4

[graval504@graval504-14u530gh50k ~]$ LIBUSB_DEBUG=9 sudo -E lsusb -v -d 0b05:0x18B4            
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000024] [00001d35] libusb: debug [libusb_init] created default context
[ 0.000040] [00001d35] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000052] [00001d35] libusb: debug [get_kernel_version] reported kernel version is 5.13.19-2-MANJARO
[ 0.000102] [00001d35] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000110] [00001d35] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000118] [00001d35] libusb: debug [op_init] sysfs is available
[ 0.000363] [00001d36] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.005057] [00001d35] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.005066] [00001d35] libusb: debug [linux_get_device_address] scan usb2
[ 0.005102] [00001d35] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.005109] [00001d35] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.005114] [00001d35] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.005247] [00001d35] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.005254] [00001d35] libusb: debug [linux_get_device_address] scan 2-2
[ 0.005298] [00001d35] libusb: debug [linux_get_device_address] bus=2 dev=20
[ 0.005304] [00001d35] libusb: debug [linux_enumerate_device] busnum 2 devaddr 20 session_id 532
[ 0.005319] [00001d35] libusb: debug [linux_enumerate_device] allocating new device for 2/20 (session 532)
[ 0.005361] [00001d35] libusb: debug [linux_get_parent_info] dev 0x561f1ad46df0 (2-2) has parent 0x561f1ad5e380 (usb2) port 2
[ 0.005472] [00001d35] libusb: debug [linux_get_device_address] getting address for device: 2-3 detached: 0
[ 0.005478] [00001d35] libusb: debug [linux_get_device_address] scan 2-3
[ 0.005512] [00001d35] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.005519] [00001d35] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.005524] [00001d35] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.005567] [00001d35] libusb: debug [linux_get_parent_info] dev 0x561f1ad46fb0 (2-3) has parent 0x561f1ad5e380 (usb2) port 3
[ 0.005661] [00001d35] libusb: debug [linux_get_device_address] getting address for device: 2-4 detached: 0
[ 0.005667] [00001d35] libusb: debug [linux_get_device_address] scan 2-4
[ 0.005703] [00001d35] libusb: debug [linux_get_device_address] bus=2 dev=3
[ 0.005709] [00001d35] libusb: debug [linux_enumerate_device] busnum 2 devaddr 3 session_id 515
[... 423 lines cut ...]
and ran again rogdrv,

[graval504@graval504-14u530gh50k ~]$ rogdrv
Traceback (most recent call last):
  File "/usr/bin/rogdrv", line 33, in <module>
    sys.exit(load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv')())
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 385, in rogdrv
    app = ROGDRV()
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 44, in __init__
    self._device = get_device()
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 47, in get_device
    return device_class()
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 111, in __init__
    self._ctl.open()
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/hid.py", line 79, in open
    self._device = hid.Device(path=self['path'])
  File "/usr/lib/python3.9/site-packages/hid/__init__.py", line 130, in __init__
    raise HIDException('unable to open device')
hid.HIDException: unable to open device

errorcode changed like that.

after i installed cython-hidapi and ran rogdrv again, but get the same message.
i don't know about how to solve this situation.
i think it can be very complicated because i ran many command, installed or removed or changed packages, etc on terminal over two days;

kyokenn, 2021-12-03:
Check the permissions on your hidraw devices:
```
$ ls -l /dev/hidraw*
```

You can change the permissions as a short solution:
```
$ sudo chmod 666 /dev/hidraw*
```

If it helps you can install udev rules - https://github.com/kyokenn/rogdrv#installation

Graval504, 2021-12-03:
tried it but not effective..

`[graval504@graval504-14u530gh50k ~]$ ls -l /dev/hidraw*
ls: cannot access '/dev/hidraw*': No such file or directory
[graval504@graval504-14u530gh50k ~]$ sudo ls -l /dev/hidraw*
[sudo] password for graval504: 
ls: cannot access '/dev/hidraw*': No such file or directory
[graval504@graval504-14u530gh50k ~]$ sudo chmod 666 /dev/hidraw*
chmod: cannot access '/dev/hidraw*': No such file or directory
`
`[graval504@graval504-14u530gh50k rogdrv]$ sudo udevadm control --reload-rules
[graval504@graval504-14u530gh50k rogdrv]$ sudo udevadm trigger
[graval504@graval504-14u530gh50k rogdrv]$ rogdrv
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 65, in run
    e = self._device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'
Traceback (most recent call last):
  File "/usr/bin/rogdrv", line 33, in <module>
    sys.exit(load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv')())
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 393, in rogdrv
    gtk3_main(app.device)
  File "/usr/lib/python3.9/site-packages/rogdrv-0.3.0-py3.9.egg/rog/ui/__init__.py", line 49, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'
`

kyokenn, 2021-12-03:
`chmod: cannot access '/dev/hidraw*': No such file or directory`
This is wierd. You are missing hidraw device files. 

Can you show output of `dmesg` after you plugged in your mouse?

closed 2021-12-03 by Graval504