kyokenn/rogdrv #28: ROG Chakram

OdinVex, 2021-12-08:
I have a `ROG Chakram` (non-`Core`), I would like to help:

Hardware Specs:

Resolution: `100~16000 DPI`
USB Polling: `1000 Hz`
Aura Sync: `Yes`
Joystick: Disabled via `udev`. It kept showing up as a controller…
Profiles: `3`
DPI Presets: `4`
DPI-On-Scroll: `Yes`
Wireless Methods: `RF` or `Bluetooth`
USB IDs: `0b05:18e3 (Wireless Wired)` `0b05:18e5 (Wireless, RF)` * I couldn't get Bluetooth to connect. Everyone knows Linux sucks donkey **** when it comes to Bluetooth. I'll guess `0b05:18e4`.

ASUS has some…weird “gaming profiles” in the manual. Not going to bother mentioning them here.

Added to udev rules, reloaded and triggered, added classes to `rog.rpy` using `Keris` as a template. 

So far I have:
`DPI Preset 1 (red): 400`
`DPI Preset 2 (purple): 800`
`DPI Preset 3 (blue): 1600` * Default
`DPI Preset 4 (green): 3200`
`1 logo | default | FF0000 | 4`
`2 wheel | default | FF0000 | 4`
`3 bottom | default | FF0000 | 4`
`Polling rate: 1000 Hz`

I'm not sure about `buttons`, `keyboard_interface`, or `control_interface`.
For buttons, I can attest only to `MBL`, `MBR`, `Scroll`, `Forward`, `Back`, `Pairing`, `DPI`.

`DPI+Forward` triggers `Profile 1` with “mouse and 4 keyboard direction keys (using analogue stick)” and `lighting effect: wave`.
`DPI+Back` triggers `Profile 2` with “mouse and analogue stick (direction arrows)” and `lighting effect: cycle`.
`DPI+Scroll` triggers `Profile 2` with “gamepad” and `lighting effect: comet (blue)`.

No idea if it needs `DoubleDPIMixin`. The Analogue Stick can act with keyboard events, guessing it might need `BitmaskMixin`?

Actions works.
LED-g/setting works.
Rate-g/setting works.
Response-g/setting works.
Sleep may be working. It showed it was set to sleep at `10 min.` but I thought I had that turned off. Other than that, it works.
Angle-g/setting works.
Version-getting works.

I had no interest in bindings. Given how lucky everything is, I'll assume it works. I'll try and create a pull-request.

kyokenn, 2021-12-08:
>No idea if it needs DoubleDPIMixin.

You can check it by comparing DPI values with ROG Armoury. You can run in from Windows VM
Most last-generation mice need it.

>DPI Preset 1 (red): 400

I believe it should be 800 by default, but I don't remember the defaults.

OdinVex, 2021-12-08:
> > No idea if it needs DoubleDPIMixin.
> 
> You can check it by comparing DPI values with ROG Armoury. You can run in from Windows VM Most last-generation mice need it.
> 
> > DPI Preset 1 (red): 400
> 
> I believe it should be 800 by default, but I don't remember the defaults.

I haven't run a Windows VM in a long time, but I can confirm with the manual and my own memory that 400 is the first profile. `rogdrv` has successfully queried the device information, entirely.

kyokenn, 2021-12-08:
> I haven't run a Windows VM in a long time, but I can confirm with the manual and my own memory that 400 is the first profile. rogdrv has successfully queried the device information, entirely.

I see. Thanks for the confirmation.

>The Analogue Stick can act with keyboard events, guessing it might need BitmaskMixin?

You can test the keyboard events with `rogdrv --debug`

OdinVex, 2021-12-08:
> I see. Thanks for the confirmation.
>
>
> You can test the keyboard events with `rogdrv --debug`

I wouldn't be able to test the `debug` feature at the moment, I would need to undo a few `udev` rules to re-enable the analogue joystick. I found most games thinking it was a gaming controller, and that just wouldn't work for me.

I added a `Pull Request` at #30 after cleaning up my first one from a horribly heavy-thumbed push.

closed 2021-12-08 by OdinVex

bam80, 2022-02-20:
@OdinVex could you check bluetooth mode?

OdinVex, 2022-02-20:
> @OdinVex could you check bluetooth mode?

Unfortunately, I've got a version of bluez that sucks. I won't be able to test it at this moment.

kyokenn, 2022-02-20:
@bam80 Why you keep spamming for every ASUS mouse model but you got a Lenovo one. ASUS uses it's own protocol, so it's completely unrelated. Most (could be all of them) ASUS mice work fine with bluetooth, but there are some hardware limitations that you can't change the mouse settings.

OdinVex, 2022-02-20:
> @bam80 Why you keep spamming for every ASUS mouse model but you got a Lenovo one. ASUS uses it's own protocol, so it's completely unrelated. Most (could be all of them) ASUS mice work fine with bluetooth, but there are some hardware limitations that you can't change the mouse settings.

I could change all of my settings over Bluetooth on Windows. *shrug* I deleted my Windows installation two years ago now, though.

bam80, 2022-02-20:
> @bam80 Why you keep spamming for every ASUS mouse model but you got a Lenovo one.

I'm considering to swap it to ASUS one just because of this problem, can you imagine it's reasonable?

> Most (could be all of them) ASUS mice work fine with bluetooth,

At last, some useful info.
I suppose you actually tried it? Because I have serious concerns about composite HID over GATT devices support in BlueZ..

> but there are some hardware limitations that you can't change the mouse settings.

As long as I can set them in Windows and they are saved to mouse, that's fine

bam80, 2022-02-20:
> Unfortunately, I've got a version of bluez that sucks. I won't be able to test it at this moment.

@OdinVex what do you mean by that, your current bluez version doesn't work?
I tried several BlueZ versions up to master, and still it doesn't work for my Lenovo mouse. Could it be the same problem?
https://www.spinics.net/lists/linux-bluetooth/msg96871.html

OdinVex, 2022-02-20:
> @OdinVex what do you mean by that, your current bluez version doesn't work? I tried several BlueZ versions up to master, and still it doesn't work for my Lenovo mouse. Could it be the same problem? [...]

It is unrelated to this thread Issue, but I'll summarize: Bluez breaks if you simply sneeze or forget say please while wearing garlic. Updates break things, to the point that bluez has particularly favored versions by its own authors/developers because of it.

OdinVex, 2022-02-21:
> @bam80 Why you keep spamming for every ASUS mouse model but you got a Lenovo one. ASUS uses it's own protocol, so it's completely unrelated. Most (could be all of them) ASUS mice work fine with bluetooth, but there are some hardware limitations that you can't change the mouse settings.

I might open a new issue for it but I did try to open rogdrv over bluetooth (not a complete connect!) and rogdrv crashed:
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/dist-packages/rog/__main__.py", line 65, in run
    e = self._device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 8, in <module>
    sys.exit(rogdrv())
  File "/usr/local/lib/python3.8/dist-packages/rog/__main__.py", line 393, in rogdrv
    gtk3_main(app.device)
  File "/usr/local/lib/python3.8/dist-packages/rog/ui/__init__.py", line 49, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'

It shouldn't have gotten that far if the device isn't fully connected, hrm.

bam80, 2022-02-21:
@OdinVex thanks, yes, please open an issue.
Does the mouse movements work at least even without rogdrv maybe?

bam80, 2022-02-21:
> I can use bluetoothctl to pair, connect, but it never “finishes” connecting on the mouse side of things. Bluez thinks it is, but it isn't.

How do you know it isn't?

OdinVex, 2022-02-21:
> > I can use bluetoothctl to pair, connect, but it never “finishes” connecting on the mouse side of things. Bluez thinks it is, but it isn't.
> 
> How do you know it isn't?

During Bluetooth pairing, the LEDs on the mouse will temporarily override color profiles and be blue and blink. They never stop, mouse input is not detected, but bluetoothctl says paired&connected&trusted.

bam80, 2022-02-21:
Hmm, my mouse seems to connects fine and stops blinking blue on pairing..
Do you have a chance to check newer Bluez? 5.53 seems really old, and BlueZ devs suggested to update it also.

OdinVex, 2022-02-21:
> Hmm, my mouse seems to connects fine and stops blinking blue on pairing.. Do you have a chance to check newer Bluez? 5.53 seems really old, and BlueZ devs suggested to update it also.

Of course, it's old, too many distros are based on Ubuntu, as is the one I'm using right now. If yours stops blinking, it should work fine. If rogdrv doesn't work for it (and you're using ASUS, not Lenovo!!!) then you can help debug the issue..in a new Issue.

bam80, 2022-02-21:
> if yours stops blinking, it should work fine. If rogdrv doesn't work for it (and you're using ASUS, not Lenovo!!!)

Mine is Lenovo, I just think about swapping it to ASUS. Hoped you get your one working before I get to the store..

> I'm not sure what you mean by mouse movements.

I thought without rogdrv mouse moves only but doesn't react to buttons presses.

Anyway, I'll go to the store with my laptop, will try the mouse with BT there, if it moves at least I'll buy it.
Hope we will sort out rest and I won't have to return it again.

Thanks.

OdinVex, 2022-02-21:
> Mine is Lenovo, I just think about swapping it to ASUS. Hoped you get your one working before I get to the store..
>
>
> I thought without rogdrv mouse moves only but doesn't react to buttons presses. [...]

You can just hook it up via wired or RF (aka ISM), set everything, then switch to Bluetooth.

bam80, 2022-02-21:
> By the way, rogdrv is not necessary for basic functionality of the mouse.

That probably true for Gen 2 devices only, see https://github.com/kyokenn/hid-asus-mouse#device-compatibility, only 2  devices listed with "no driver needed".
I'm aiming at Pugio 2 which is Gen 1 device I suppose.

OdinVex, 2022-02-21:
> > By the way, rogdrv is not necessary for basic functionality of the mouse.
> 
> That's probably true for Gen 2 devices only, see https://github.com/kyokenn/hid-asus-mouse#device-compatibility, only 2 devices listed with "no driver needed". I'm aiming at Pugio 2 which is Gen 1 device I suppose.

No, almost all mice have supported basic mouse functionality since 1995.

bam80, 2022-02-21:
How to interpret that table then? Most mice need [gen1 driver](https://github.com/kyokenn/hid-asus-mouse/blob/master/gen1) for Keyboard events according to it.

OdinVex, 2022-02-21:
> How to interpret that table then? Most mice need [gen1 driver](https://github.com/kyokenn/hid-asus-mouse/blob/master/gen1) for Keyboard events according to it.

Which has nothing to do with basic functionality. You're talking about extended functionality, such as setting LED colors, profiles, macros & mapping. I never bothered to test those functionalities because I don't use that. If yours needs it, you can submit a pull request after you add it to gen1 and test.

bam80, 2022-02-21:
I just need basic, I thought that Keyboard events are needed to report button presses for that mice. Maybe Readme could be improved then, it's not obvious.
Thanks for clarifying.

kyokenn, 2022-02-21:
>I'm considering to swap it to ASUS one just because of this problem, can you imagine it's reasonable?

Yes. That's a fair point, but you could open an issue and just ask.

>I suppose you actually tried it?

Yes. I have tested few mice over BT by myself and they work just fine without any extra software/drivers, but you have to switch to RF/Cable to change the settings with rogdrv/ratbag/armoury.

>As long as I can set them in Windows and they are saved to mouse, that's fine

Yes. The settings are stored in the mouse.

So you can get any ASUS mouse, newer - better. Just don't buy some old one which was released before Gladius 2, because most of them have a broken HID descriptor. It's still possible to fix, but you have to dive into the kernel development stuff.

kyokenn, 2022-02-21:
>How to interpret that table then? Most mice need [gen1 driver](https://github.com/kyokenn/hid-asus-mouse/blob/master/gen1) for Keyboard events according to it.

That's true. Mouse button events would work without any driver. But if you bind keyboard events you would need a HID driver. Also you don't need a HID driver in BT mode, but it could depends on the mouse.

OdinVex, 2022-02-21:
> I just need basic, I thought that Keyboard events are needed to report button presses for that mice. Maybe Readme could be improved then, it's not obvious. Thanks for clarifying.

LMB, RMB, MMB, Scroll, Back, Forward, are not keyboard presses. They're commands. The readme doesn't need an update on that, this is general common knowledge.

> Yes. I have tested few mice over BT by myself and they work just fine without any extra software/drivers, but you have to switch to RF/Cable to change the settings with rogdrv/ratbag/armoury.

And as soon as Bluez gets updated for my distro I'll test my Chakram out. I don't think it would need RF/USB, but I'm a hopeful guy. :)

> So you can get any ASUS mouse, newer - better. Just don't buy some old one which was released before Gladius 2, because most of them have a broken HID descriptor. It's still possible to fix, but you have to dive into the kernel development stuff.

A few years ago, I was able to create a mapped ID via grub cfg. I think it was for one of those fake 'Dr.' Dre Realtek chipsets which was only a rebadge with a ID change with biased presets to market his fake crap. It was on an HP laptop.

bam80, 2022-02-22:
Well, I bought ROG Pugio 2, it works with BT indeed, which is nice!
But there is a problem still - it always keeps BT connected which is not good for batteries I guess..