kyokenn/rogdrv #31: Tray Icon Image not Visible

OdinVex, 2021-12-08:
I'm running `Linux Mint 20.2 x64`. Tray Icon Image is not visible, but the entry is there.

I checked `/usr/local/share/pixmaps`, the icons 'rog-symbolic.symbolic.png' and 'rog.png' are there. `rogdrv` doesn't seem to be loaded `input-mouse-symbolic` or `input-mouse` either.

kyokenn, 2021-12-08:
Are you sure it's not just a black icon on a black background?

OdinVex, 2021-12-08:
> Are you sure it's not just a black icon on a black background?

Quite certain. I changed my theme, as I mentioned. It's just not showing the actual images in any of the menus. I even replaced the icons with simple solid-red pngs to no avail.

kyokenn, 2021-12-08:
> Edit: I found it interesting that if I set APP_INDICATOR_SUPPORT to false, I get an icon.

I see. So the new app indicator is broken for you, and you got working the old-style tray icon.

What desktop environment are you using? Is it generic gnome-shell?

OdinVex, 2021-12-08:
> > Edit: I found it interesting that if I set APP_INDICATOR_SUPPORT to false, I get an icon.
> 
> I see. So the new app indicator is broken for you, and you got working the old-style tray icon.
> 
> What desktop environment are you using? Is it generic gnome-shell?

Cinnamon.