kyokenn/rogdrv #33: Back with "/dev/uinput" cannot be opened for writing

slacking247, 2022-01-05:
``` $ rogdrv --debug
[2022-01-06 00:50:42,005] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-01-06 00:50:42,006] [DEBUG] getting list of devices using "cython-hidapi"
[2022-01-06 00:50:42,008] [DEBUG] 0 devices found
[2022-01-06 00:50:42,008] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-01-06 00:50:42,008] [DEBUG] getting list of devices using "cython-hidapi"
[2022-01-06 00:50:42,010] [DEBUG] 0 devices found
[2022-01-06 00:50:42,010] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-01-06 00:50:42,010] [DEBUG] getting list of devices using "cython-hidapi"
[2022-01-06 00:50:42,012] [DEBUG] 0 devices found
[2022-01-06 00:50:42,012] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2022-01-06 00:50:42,012] [DEBUG] getting list of devices using "cython-hidapi"
[2022-01-06 00:50:42,014] [DEBUG] found 5 subdevices:
[2022-01-06 00:50:42,014] [DEBUG] /dev/hidraw3: ASUS ROG PUGIO interface 0
[2022-01-06 00:50:42,014] [DEBUG] /dev/hidraw3: ASUS ROG PUGIO interface 0
[2022-01-06 00:50:42,014] [DEBUG] /dev/hidraw4: ASUS ROG PUGIO interface 1 [using as keyboard]
[2022-01-06 00:50:42,014] [DEBUG] /dev/hidraw4: ASUS ROG PUGIO interface 1 [using as keyboard]
[2022-01-06 00:50:42,014] [DEBUG] /dev/hidraw5: ASUS ROG PUGIO interface 2 [using as control]
[2022-01-06 00:50:42,014] [DEBUG] opening control subdevice
[2022-01-06 00:50:42,014] [DEBUG] opening device "/dev/hidraw5"
[2022-01-06 00:50:42,066] [DEBUG] opening keyboard subdevice
[2022-01-06 00:50:42,066] [DEBUG] opening device "/dev/hidraw4"
Traceback (most recent call last):
  File "/usr/bin/rogdrv", line 33, in <module>
    sys.exit(load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv')())
  File "/usr/lib/python3.10/site-packages/rogdrv-0.3.0-py3.10.egg/rog/__main__.py", line 385, in rogdrv
    app = ROGDRV()
  File "/usr/lib/python3.10/site-packages/rogdrv-0.3.0-py3.10.egg/rog/__main__.py", line 45, in __init__
    self._handler = DeviceEventHandler()
  File "/usr/lib/python3.10/site-packages/rogdrv-0.3.0-py3.10.egg/rog/event.py", line 22, in __init__
    self._uinput = uinput.UInput(events=event)
  File "/usr/lib/python3.10/site-packages/evdev/uinput.py", line 127, in __init__
    self._verify()
  File "/usr/lib/python3.10/site-packages/evdev/uinput.py", line 274, in _verify
    raise UInputError(msg.format(self.devnode))
evdev.uinput.UInputError: "/dev/uinput" cannot be opened for writing
```
Arch, udev rules installed, rebooting did not help.

kyokenn, 2022-01-12:
Wierd, udev rules have line ```KERNEL=="uinput", MODE="0666"```, so this should be enough.

What permissions you have on /dev/uinput ?
Can you show output of  ```stat /dev/uinput``` and ```getfacl /dev/uinput``` ?

slacking247, 2022-01-12:
That's weird, I did see that line there and yet:
```stat /dev/uinput
  File: /dev/uinput
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 0,5     Inode: 516         Links: 1     Device type: 10,223
Access: (0660/crw-rw----)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2022-01-12 10:20:20.314052722 +0200
Modify: 2022-01-12 10:20:20.314052722 +0200
Change: 2022-01-12 10:20:38.787052535 +0200
 Birth: -
```

```getfacl /dev/uinput
getfacl: Removing leading '/' from absolute path names
# file: dev/uinput
# owner: root
# group: root
user::rw-
user:kusr:rw-
group::---
mask::rw-
other::---
```

kyokenn, 2022-01-13:
You don't have any extra ACL's on file so it's just an udev-related issue.
I'm not familiar with arch linux, can you check that you don't have any other udev rules which could override this one?
Does the hidraw rules applied to /dev/hidraw* files?

Also as temporarily solution you can do ```chmod 666 /dev/uinput/```

slacking247, 2022-01-13:
/etc/udev/rules.d has only 50-rogdrv.rules in it, hidraw3 to hidraw5 have 660 permissions.

kyokenn, 2023-03-02:
uinput was removed in favor of kernel module - https://github.com/kyokenn/hid-asus-mouse

closed 2023-03-02 by kyokenn