kyokenn/rogdrv #34: Does it support ?

Duoslow, 2022-01-16:
I'm planning to buy ROG Pugio II, but does this driver support it, can anyone answer, thanks

kyokenn, 2022-01-16:
It's not supported atm, but if the ASUS hasn't changed much of the protocol and mouse doesn't have a broken descriptor (both are really rare cases), then it could be easily added. You just have to hope for the best.

Duoslow, 2022-01-17:
hi i bought the pugio 2 today and here is the command results 
0b05:1908 = https://pastebin.com/raw/mx2QYgZ8
0b05:1906 = https://pastebin.com/raw/3ep4xV8n

Duoslow, 2022-01-17:
I hope it's not broken 😃

kyokenn, 2022-01-17:
Looks good.
I have added device ID's to the driver. You can test it.
You also have to guess the button mappings by yourself.
It's using the Pugio's mappings atm, but Pugio had 10 buttons and Pugio 2 has 9 buttons.
Also some new mice may need DoubleDPIMixin because they have a high upper DPI limits.

bam80, 2022-02-18:
@Duoslow could you check if your mouse works in Bluetooth mode, with or without this driver?

closed 2022-03-15 by kyokenn

bam80, 2022-03-15:
> I'm planning to buy ROG Pugio II

I don't know if it's all game mouses property but mine Pugio 2 lasted in BT mode less than month on one charge, without any gaming and less then average usage.
It's considerably less than any other BT mouses I had.