kyokenn/rogdrv #35: Lenovo Legion M600 Wireless Gaming Mouse - doesn't work in Bluetooth mode

bam80, 2022-02-18:
Hi, many thanks for this project!

My question might be out of scope here still I hope for your help.
Just bought mentioned mouse, it works fine on Linux in USB cable and wireless adapter modes but in Bluetooth mode doesn't show any evidence of life after being successfully paired and connected.
`btmon` shows events, though - here is the dump in binary and text formats:
https://www.spinics.net/lists/linux-bluetooth/msg96904.html

Important note the mouse works fine with BT on my ancient tablet with Android 8.10. This mouse didn't even exist when it stopped updating..

Could you help me to understand what's wrong with this mouse on Linux?
I need to make a decision if I return it to store or keep it in hope it will be fixed soon..

kyokenn, 2022-02-18:
Some mice requires a specific version of BT, some of them requires BT LE. Make sure to check the mouse and BT adapter specs. You can also boot into Windows VM and pass BT adapter into VM.

bam80, 2022-02-18:
I have Windows 10 as a second system installed on the same PC. The mouse works there via BT out of the box, even before any additional Lenovo software installed.
Also, as seen in the logs above - in Linux in pairs and connects, and even events can be seen with `btmon`. It just doesn't work in BT mode, but works in RF an USB cable modes.

So I wonder does it use some custom HID protocol via BT, or the protocol is standard and the problem in something else, considering it also works on Android via BT?

bam80, 2022-03-15:
It's upstream problem with linux BT/HID stack, see the link above.

closed 2022-03-15 by bam80