kyokenn/rogdrv #36: OSError: Failed to write to HID device.

bam80, 2022-02-23:
The mouse connected via RF dongle:
```
$ lsusb 
Bus 002 Device 007: ID 0b05:1908 ASUSTek Computer, Inc. ROG PUGIO II
```
Loaded the custom udev rules as per Readme.

On Profiles menu unfold:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/ui/handler.py", line 82, in on_profile_choice
    profile, _, _ = self._device.get_profile_version()
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 375, in get_profile_version
    response = self.query(bytes(request))
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 200, in query
    self.write(request)
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 183, in write
    self._ctl.write(data)
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/hid.py", line 35, in write
    self._device.write(data)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 186, in write
    raise IOError("Failed to write to HID device.")
OSError: Failed to write to HID device.

kyokenn, 2022-02-28:
Can you show the permissions of hidraw files?
```ls -lah /dev/hidraw*```
You can figure out which files are related to mouse by unplugging it. You can also check ```dmesg```

bam80, 2022-03-05:
```
$  ls -lah /dev/hidraw*
crw------- 1 root root 239, 0 фев 24 22:23 /dev/hidraw0
crw------- 1 root root 239, 1 мар  5 22:20 /dev/hidraw1
```
hidraw1 corresponds to the mouse

bam80, 2022-03-05:
Oh, my bad - with the RF mode it's following:
```
$ ll /dev/hidraw*
crw-------  1 root root    239, 0 фев 24 22:23 /dev/hidraw0
crw-rw----+ 1 root plugdev 239, 1 мар  5 22:28 /dev/hidraw1
crw-rw----+ 1 root plugdev 239, 2 мар  5 22:28 /dev/hidraw2
crw-rw----+ 1 root plugdev 239, 3 мар  5 22:28 /dev/hidraw3
crw-rw----+ 1 root plugdev 239, 4 мар  5 22:28 /dev/hidraw4

kyokenn, 2022-03-10:
Permissions looks good. Does it work with sudo?
For example:
```
sudo rogdrv-config profile --debug
```

bam80, 2022-03-10:
It doesn't:
```
$ sudo rogdrv-config profile --debug
[sudo] password for bam: 
[2022-03-10 15:41:51,933] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-03-10 15:41:52,034] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,037] [DEBUG] 0 devices found
[2022-03-10 15:41:52,037] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-03-10 15:41:52,037] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,038] [DEBUG] 0 devices found
[2022-03-10 15:41:52,038] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-03-10 15:41:52,038] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,039] [DEBUG] 0 devices found
[2022-03-10 15:41:52,039] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2022-03-10 15:41:52,039] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,041] [DEBUG] 0 devices found
[2022-03-10 15:41:52,041] [DEBUG] searching for device PugioGladiusII (VendorID: 0x0B05, ProductID: 0x1851)
[2022-03-10 15:41:52,041] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,042] [DEBUG] 0 devices found
[2022-03-10 15:41:52,042] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2022-03-10 15:41:52,042] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,043] [DEBUG] 0 devices found
[2022-03-10 15:41:52,043] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2022-03-10 15:41:52,043] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,044] [DEBUG] 0 devices found
[2022-03-10 15:41:52,044] [DEBUG] searching for device StrixImpactIIWirelessWired (VendorID: 0x0B05, ProductID: 0x1947)
[2022-03-10 15:41:52,044] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,046] [DEBUG] 0 devices found
[2022-03-10 15:41:52,046] [DEBUG] searching for device StrixImpactIIWireless (VendorID: 0x0B05, ProductID: 0x1949)
[2022-03-10 15:41:52,046] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,047] [DEBUG] 0 devices found
[2022-03-10 15:41:52,047] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2022-03-10 15:41:52,047] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,049] [DEBUG] 0 devices found
[2022-03-10 15:41:52,049] [DEBUG] searching for device KerisWireless (VendorID: 0x0B05, ProductID: 0x1960)
[2022-03-10 15:41:52,049] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,051] [DEBUG] 0 devices found
[2022-03-10 15:41:52,051] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x195E)
[2022-03-10 15:41:52,051] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,052] [DEBUG] 0 devices found
[2022-03-10 15:41:52,052] [DEBUG] searching for device ChakramWireless (VendorID: 0x0B05, ProductID: 0x18E5)
[2022-03-10 15:41:52,053] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,055] [DEBUG] 0 devices found
[2022-03-10 15:41:52,055] [DEBUG] searching for device ChakramWirelessWired (VendorID: 0x0B05, ProductID: 0x18E3)
[2022-03-10 15:41:52,055] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,057] [DEBUG] 0 devices found
[2022-03-10 15:41:52,057] [DEBUG] searching for device Pugio2 (VendorID: 0x0B05, ProductID: 0x1906)
[2022-03-10 15:41:52,057] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,059] [DEBUG] 0 devices found
[2022-03-10 15:41:52,059] [DEBUG] searching for device Pugio2Wired (VendorID: 0x0B05, ProductID: 0x1908)
[2022-03-10 15:41:52,059] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 15:41:52,062] [DEBUG] found 8 subdevices:
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw1: ASUSTeK ROG PUGIO II interface 0
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw2: ASUSTeK ROG PUGIO II interface 1 [using as keyboard]
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw2: ASUSTeK ROG PUGIO II interface 1 [using as keyboard]
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as control]
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as control]
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as control]
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as control]
[2022-03-10 15:41:52,062] [DEBUG] /dev/hidraw4: ASUSTeK ROG PUGIO II interface 3
[2022-03-10 15:41:52,062] [DEBUG] opening control subdevice
[2022-03-10 15:41:52,063] [DEBUG] opening device "/dev/hidraw3"
[2022-03-10 15:41:52,063] [DEBUG] opening keyboard subdevice
[2022-03-10 15:41:52,063] [DEBUG] opening device "/dev/hidraw2"
[2022-03-10 15:41:52,067] [DEBUG] getting profile and firmware versions
[2022-03-10 15:41:52,067] [DEBUG] > 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 33, in <module>
    sys.exit(load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv-config')())
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 402, in rogdrv_config
    app.run()
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 200, in query
    self.write(request)
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 183, in write
    self._ctl.write(data)
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/hid.py", line 35, in write
    self._device.write(data)
  File "/usr/lib/python3/dist-packages/hidapi.py", line 186, in write
    raise IOError("Failed to write to HID device.")
OSError: Failed to write to HID device.

kyokenn, 2022-03-10:
Can you confirm that /dev/hidraw3 is a "Consumer Control" interface and not just keyboard or mouse? You can check it in ```dmesg```

kyokenn, 2022-03-10:
You can try to set those options:
```
    control_interface = 0
    keyboard_interface = 2
```

there - https://github.com/kyokenn/rogdrv/blob/master/rog/device/rog.py#L227

And also try other values for "control_interface". It's an interface number.

bam80, 2022-03-10:
Yep, that worked

bam80, 2022-03-15:
Do you need me to make a PR?

commit d4a04133 (2022-03-15): Pugio 2: OSError: Failed to write to HID device.

Fixes #36
closed 2022-03-15 by kyokenn