kyokenn/rogdrv #38: Pugio 2: DPI, Polling rate, Performance menus do not work

bam80, 2022-03-10:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/ui/handler.py", line 106, in on_dpi_choice
    dpis, _, _, _ = self._device.get_dpi_rate_response_snapping()
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 434, in get_dpi_rate_response_snapping
    rate = defs.POLLING_RATES[response[i]]
KeyError: 15
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/ui/handler.py", line 117, in on_rate_choice
    _, rate, _, _ = self._device.get_dpi_rate_response_snapping()
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 434, in get_dpi_rate_response_snapping
    rate = defs.POLLING_RATES[response[i]]
KeyError: 15
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/ui/handler.py", line 142, in on_perf_choice
    _, _, _, snapping = self._device.get_dpi_rate_response_snapping()
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 434, in get_dpi_rate_response_snapping
    rate = defs.POLLING_RATES[response[i]]
KeyError: 15

kyokenn, 2022-03-10:
Can you run with "--debug" option? I need to see the actual packets.

bam80, 2022-03-10:
```
bam@Feedme:~$ rogdrv --debug
[2022-03-10 16:24:23,628] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-03-10 16:24:23,688] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,690] [DEBUG] 0 devices found
[2022-03-10 16:24:23,690] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-03-10 16:24:23,690] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,692] [DEBUG] 0 devices found
[2022-03-10 16:24:23,692] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-03-10 16:24:23,692] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,694] [DEBUG] 0 devices found
[2022-03-10 16:24:23,694] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2022-03-10 16:24:23,694] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,696] [DEBUG] 0 devices found
[2022-03-10 16:24:23,696] [DEBUG] searching for device PugioGladiusII (VendorID: 0x0B05, ProductID: 0x1851)
[2022-03-10 16:24:23,696] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,698] [DEBUG] 0 devices found
[2022-03-10 16:24:23,698] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2022-03-10 16:24:23,698] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,700] [DEBUG] 0 devices found
[2022-03-10 16:24:23,700] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2022-03-10 16:24:23,700] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,703] [DEBUG] 0 devices found
[2022-03-10 16:24:23,703] [DEBUG] searching for device StrixImpactIIWirelessWired (VendorID: 0x0B05, ProductID: 0x1947)
[2022-03-10 16:24:23,703] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,705] [DEBUG] 0 devices found
[2022-03-10 16:24:23,705] [DEBUG] searching for device StrixImpactIIWireless (VendorID: 0x0B05, ProductID: 0x1949)
[2022-03-10 16:24:23,705] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,707] [DEBUG] 0 devices found
[2022-03-10 16:24:23,707] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2022-03-10 16:24:23,707] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,709] [DEBUG] 0 devices found
[2022-03-10 16:24:23,709] [DEBUG] searching for device KerisWireless (VendorID: 0x0B05, ProductID: 0x1960)
[2022-03-10 16:24:23,709] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,711] [DEBUG] 0 devices found
[2022-03-10 16:24:23,711] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x195E)
[2022-03-10 16:24:23,711] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,712] [DEBUG] 0 devices found
[2022-03-10 16:24:23,713] [DEBUG] searching for device ChakramWireless (VendorID: 0x0B05, ProductID: 0x18E5)
[2022-03-10 16:24:23,713] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,714] [DEBUG] 0 devices found
[2022-03-10 16:24:23,714] [DEBUG] searching for device ChakramWirelessWired (VendorID: 0x0B05, ProductID: 0x18E3)
[2022-03-10 16:24:23,715] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,716] [DEBUG] 0 devices found
[2022-03-10 16:24:23,716] [DEBUG] searching for device Pugio2 (VendorID: 0x0B05, ProductID: 0x1906)
[2022-03-10 16:24:23,716] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,718] [DEBUG] 0 devices found
[2022-03-10 16:24:23,718] [DEBUG] searching for device Pugio2Wired (VendorID: 0x0B05, ProductID: 0x1908)
[2022-03-10 16:24:23,718] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-10 16:24:23,721] [DEBUG] found 8 subdevices:
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw1: ASUSTeK ROG PUGIO II interface 0 [using as control]
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw2: ASUSTeK ROG PUGIO II interface 1
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw2: ASUSTeK ROG PUGIO II interface 1
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw3: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-10 16:24:23,721] [DEBUG] /dev/hidraw4: ASUSTeK ROG PUGIO II interface 3
[2022-03-10 16:24:23,721] [DEBUG] opening control subdevice
[2022-03-10 16:24:23,721] [DEBUG] opening device "/dev/hidraw1"
[2022-03-10 16:24:23,778] [DEBUG] opening keyboard subdevice
[2022-03-10 16:24:23,778] [DEBUG] opening device "/dev/hidraw3"
[2022-03-10 16:24:23,939] [DEBUG] Using selector: EpollSelector
[2022-03-10 16:24:39,944] [DEBUG] getting DPI and polling rate
[2022-03-10 16:24:39,944] [DEBUG] number of dpi presets: 2
[2022-03-10 16:24:39,945] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-03-10 16:24:39,952] [DEBUG] < 12 04 00 00 03 00 07 00 0F 00 1F 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Traceback (most recent call last):
[... cut ...]
[2022-03-10 16:24:53,003] [DEBUG] < 12 04 00 00 03 00 07 00 0F 00 1F 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/ui/handler.py", line 142, in on_perf_choice
    _, _, _, snapping = self._device.get_dpi_rate_response_snapping()
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 434, in get_dpi_rate_response_snapping
    rate = defs.POLLING_RATES[response[i]]
KeyError: 15

kyokenn, 2022-03-10:
Add this:
```
    dpis = 4
```
to Pugio2 class and try again.

bam80, 2022-03-10:
Now I can see 4 DPI presets in submenu but I can't choose one - all the 
items are greyed out.

kyokenn, 2022-03-10:
Yeah, it's fine. You can select DPI preset using mouse buttons only. That's how all ASUS mice work. It's not possible to switch active preset from software.

bam80, 2022-03-10:
Is it possible to show the current DPI preset at least?
Also I somehow messed up the original DPI by switching profiles and return back. I had to restore it with the button..

closed 2022-03-15 by kyokenn

bam80, 2023-04-10:
> 3rd DPI preset (often marked by blue LED) is the default one (on mice with 4 DPI presets). It always get selected when switching a profile.

Could we pull this info to the Readme please?
Otherwise, it's hard to predict what's going on and perceived as a bug..

BTW, I just got it somehow reset even without  switching profiles.

bam80, 2023-04-10:
> > Is it possible to show the current DPI preset at least?
> No.

I can do this in the Windows setup app - it shows current DPI preset and dynamically updates it as it switched on the mouse.
The switching itself is seems possible via mouse itself only indeed.

kyokenn, 2023-04-13:
>The DPI got reset on ratbagcli list command issued. Is it a bug @kyokenn ?

It could be a bug, but I will need logs from ratbagd to check it.

>I can do this in the Windows setup app - it shows current DPI preset and dynamically updates it as it switched on the mouse.

I think the mouse sends key event, but doesn't stores the current DPI preset in onboard memory. So it could be implemented in the HID kernel module only.

commit 5f6f74ac (2022-03-15): Pugio 2: DPI, Polling rate, Performance menus do not work

Fixes #38