kyokenn/rogdrv #42: Pugio 2: wrong battery status

bam80, 2022-03-15:
I fully charged the mouse as seen in Gnome's Power Setting for Bluetooth devices, still get 0% charge in Battery menu when in RF mode: 
```
bam@Box:~/tmp/rogdrv$ rogdrv --debug
[2022-03-16 00:51:33,818] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-03-16 00:51:33,859] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,861] [DEBUG] 0 devices found
[2022-03-16 00:51:33,861] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-03-16 00:51:33,861] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,863] [DEBUG] 0 devices found
[2022-03-16 00:51:33,863] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-03-16 00:51:33,863] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,865] [DEBUG] 0 devices found
[2022-03-16 00:51:33,865] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2022-03-16 00:51:33,865] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,867] [DEBUG] 0 devices found
[2022-03-16 00:51:33,867] [DEBUG] searching for device PugioGladiusII (VendorID: 0x0B05, ProductID: 0x1851)
[2022-03-16 00:51:33,867] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,869] [DEBUG] 0 devices found
[2022-03-16 00:51:33,869] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2022-03-16 00:51:33,869] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,871] [DEBUG] 0 devices found
[2022-03-16 00:51:33,871] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2022-03-16 00:51:33,871] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,873] [DEBUG] 0 devices found
[2022-03-16 00:51:33,873] [DEBUG] searching for device StrixImpactIIWirelessWired (VendorID: 0x0B05, ProductID: 0x1947)
[2022-03-16 00:51:33,873] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,875] [DEBUG] 0 devices found
[2022-03-16 00:51:33,875] [DEBUG] searching for device StrixImpactIIWireless (VendorID: 0x0B05, ProductID: 0x1949)
[2022-03-16 00:51:33,875] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,876] [DEBUG] 0 devices found
[2022-03-16 00:51:33,876] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2022-03-16 00:51:33,876] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,878] [DEBUG] 0 devices found
[2022-03-16 00:51:33,878] [DEBUG] searching for device KerisWireless (VendorID: 0x0B05, ProductID: 0x1960)
[2022-03-16 00:51:33,878] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,880] [DEBUG] 0 devices found
[2022-03-16 00:51:33,880] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x195E)
[2022-03-16 00:51:33,880] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,882] [DEBUG] 0 devices found
[2022-03-16 00:51:33,882] [DEBUG] searching for device ChakramWireless (VendorID: 0x0B05, ProductID: 0x18E5)
[2022-03-16 00:51:33,882] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,884] [DEBUG] 0 devices found
[2022-03-16 00:51:33,884] [DEBUG] searching for device ChakramWirelessWired (VendorID: 0x0B05, ProductID: 0x18E3)
[2022-03-16 00:51:33,884] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,886] [DEBUG] 0 devices found
[2022-03-16 00:51:33,886] [DEBUG] searching for device Pugio2 (VendorID: 0x0B05, ProductID: 0x1906)
[2022-03-16 00:51:33,886] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,888] [DEBUG] 0 devices found
[2022-03-16 00:51:33,888] [DEBUG] searching for device Pugio2Wired (VendorID: 0x0B05, ProductID: 0x1908)
[2022-03-16 00:51:33,888] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-03-16 00:51:33,890] [DEBUG] found 8 subdevices:
[2022-03-16 00:51:33,890] [DEBUG] /dev/hidraw9: ASUSTeK ROG PUGIO II interface 0 [using as control]
[2022-03-16 00:51:33,891] [DEBUG] /dev/hidraw10: ASUSTeK ROG PUGIO II interface 1
[2022-03-16 00:51:33,891] [DEBUG] /dev/hidraw10: ASUSTeK ROG PUGIO II interface 1
[2022-03-16 00:51:33,891] [DEBUG] /dev/hidraw11: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-16 00:51:33,891] [DEBUG] /dev/hidraw11: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-16 00:51:33,891] [DEBUG] /dev/hidraw11: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-16 00:51:33,891] [DEBUG] /dev/hidraw11: ASUSTeK ROG PUGIO II interface 2 [using as keyboard]
[2022-03-16 00:51:33,891] [DEBUG] /dev/hidraw12: ASUSTeK ROG PUGIO II interface 3
[2022-03-16 00:51:33,891] [DEBUG] opening control subdevice
[2022-03-16 00:51:33,891] [DEBUG] opening device "/dev/hidraw9"
[2022-03-16 00:51:33,950] [DEBUG] opening keyboard subdevice
[2022-03-16 00:51:33,950] [DEBUG] opening device "/dev/hidraw11"
[2022-03-16 00:51:34,110] [DEBUG] Using selector: EpollSelector
[2022-03-16 00:51:43,735] [DEBUG] getting sleep timeout and battery alert level
[2022-03-16 00:51:43,736] [DEBUG] > 12 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-03-16 00:51:43,742] [DEBUG] < 12 07 00 00 04 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-03-16 00:51:43,742] [DEBUG] current battery alert level is 25%
[2022-03-16 00:51:43,742] [DEBUG] getting sleep timeout and battery alert level
[2022-03-16 00:51:43,742] [DEBUG] > 12 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-03-16 00:51:43,747] [DEBUG] < 12 07 00 00 04 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

bam80, 2022-03-16:
Now it shows correct charge, thanks.
About alert levels, I don't see 75% as an option, it was available before.
How can I test sleep timeout, should it apply to BT connection also, or only to RF one?

kyokenn, 2022-03-16:
You can't set alert level higher than 50% as far as I remember. It looks pointless to have an alert on such high charge level, because mouse could drop from 100% to 75% really fast and stay at 75% for longer period.

Sleep timeout applies to any wireless types of connection.

bam80, 2022-03-16:
75% was definitely an option, I set it in orig Windows app for testing purposes and your app displayed it as such.
Then I lowered the level at some point (via your app) and now can't see the 75% any more.

I have sleep timeout set to 2 min but can't see any evidence of sleepiness even for RF mode.

bam80, 2023-04-05:
Could I resurrect the report?
Battery status doesn't work for me again:
[image]

Also setting sleep timeouts doesn't seem to have any affect and doesn't preserve between the app rerun:
```
$ rogdrv --debug
[2023-04-05 19:28:40,921] [DEBUG] Loading drivers
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'etekcity'
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'sinowealth'
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'gskill'
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'logitech_g300'
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'logitech_g600'
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'sinowealth_nubwo'
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'roccat-kone-emp'
[2023-04-05 19:28:40,973] [ERROR] Unable to find driver 'roccat-kone-pure'
[2023-04-05 19:28:40,974] [ERROR] Unable to find driver 'steelseries'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/rog/ui/handler.py", line 246, in on_battery_choice
    _, charge, alert = self._device.get_sleep_charge_alert()
AttributeError: 'NoneType' object has no attribute 'get_sleep_charge_alert'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/rog/ui/handler.py", line 246, in on_battery_choice
    _, charge, alert = self._device.get_sleep_charge_alert()
AttributeError: 'NoneType' object has no attribute 'get_sleep_charge_alert'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/rog/ui/handler.py", line 246, in on_battery_choice
    _, charge, alert = self._device.get_sleep_charge_alert()
AttributeError: 'NoneType' object has no attribute 'get_sleep_charge_alert'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/rog/ui/handler.py", line 221, in on_sleep_choice
    sleep, _, _ = self._device.get_sleep_charge_alert()
AttributeError: 'NoneType' object has no attribute 'get_sleep_charge_alert'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/rog/ui/handler.py", line 233, in on_sleep
    sleep_old, _, alert = self._device.get_sleep_charge_alert()
AttributeError: 'NoneType' object has no attribute 'get_sleep_charge_alert'

bam80, 2023-04-05:
UPDATE:
the reason might be `ratbagcli` doesn't see the device:
```
$ ratbagcli list 
ERROR: ratbag: Unable to find driver 'etekcity'
ERROR: ratbag: Unable to find driver 'sinowealth'
ERROR: ratbag: Unable to find driver 'gskill'
ERROR: ratbag: Unable to find driver 'logitech_g300'
ERROR: ratbag: Unable to find driver 'logitech_g600'
ERROR: ratbag: Unable to find driver 'sinowealth_nubwo'
ERROR: ratbag: Unable to find driver 'roccat-kone-emp'
ERROR: ratbag: Unable to find driver 'roccat-kone-pure'
ERROR: ratbag: Unable to find driver 'steelseries'
# No supported devices available
```
Thoughts?

bam80, 2023-04-10:
@kyokenn friendly ping :)

kyokenn, 2023-04-10:
>Battery status doesn't work for me again

It's not implemented in ratbag-python so it will not be available in rogdrv. I want to move the battery settings and battery status to the kernel module - https://github.com/kyokenn/hid-asus-mouse . So the battery status will be available in any DE like GNOME and others.

>the reason might be ratbagcli doesn't see the device:

Can you run "sudo ratbagd --console-log-level debug" and show the output?

bam80, 2023-04-10:
> So the battery status will be available in any DE like GNOME and others.

Right now I see some evidence of the mouse discharge in Gnome Settings:
[image]
It probably is not precise because it jumped from 100 to 75 suddenly.
But still, it can work somehow without your driver?

Can you run "sudo ratbagd --console-log-level debug" and show the output?
```
$ sudo ratbagd --console-log-level debug
[sudo] password for bam: 
D|ratbag.driver: Loading drivers
Traceback (most recent call last):
  File "/usr/local/bin/ratbagd", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/ratbag/cli/ratbagd_dasbus.py", line 603, in main
    ratbagd.start()
  File "/usr/local/lib/python3.10/dist-packages/ratbag/cli/ratbagd_dasbus.py", line 521, in start
    self.bus.register_service(NAME_PREFIX)
  File "/usr/local/lib/python3.10/dist-packages/dasbus/connection.py", line 259, in register_service
    result = self.proxy.RequestName(service_name, flags)
  File "/usr/local/lib/python3.10/dist-packages/dasbus/client/handler.py", line 450, in _call_method
    return self._get_method_reply(
  File "/usr/local/lib/python3.10/dist-packages/dasbus/client/handler.py", line 483, in _get_method_reply
    return self._handle_method_error(error)
  File "/usr/local/lib/python3.10/dist-packages/dasbus/client/handler.py", line 509, in _handle_method_error
    raise exception from None
dasbus.error.DBusError: Connection ":1.29717" is not allowed to own the service "org.freedesktop.ratbag1" due to security policies in the configuration file

kyokenn, 2023-04-10:
>But still, it can work somehow without your driver?

Is it connected via bluetooth? ASUS mice can actually send the battery info over bluetooth, they are using HID-compatible protocol.

If the mouse is connected via cable or RF then it sends the battery info using it's own protocol. I haven't seen the other cases yet.

kyokenn, 2023-04-10:
>dasbus.error.DBusError: Connection ":1.29717" is not allowed to own the service "org.freedesktop.ratbag1" due to security policies in the configuration file

You need to copy dbus config file:
```
sudo cp -fv ./ratbag-python/dbus/org.freedesktop.ratbag1.conf /etc/dbus-1/system.d/
```

bam80, 2023-04-10:
> Is it connected via bluetooth? ASUS mice can actually send the battery info over bluetooth, they are using HID-compatible protocol.

Yes, it is. What is the point in your `hid-asus-mouse` driver then?

bam80, 2023-04-10:
> You need to copy dbus config file:
> sudo cp -fv ./ratbag-python/dbus/org.freedesktop.ratbag1.conf /etc/dbus-1/system.d/

Still doesn't seem to work:
```
$ sudo ratbagd --console-log-level debug
D|ratbag.driver: Loading drivers
```
no more output here.

If I run the `ratbagcli` in the other console, nothing changes:
```
bam@Feedme:~/tmp/rogdrv/ratbag-python$ ratbagcli list
D|ratbag.driver: Loading drivers
E|ratbag: Unable to find driver 'etekcity'
E|ratbag: Unable to find driver 'sinowealth'
E|ratbag: Unable to find driver 'gskill'
E|ratbag: Unable to find driver 'logitech_g300'
E|ratbag: Unable to find driver 'logitech_g600'
E|ratbag: Unable to find driver 'sinowealth_nubwo'
E|ratbag: Unable to find driver 'roccat-kone-emp'
E|ratbag: Unable to find driver 'roccat-kone-pure'
E|ratbag: Unable to find driver 'steelseries'
# No supported devices available
bam@Feedme:~/tmp/rogdrv/ratbag-python$ sudo ratbagcli list
[sudo] password for bam: 
D|ratbag.driver: Loading drivers
E|ratbag: Unable to find driver 'etekcity'
E|ratbag: Unable to find driver 'sinowealth'
E|ratbag: Unable to find driver 'gskill'
E|ratbag: Unable to find driver 'logitech_g300'
E|ratbag: Unable to find driver 'logitech_g600'
E|ratbag: Unable to find driver 'sinowealth_nubwo'
E|ratbag: Unable to find driver 'roccat-kone-emp'
E|ratbag: Unable to find driver 'roccat-kone-pure'
E|ratbag: Unable to find driver 'steelseries'
# No supported devices available

kyokenn, 2023-04-10:
>Yes, it is. What is the point in your hid-asus-mouse driver then (for this particular mouse)?

This driver can be used with USB/RF connection only. Also both rogdrv + ratbag-python and libratbag can be used when thee mouse is connected over USB or RF.

>No supported devices available

Make sure that you have connected your mouse via USB or RF.

bam80, 2023-04-10:
> Make sure that you have connected your mouse via USB or RF.

Oh, can we make it explicitly clear from the Readme, please?
I can assure you it's not obvious even for those who used the driver in the past (so I completely forgot about it after a year of not using it)

bam80, 2023-04-10:
> You need to copy dbus config file:
sudo cp -fv ./ratbag-python/dbus/org.freedesktop.ratbag1.conf /etc/dbus-1/system.d/

This should also be made more clear in Readme, IMO.
Right now it's mentioned in "See also" chapter and perceived as something auxiliary and not strictly needed to run the app.

bam80, 2023-04-10:
> > Battery status doesn't work for me again

>It's not implemented in ratbag-python so it will not be available in rogdrv. I want to move the battery settings and battery status to the kernel module - https://github.com/kyokenn/hid-asus-mouse . So the battery status will be available in any DE like GNOME and others.

What about sleep settings? I do not see them in the new app version too.

kyokenn, 2023-04-13:
>Oh, can we make it explicitly clear from the Readme, please?
>I can assure you it's not obvious even for those who used the driver in the past (so I completely forgot about it after a year of not using it)

Done.

>This should also be made more clear in Readme, IMO.
>Right now it's mentioned in "See also" chapter and perceived as something auxiliary and not strictly needed to run the app.

Yes, it's not needed for the main app rogdrv/rogdrv-config. But if you want to use ratbagd, then you will need it.

>What about sleep settings? I do not see them in the new app version too.

It's not implemented yet.