kyokenn/rogdrv #43: Asus ROG Gladius II Origin

JoneKone, 2022-03-22:
```
@:~/git/rogdrv$ rogdrv-config led
 LED       | MODE    | COLOR  | BRIGHTNESS
-----------|---------|--------|-----------
 1 logo    | default | 000000 | 4
 2 wheel   | wave    | 000000 | 1
 3 bottom  | wave    | 000000 | 1
jonekone@TreeOfLight:~/git/rogdrv$ rogdrv-config led 1
usage: rogdrv-config led [-h] [-l LED] [-c COLOR] [-b BRIGHTNESS] [-m MODE]
rogdrv-config led: error: unrecognized arguments: 1
@:~/git/rogdrv$ rogdrv-config led -h
usage: rogdrv-config led [-h] [-l LED] [-c COLOR] [-b BRIGHTNESS] [-m MODE]

optional arguments:
  -h, --help            show this help message and exit
  -l LED, --led LED     LED to set: all (default), logo, wheel, bottom
  -c COLOR, --color COLOR
                        Color in hex. HTML format, ex.: aabbcc
  -b BRIGHTNESS, --brightness BRIGHTNESS
                        Brightness: 0 - 4, default is 0
  -m MODE, --mode MODE  LED mode: default, breath, rainbow, wave, reactive, flasher, battery
@:~/git/rogdrv$ rogdrv-config led -l LED -b 0
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 33, in <module>
    sys.exit(load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv-config')())
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 402, in rogdrv_config
    app.run()
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 93, in run
    method()
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/__main__.py", line 223, in led
    self._device.set_led(
  File "/usr/local/lib/python3.9/dist-packages/rogdrv-0.3.0-py3.9.egg/rog/device/base.py", line 357, in set_led
    request[2] = defs.LED_NAMES.index(name)
ValueError: tuple.index(x): x not in tuple
@:~/git/rogdrv$ 
```

I tried the gui, didn't turn off this horribele rgb
I tried this command xD i have no idea what went wrong.

JoneKone, 2022-03-23:
it somehow worked btw. i managed to turn everything off

closed 2022-09-01 by kyokenn

JoneKone, 2022-09-08:
You haven't actually fixed anything still won't work. Ubuntu 24.04.1

kyokenn, 2022-09-08:
The "LED" should be replaced with actual LED's name: "logo", "wheel", "bottom", or "all" (to set all LEDs).
So the command will looks like: "rogdrv-config led -l all -b 0"