kyokenn/rogdrv #44: Chakram X support ???

1412kauti, 2022-06-15:
```
[2022-06-15 12:11:02,631] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-06-15 12:11:02,657] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,657] [DEBUG] 0 devices found
[2022-06-15 12:11:02,657] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-06-15 12:11:02,657] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,657] [DEBUG] 0 devices found
[2022-06-15 12:11:02,657] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-06-15 12:11:02,657] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,657] [DEBUG] 0 devices found
[2022-06-15 12:11:02,657] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2022-06-15 12:11:02,657] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,658] [DEBUG] 0 devices found
[2022-06-15 12:11:02,658] [DEBUG] searching for device PugioGladiusII (VendorID: 0x0B05, ProductID: 0x1851)
[2022-06-15 12:11:02,658] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,658] [DEBUG] 0 devices found
[2022-06-15 12:11:02,658] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2022-06-15 12:11:02,658] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,658] [DEBUG] 0 devices found
[2022-06-15 12:11:02,658] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2022-06-15 12:11:02,658] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,658] [DEBUG] 0 devices found
[2022-06-15 12:11:02,658] [DEBUG] searching for device StrixImpactIIWireless (VendorID: 0x0B05, ProductID: 0x1949)
[2022-06-15 12:11:02,658] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,658] [DEBUG] 0 devices found
[2022-06-15 12:11:02,658] [DEBUG] searching for device StrixImpactIIWirelessWired (VendorID: 0x0B05, ProductID: 0x1947)
[2022-06-15 12:11:02,658] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,658] [DEBUG] 0 devices found
[2022-06-15 12:11:02,658] [DEBUG] searching for device StrixImpactIIElectroPunk (VendorID: 0x0B05, ProductID: 0x1956)
[2022-06-15 12:11:02,658] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,659] [DEBUG] 0 devices found
[2022-06-15 12:11:02,659] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2022-06-15 12:11:02,659] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,659] [DEBUG] 0 devices found
[2022-06-15 12:11:02,659] [DEBUG] searching for device KerisWireless (VendorID: 0x0B05, ProductID: 0x1960)
[2022-06-15 12:11:02,659] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,659] [DEBUG] 0 devices found
[2022-06-15 12:11:02,659] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x195E)
[2022-06-15 12:11:02,659] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,659] [DEBUG] 0 devices found
[2022-06-15 12:11:02,659] [DEBUG] searching for device ChakramWireless (VendorID: 0x0B05, ProductID: 0x18E5)
[2022-06-15 12:11:02,659] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,659] [DEBUG] 0 devices found
[2022-06-15 12:11:02,659] [DEBUG] searching for device ChakramWirelessWired (VendorID: 0x0B05, ProductID: 0x18E3)
[2022-06-15 12:11:02,659] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,659] [DEBUG] 0 devices found
[2022-06-15 12:11:02,660] [DEBUG] searching for device Pugio2 (VendorID: 0x0B05, ProductID: 0x1906)
[2022-06-15 12:11:02,660] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,660] [DEBUG] 0 devices found
[2022-06-15 12:11:02,660] [DEBUG] searching for device Pugio2Wired (VendorID: 0x0B05, ProductID: 0x1908)
[2022-06-15 12:11:02,660] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-06-15 12:11:02,660] [DEBUG] 0 devices found
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/__main__.py", line 65, in run
    e = self._device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 11, in <module>
    load_entry_point('rogdrv==0.3.0', 'console_scripts', 'rogdrv')()
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/__main__.py", line 393, in rogdrv
    gtk3_main(app.device)
  File "/usr/local/lib/python3.8/dist-packages/rogdrv-0.3.0-py3.8.egg/rog/ui/__init__.py", line 49, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'
[2022-06-15 12:11:02,955] [DEBUG] Using selector: EpollSelector
```
Im using a new ROG Chakram X...can the support for it be added please ?

kyokenn, 2022-06-21:
What Product ID is it? You can check in ```lsusb```

1412kauti, 2022-06-22:
`Bus 001 Device 003: ID 0b05:1a1a ASUSTek Computer, Inc. ROG CHAKRAM X
`