kyokenn/rogdrv #45: Gladius III support

user030276, 2022-08-29:
Kyoken,

Greetings! Thank you for your project, rogdrv. I wonder if we can add my Gladius III.

Please say if you have any thoughts.

All the best,

```
user@m5:~$ LIBUSB_DEBUG=9 sudo -E lsusb -v -d 0b05:197b
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000003] [00003272] libusb: debug [libusb_init] created default context
[ 0.000016] [00003272] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.000041] [00003272] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000049] [00003272] libusb: debug [get_kernel_version] reported kernel version is 4.19.0-16-amd64
[ 0.000051] [00003272] libusb: debug [op_init] bulk continuation flag supported
[ 0.000053] [00003272] libusb: debug [op_init] zero length packet flag supported
[ 0.000054] [00003272] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
[ 0.000059] [00003272] libusb: debug [op_init] sysfs can relate devices
[ 0.000060] [00003272] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000298] [00003273] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.000770] [00003272] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.000775] [00003272] libusb: debug [linux_get_device_address] scan usb1
[ 0.000792] [00003272] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.000793] [00003272] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.000796] [00003272] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.000852] [00003272] libusb: debug [linux_get_device_address] getting address for device: 1-5 detached: 0
[ 0.000855] [00003272] libusb: debug [linux_get_device_address] scan 1-5
[ 0.000897] [00003272] libusb: debug [linux_get_device_address] bus=1 dev=46
[ 0.000899] [00003272] libusb: debug [linux_enumerate_device] busnum 1 devaddr 46 session_id 302
[ 0.000901] [00003272] libusb: debug [linux_enumerate_device] allocating new device for 1/46 (session 302)
[ 0.000939] [00003272] libusb: debug [linux_get_parent_info] Dev 0x55e5d7d10a80 (1-5) has parent 0x55e5d7d0b500 (usb1) port 5
[ 0.000955] [00003272] libusb: debug [linux_get_device_address] getting address for device: 1-6 detached: 0
[ 0.000970] [00003272] libusb: debug [linux_get_device_address] scan 1-6
[ 0.000996] [00003272] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.000999] [00003272] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.001000] [00003272] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.001041] [00003272] libusb: debug [linux_get_parent_info] Dev 0x55e5d7d10b40 (1-6) has parent 0x55e5d7d0b500 (usb1) port 6
[ 0.001057] [00003272] libusb: debug [linux_get_device_address] getting address for device: 1-6.1 detached: 0
[ 0.001060] [00003272] libusb: debug [linux_get_device_address] scan 1-6.1
[ 0.001074] [00003272] libusb: debug [linux_get_device_address] bus=1 dev=21
[ 0.001076] [00003272] libusb: debug [linux_enumerate_device] busnum 1 devaddr 21 session_id 277
[ 0.001079] [00003272] libusb: debug [linux_enumerate_device] allocating new device for 1/21 (session 277)
[ 0.001092] [00003272] libusb: debug [linux_get_parent_info] Dev 0x55e5d7d10c00 (1-6.1) has parent 0x55e5d7d10b40 (1-6) port 1
[ 0.001107] [00003272] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.001109] [00003272] libusb: debug [linux_get_device_address] scan usb2
[ 0.001122] [00003272] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.001124] [00003272] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.001127] [00003272] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.001145] [00003272] libusb: debug [usbi_add_pollfd] add fd 7 events 1
[ 0.001151] [00003272] libusb: debug [usbi_io_init] using timerfd for timeouts
[ 0.001154] [00003272] libusb: debug [usbi_add_pollfd] add fd 9 events 1
[ 0.001158] [00003272] libusb: debug [libusb_get_device_list]
[ 0.001164] [00003272] libusb: debug [libusb_get_device_descriptor]
[ 0.001166] [00003272] libusb: debug [libusb_get_device_descriptor]
[ 0.001169] [00003272] libusb: debug [libusb_get_device_descriptor]
[ 0.001171] [00003272] libusb: debug [libusb_get_device_descriptor]

Bus 001 Device 046: ID 0b05:197b ASUSTek Computer, Inc.
[ 0.001252] [00003272] libusb: debug [libusb_open] open 1.46
[ 0.001276] [00003272] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 0.001279] [00003272] libusb: debug [libusb_get_device_descriptor]
[ 0.001458] [00003272] libusb: debug [libusb_alloc_transfer] transfer 0x55e5d7d0c250
[ 0.001461] [00003272] libusb: debug [libusb_submit_transfer] transfer 0x55e5d7d0c250
[ 0.001464] [00003272] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.001477] [00003272] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.001481] [00003272] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.001484] [00003272] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.001603] [00003272] libusb: debug [handle_events] poll() returned 1
[... 204 lines cut ...]
[ 0.003238] [00003272] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.003240] [00003272] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.003331] [00003272] libusb: debug [handle_events] poll() returned 1
[ 0.003335] [00003272] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=2
[ 0.003338] [00003272] libusb: debug [handle_control_completion] handling completion status 0
[ 0.003340] [00003272] libusb: debug [disarm_timerfd]
[ 0.003343] [00003272] libusb: debug [usbi_handle_transfer_completion] transfer 0x55e5d7d0cfd0 has callback 0x7f128b8302c0
[ 0.003346] [00003272] libusb: debug [sync_transfer_cb] actual_length=2
[ 0.003349] [00003272] libusb: debug [libusb_free_transfer] transfer 0x55e5d7d0cfd0
Device Status:     0x0000
  (Bus Powered)
[ 0.003357] [00003272] libusb: debug [libusb_close]
[ 0.003361] [00003272] libusb: debug [usbi_remove_pollfd] remove fd 10
[ 0.003366] [00003272] libusb: debug [libusb_get_device_descriptor]
[ 0.003391] [00003272] libusb: debug [libusb_exit]
[ 0.003394] [00003272] libusb: debug [libusb_exit] destroying default context
[ 0.003397] [00003272] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.003400] [00003272] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.003404] [00003272] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[ 0.003407] [00003272] libusb: debug [handle_events] poll() returned 0
[ 0.003410] [00003272] libusb: debug [usbi_remove_pollfd] remove fd 7
[ 0.003416] [00003272] libusb: debug [usbi_remove_pollfd] remove fd 9
[ 0.003427] [00003273] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
user@m5:~$
```

kyokenn, 2022-08-31:
I have added Gladius III to the rogdrv. You can test if it works.

user030276, 2022-08-31:
Thank you kyokenn! You rule! 
```
user@m5:~/dev/rogdrv$ rogdrv --console --debug
[2022-08-31 16:08:54,447] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-08-31 16:08:54,485] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:08:54,486] [DEBUG] 0 devices found
[2022-08-31 16:08:54,486] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-08-31 16:08:54,486] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:08:54,486] [DEBUG] 0 devices found
[2022-08-31 16:08:54,486] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-08-31 16:08:54,486] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:08:54,486] [DEBUG] 0 devices found
[2022-08-31 16:08:54,486] [DEBUG] searching for device Gladius3 (VendorID: 0x0B05, ProductID: 0x197B)
[2022-08-31 16:08:54,486] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:08:54,487] [DEBUG] found 3 subdevices:
[2022-08-31 16:08:54,487] [DEBUG] /dev/hidraw1: ASUSTeK ROG GLADIUS III interface 0
[2022-08-31 16:08:54,487] [DEBUG] /dev/hidraw2: ASUSTeK ROG GLADIUS III interface 1 [using as keyboard]
[2022-08-31 16:08:54,487] [DEBUG] /dev/hidraw3: ASUSTeK ROG GLADIUS III interface 2 [using as control]
[2022-08-31 16:08:54,487] [DEBUG] opening control subdevice
[2022-08-31 16:08:54,487] [DEBUG] opening device "/dev/hidraw3"
[2022-08-31 16:08:54,487] [DEBUG] opening keyboard subdevice
[2022-08-31 16:08:54,487] [DEBUG] opening device "/dev/hidraw2"
```
So far, so good. Next: 
`user@m5:~$ rogdrv-config profile -p 1`
All 4 light profiles are correctly set, starting from 1. 

 $rogdvr-config dpi --preset
The number of DPI presets prints as 2. The actual number should be 4. 
No change with any number and --preset 
Pressing the DPI button changes the DPI, and prints 'device is in invalid state...' and is shown below: 

```
user@m5:~$ rogdrv-config dpi -p 1 --debug
[2022-08-31 16:21:53,057] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-08-31 16:21:53,095] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:21:53,096] [DEBUG] 0 devices found
[2022-08-31 16:21:53,096] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-08-31 16:21:53,096] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:21:53,096] [DEBUG] 0 devices found
[2022-08-31 16:21:53,096] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-08-31 16:21:53,096] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:21:53,096] [DEBUG] 0 devices found
[2022-08-31 16:21:53,096] [DEBUG] searching for device Gladius3 (VendorID: 0x0B05, ProductID: 0x197B)
[2022-08-31 16:21:53,096] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:21:53,097] [DEBUG] found 3 subdevices:
[2022-08-31 16:21:53,097] [DEBUG] /dev/hidraw1: ASUSTeK ROG GLADIUS III interface 0
[2022-08-31 16:21:53,097] [DEBUG] /dev/hidraw2: ASUSTeK ROG GLADIUS III interface 1 [using as keyboard]
[2022-08-31 16:21:53,097] [DEBUG] /dev/hidraw3: ASUSTeK ROG GLADIUS III interface 2 [using as control]
[2022-08-31 16:21:53,097] [DEBUG] opening control subdevice
[2022-08-31 16:21:53,097] [DEBUG] opening device "/dev/hidraw3"
[2022-08-31 16:21:53,097] [DEBUG] opening keyboard subdevice
[2022-08-31 16:21:53,097] [DEBUG] opening device "/dev/hidraw2"
[2022-08-31 16:21:53,098] [DEBUG] getting DPI and polling rate
[2022-08-31 16:21:53,098] [DEBUG] number of dpi presets: 2
[2022-08-31 16:21:53,098] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:55,409] [DEBUG] < 03 12 08 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:55,409] [DEBUG] device is in invalid state, retrying (1/10)
[2022-08-31 16:21:55,410] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:56,815] [DEBUG] < 03 12 08 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:56,815] [DEBUG] device is in invalid state, retrying (2/10)
[2022-08-31 16:21:56,816] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:58,018] [DEBUG] < 03 12 08 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:58,018] [DEBUG] device is in invalid state, retrying (3/10)
[2022-08-31 16:21:58,019] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:59,006] [DEBUG] < 03 12 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:59,006] [DEBUG] device is in invalid state, retrying (4/10)
[2022-08-31 16:21:59,007] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
    super().set_dpi(dpi / 2, preset=preset)
  File "/usr/local/lib/python3.7/dist-packages/rog/device/base.py", line 463, in set_dpi
    self.query(bytes(request))
ValueError: bytes must be in range(0, 256)
user@m5:~$ 
```

user@m5:~$ rogdrv-config dump -o omg.json ## creates 0 byte file, hangs. 

user@m5:~$ rogdrv-config rate -r 500 --debug #40 
^polling rate does not change, but this rate command does change DPI

OK ! Thank you very much again. Please tell me if you want anything specific tested or done. 
Good luck,

user030276, 2022-08-31:
setting LED mode partly works
the brightness 4 is dim
there is a second color displayed --color aaaaaa gives white and then red in --mode reactive
the different modes are all correct
the side LED does not work in reactive mode, seems to work in all other modes

 ```
user@m5:~$ rogdrv-config led --debug --mode reactive  --color aaaaaa --brightness 4 
[2022-08-31 16:52:21,736] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-08-31 16:52:21,774] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:52:21,774] [DEBUG] 0 devices found
[2022-08-31 16:52:21,775] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-08-31 16:52:21,775] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:52:21,775] [DEBUG] 0 devices found
[2022-08-31 16:52:21,775] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-08-31 16:52:21,775] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:52:21,775] [DEBUG] 0 devices found
[2022-08-31 16:52:21,775] [DEBUG] searching for device Gladius3 (VendorID: 0x0B05, ProductID: 0x197B)
[2022-08-31 16:52:21,775] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-08-31 16:52:21,776] [DEBUG] found 3 subdevices:
[2022-08-31 16:52:21,776] [DEBUG] /dev/hidraw1: ASUSTeK ROG GLADIUS III interface 0
[2022-08-31 16:52:21,776] [DEBUG] /dev/hidraw2: ASUSTeK ROG GLADIUS III interface 1 [using as keyboard]
[2022-08-31 16:52:21,776] [DEBUG] /dev/hidraw3: ASUSTeK ROG GLADIUS III interface 2 [using as control]
[2022-08-31 16:52:21,776] [DEBUG] opening control subdevice
[2022-08-31 16:52:21,776] [DEBUG] opening device "/dev/hidraw3"
[2022-08-31 16:52:21,776] [DEBUG] opening keyboard subdevice
[2022-08-31 16:52:21,776] [DEBUG] opening device "/dev/hidraw2"
[2022-08-31 16:52:21,777] [DEBUG] setting LED all color to rgb(170,170,170) with mode reactive and brightness 4
[2022-08-31 16:52:21,777] [DEBUG] > 51 28 03 00 04 04 AA AA AA 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

user030276, 2022-09-01:
left click generates some random keyboard output. the output keeps changing.  here is left click.
```
[2022-09-01 09:31:44,182] [DEBUG] < 00 00 00 00 01 00 03 00 00 00 00
[2022-09-01 09:31:44,182] [DEBUG] < 00 00 00 00 01 00 03 27 00 00 00
[2022-09-01 09:31:44,313] [DEBUG] < 00 00 00 00 00 00 03 00 00 00 00
```
^ bytes  "27 00" changes each click.

right click is OK. 

middle mouse button / scroll wheel generates letter "a" or number 0 and maybe new line, and other characters. here is scroll wheel click. 
```

[2022-09-01 09:27:36,184] [DEBUG] < 00 00 00 00 04 00 03 00 00 00 00
[2022-09-01 09:27:36,288] [DEBUG] < 00 00 00 00 00 00 03 00 00 00 00

```
back button outputs . character
```
[2022-09-01 09:39:00,733] [DEBUG] < 00 00 00 00 08 00 03 00 00 00 00
[2022-09-01 09:39:00,982] [DEBUG] < 00 00 00 00 00 00 03 00 00 00 00
```

forward button outputs m
```
[2022-09-01 09:39:46,844] [DEBUG] < 00 00 00 00 10 00 03 00 00 00 00
[2022-09-01 09:39:46,980] [DEBUG] < 00 00 00 00 00 00 03 00 00 00 00

```

kyokenn, 2022-09-01:
> The number of DPI presets prints as 2. The actual number should be 4.

I have changed number of DPI presets, forgot that all recent mice have 4 presets.

>user@m5:~$ rogdrv-config rate -r 500 --debug
>^polling rate does not change, but this rate command does change DPI

Correct number of DPI presets should solve it and also other options like angle snapping and button response delay.

```
[2022-08-31 16:21:53,098] [DEBUG] number of dpi presets: 2
[2022-08-31 16:21:53,098] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:55,409] [DEBUG] < 03 12 08 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-08-31 16:21:55,409] [DEBUG] device is in invalid state, retrying (1/10)
```
This is bad, you can try to replug the mouse. Maybe it just stuck in some wrong state. If it still doesn't helps then the query command has changed and you will need an USB debugger to find the correct one.

>$ rogdrv-config dpi -d 800 --debug

Can you show debug output of this one?

>middle mouse button / scroll wheel generates letter "a" or number 0 and maybe new line, and other characters. here is scroll wheel click.

This is weird. Does the buttons actually work without a driver? You may not need to debug the events if the buttons are already working. :smiley:

user030276, 2022-09-01:
Thank you for your reply kyokenn, and thank you for the help.

dpi presets do not change mouse dpi 

```
user@m5:~$ rogdrv-config dpi --preset 3 --debug4
[2022-09-01 12:29:31,435] [DEBUG] number of dpi presets: 4
[2022-09-01 12:29:31,435] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
dpi flag does not change dpi
```
user@m5:~$ rogdrv-config dpi --dpi 400 --debug
[2022-09-01 12:31:04,643] [DEBUG] setting DPI to 200.0 for preset 1
[2022-09-01 12:31:04,643] [DEBUG] > 51 31 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

All buttons work without a driver.  
With your driver they work and give the extra keyboard outputs. 

Mouse polling is now correct! Way to go!
```
user@m5:~$ rogdrv-config rate -r 500 --debug
[2022-09-01 12:35:23,018] [DEBUG] setting polling rate to 500
[2022-09-01 12:35:23,018] [DEBUG] > 51 31 04 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

rogdrv-config prints that device is in invalid state whenever DPI button is pushed. The DPI changes normally. 

Thank you again !

kyokenn, 2022-09-01:
```
[2022-09-01 12:29:31,435] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
This is only request (what the driver is sending to the mouse), but I also need a response (what the mouse answers) marked as "<".

user030276, 2022-09-01:
Thank you for the reply! 

with rogdrv
There is no response.  
With  ` $ rogdrv --console --debug` running, and then running 
 `$ rogdrv-config dpi --preset 2 --debug` there is no response, no < output in any debug window.

kyokenn, 2022-09-01:
> With $ rogdrv --console --debug running, and then running

You may need to close it, because it could block the reading from device. Those 2 tools should not be running at the same time.

user030276, 2022-09-01:
Thank you again.

With only $ rogdrv-config running the debug ouput and effect is the same

sorry if this is a double post. i am not sure what reply by email does.

kyokenn, 2022-09-01:
> With only $ rogdrv-config running the debug ouput and effect is the same

I see. So it's stuck and waiting for response indefinitely. You will need a USB debugger (like https://github.com/djpnewton/busdog ) and Windows PC or VM with ASUS Armoury to get the actual commands for getting/setting DPI.

kyokenn, 2023-03-02:
I have switched back-end to ratbag-python, so there is a chance that I could work now.