kyokenn/rogdrv #46: ASUSTek TUF Gaming M3

badmansan, 2022-09-10:
Good day! I have got an error while execute rogdrv:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/dist-packages/rog/__main__.py", line 65, in run
    e = self._device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 8, in <module>
    sys.exit(rogdrv())
  File "/usr/local/lib/python3.8/dist-packages/rog/__main__.py", line 393, in rogdrv
    gtk3_main(app.device)
  File "/usr/local/lib/python3.8/dist-packages/rog/ui/__init__.py", line 49, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'
```
System: Linux Mint 20.2 (Ubuntu 20.04)
Seems the mouse is supported (I see "Consumer Control"):
```
[    1.488311] usb 1-1.1: New USB device found, idVendor=0b05, idProduct=1910, bcdDevice= 0.09
[    1.488313] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.488314] usb 1-1.1: Product: TUF GAMING M3
[    1.488314] usb 1-1.1: Manufacturer: ASUSTek
[    1.491333] hidraw: raw HID events driver (C) Jiri Kosina
[    1.497228] usbcore: registered new interface driver usbhid
[    1.497228] usbhid: USB HID core driver
[    1.499109] input: ASUSTek TUF GAMING M3 Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:0B05:1910.0001/input/input3
[    1.499179] input: ASUSTek TUF GAMING M3 as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:0B05:1910.0001/input/input4
[    1.499230] input: ASUSTek TUF GAMING M3 Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:0B05:1910.0001/input/input5
```
Is this mouse can be configured with rogdrv?

kyokenn, 2022-09-14:
I have added the mouse to the driver. You can test if it works.

badmansan, 2022-09-17:
Thank's @kyokenn. Unfortunately, I give back the mouse to the shop, so I can't test you fix. Sorry for that.

closed 2022-09-17 by badmansan