kyokenn/rogdrv #47: [ Chakram ] Missing button binding feature

k5nn, 2022-10-10:
I noticed that the button binding cell for the Chakram is "?" would like to confirm that the current version does not support it is there anything that I can help with? would really like the button binding feature be implement for this mouse

kyokenn, 2022-10-11:
"?" means untested, it still should work, but maybe not for all buttons. You can test it and tell which works and which doesn't.

k5nn, 2022-10-11:
My bad the joystick specifically has the behavior of the arrow buttons

kyokenn, 2022-10-12:
Can you run ```rogdrv-config bind --debug``` and post an output ?

It will show the current button bindings.

k5nn, 2022-12-08:
hello sorry it's taken me this long to get back to you I was busy with IRL stuff but here is the debug log for the ChakramWireless

`[2022-12-08 21:26:09,566] [DEBUG] searching for device ChakramWireless (VendorID: 0x0B05, ProductID: 0x18E5)
[2022-12-08 21:26:09,566] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-12-08 21:26:09,570] [DEBUG] found 8 subdevices:
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw2: ASUSTeK ROG CHAKRAM interface 0 [using as control]
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw3: ASUSTeK ROG CHAKRAM interface 1
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw3: ASUSTeK ROG CHAKRAM interface 1
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw4: ASUSTeK ROG CHAKRAM interface 2 [using as keyboard]
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw4: ASUSTeK ROG CHAKRAM interface 2 [using as keyboard]
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw4: ASUSTeK ROG CHAKRAM interface 2 [using as keyboard]
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw4: ASUSTeK ROG CHAKRAM interface 2 [using as keyboard]
[2022-12-08 21:26:09,570] [DEBUG] /dev/hidraw5: ASUSTeK ROG CHAKRAM interface 3
[2022-12-08 21:26:09,570] [DEBUG] opening control subdevice
[2022-12-08 21:26:09,570] [DEBUG] opening device "/dev/hidraw2"
[2022-12-08 21:26:09,627] [DEBUG] opening keyboard subdevice
[2022-12-08 21:26:09,627] [DEBUG] opening device "/dev/hidraw4"
[2022-12-08 21:26:09,630] [DEBUG] > 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-12-08 21:26:09,639] [DEBUG] < 12 05 00 00 F0 01 F1 01 F2 01 E4 01 E5 01 E6 01 E8 01 E9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

 BUTTON               | TYPE  | CODE - ACTION
----------------------|-------|--------------
 1 MOUSE_LEFT         | MOUSE | 0xF0 - MOUSE_LEFT
 2 MOUSE_RIGHT        | MOUSE | 0xF1 - MOUSE_RIGHT
 3 MOUSE_MIDDLE       | MOUSE | 0xF2 - MOUSE_MIDDLE
 4 MOUSE_SCROLL_UP    | MOUSE | 0xE8 - MOUSE_SCROLL_UP
 5 MOUSE_SCROLL_DOWN  | MOUSE | 0xE9 - MOUSE_SCROLL_DOWN`

kyokenn, 2022-12-09:
I have fixed number of buttons. So generic buttons will show up.

For the joystick settings I would need some logs from USB debugger (like busdog) and Armoury on Windows.