kyokenn/rogdrv #48: [Gladius II] make "DPI Target" button configurable

BierLiebHaber, 2022-10-13:
The Gladius II's DPI Target button is also configurable via software.
The corresponding mouse action is: 0xE7
I added the action here: https://github.com/BierLiebHaber/rogdrv/blob/master/rog/defs.py
But I'm not sure how to set it up correctly, so it gets used as the 9th Button for configuration (so right now its only settable as button 11)

If you can tell me how I can make it configurable as button 9, I can also create a  pull request with the necessary changes.

kyokenn, 2022-10-13:
This is so cool. Thanks for figuring out this and fixing that stuff. I checked action 0xE7 and it works for me too even that I don't have such button.
Can you run ```rogdrv-config bind --debug``` and post an output?

BierLiebHaber, 2022-10-13:
$rogdrv-config bind --debug
[2022-10-13 14:29:35,722] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-10-13 14:29:35,773] [DEBUG] getting list of devices using "hidapi-cffi"
[2022-10-13 14:29:35,775] [DEBUG] found 3 subdevices:
[2022-10-13 14:29:35,775] [DEBUG] /dev/hidraw0: ASUS ROG Gladius II interface 0
[2022-10-13 14:29:35,775] [DEBUG] /dev/hidraw1: ASUS ROG Gladius II interface 1 [using as keyboard]
[2022-10-13 14:29:35,775] [DEBUG] /dev/hidraw2: ASUS ROG Gladius II interface 2 [using as control]
[2022-10-13 14:29:35,775] [DEBUG] opening control subdevice
[2022-10-13 14:29:35,775] [DEBUG] opening device "/dev/hidraw2"
[2022-10-13 14:29:35,829] [DEBUG] opening keyboard subdevice
[2022-10-13 14:29:35,829] [DEBUG] opening device "/dev/hidraw1"
[2022-10-13 14:29:35,830] [DEBUG] > 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2022-10-13 14:29:35,833] [DEBUG] < 12 05 00 00 F0 01 F1 01 F2 01 E4 01 E5 01 E6 01 F2 01 E8 01 E9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 BUTTON               | TYPE  | CODE - ACTION
----------------------|-------|--------------
 1 MOUSE_LEFT         | MOUSE | 0xF0 - MOUSE_LEFT
 2 MOUSE_RIGHT        | MOUSE | 0xF1 - MOUSE_RIGHT
 3 MOUSE_MIDDLE       | MOUSE | 0xF2 - MOUSE_MIDDLE
 4 MOUSE_SCROLL_UP    | MOUSE | 0xE8 - MOUSE_SCROLL_UP
 5 MOUSE_SCROLL_DOWN  | MOUSE | 0xE9 - MOUSE_SCROLL_DOWN
 6 MOUSE_DPI          | MOUSE | 0xE6 - MOUSE_DPI
 7 MOUSE_BACKWARD     | MOUSE | 0xE4 - MOUSE_BACKWARD
 8 MOUSE_FORWARD      | MOUSE | 0xE5 - MOUSE_FORWARD
 9 MOUSE_BACKWARD_ALT | MOUSE | 0xF2 - MOUSE_MIDDLE

This is after adding the button here as well: https://github.com/BierLiebHaber/rogdrv/blob/master/rog/device/rog.py

kyokenn, 2022-10-13:
Looks fine to me. Does the button rebinding (to any other action) works?

BierLiebHaber, 2022-10-13:
yeah I have it set to 0xF2 (Middle mouse) and it works perfectly :D

kyokenn, 2022-10-13:
Ok. You can create pull request then.

BierLiebHaber, 2022-11-21:
sorry I was busy for a while but I just found the time to create a PR #49 
if there is a better way to add the button and action please let me know.

kyokenn, 2022-11-21:
It's fine. The PR is OK for the current architecture of the project.

I spend some time thinking about the architecture flaws that prevents to use the better solution. I should not have used the button indices at the first place for the button mapping, because those indices depends on the mouse.

My idea is to use the actual button codes instead of shared indices. I'm already working on some changes for the ratbag:
https://github.com/kyokenn/libratbag/commit/916ef1462b0da298aa5174232a6bb9e29c2ca3d8#diff-818f06bb968064ea18e7728af0beee451ce19be11eb3cc8fc8d2417ac6d2668fL12
https://github.com/kyokenn/libratbag/commit/916ef1462b0da298aa5174232a6bb9e29c2ca3d8#diff-d489dff50fdd3865cb39ff0393a31dd02d3bc9f50d4c206d3ddd11c7e152e114L41

closed 2022-11-26 by BierLiebHaber