kyokenn/rogdrv #50: Asus Spatha X support

Red-M, 2022-11-23:
I've had a good read of #1 but I've noticed the Spatha X has 4 hidraw devices on my Debian stable machine:
```
input: ASUSTeK ROG SPATHA X Consumer Control as /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb6/6-2/6-2:1.2/0003:0B05:1979.000A/input/input38
input: ASUSTeK ROG SPATHA X System Control as /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb6/6-2/6-2:1.2/0003:0B05:1979.000A/input/input39
input: ASUSTeK ROG SPATHA X as /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb6/6-2/6-2:1.2/0003:0B05:1979.000A/input/input40
input: ASUSTeK ROG SPATHA X Keyboard as /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb6/6-2/6-2:1.2/0003:0B05:1979.000A/input/input41
```

I've able to get ratbagd 0.17 to recognise the mouse with a custom device file but it turns off all the LEDs and sets the DPI to the highest value when the mouse is plugged in (sometimes the LEDs for a very brief moment stay as what is set in the mouse then they turn off). The device file is as follows:
```
[Device]
Name=ASUS ROG Spatha X
DeviceMatch=usb:0b05:1979
Driver=asus
LedTypes=logo;side;wheel

[Driver/asus]
Profiles=5
Buttons=12
Leds=3
Dpis=4
Wireless=1
DpiRange=100:19000@50
Quirks=DOUBLE_DPI
```

I'm happy to assist with testing new builds of this project or of ratbagd/ratbagctl to get this device supported. I can take source code to compile and test and provide the results. Taking USB packet captures might be a little difficult but I'm sure I can hook up a VM in my linux host to get the armoury crate captures from.

kyokenn, 2022-11-23:
Can you show output of ```ratbag-command -vvv /dev/hidrawX info``` ?
You can find the hidraw index with ```ratbag-command list```

Red-M, 2022-11-23:
ah ok, also piper seems to be able to read some of the state of the mouse now?
[image]

ah, Debian has it named as `lur-command` but it can't seem to open any of the hidraw devices.

Red-M, 2022-11-23:
Its horribly broken for writing quite a few things to the mouse.

Red-M, 2022-11-23:
```
# ./ratbag-command list
ratbag error: Can't talk to the mouse: 'Broken pipe' (-32)
ratbag error: Can't talk to the mouse: 'Broken pipe' (-32)
/dev/hidraw1:        ASUSTeK ROG SPATHA X```

```
# ./ratbag-command -vvv /dev/hidraw1 info
ratbag debug: New device: ASUSTeK ROG SPATHA X
ratbag debug: Using data directory '/root/git/libratbag/data/devices'
ratbag debug: device assigned driver asus
ratbag debug: hidraw info: bus 0x03 vendor 0xb05 product 0x1979
ratbag debug: ASUSTeK ROG SPATHA X is device '/dev/hidraw1'.
ratbag debug: Parsing HID report descriptor
ratbag debug: Parsing HID report descriptor
ratbag raw: output report: 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 00 00 00 22 00 01 00 05 07 00 03 14 29 91 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Initial profile is 0
ratbag debug: Primary version 11.91.29
ratbag debug: Secondary version 01.00.22
ratbag debug: Loading buttons data
ratbag raw: output report: 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 05 00 00 f0 01 f1 01 f2 01 1f 00 1e 00 e3 01 e8 01 e9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Unknown action code e3
ratbag debug: No mapping for button 8
ratbag debug: No mapping for button 9
ratbag debug: No mapping for button 10
ratbag debug: No mapping for button 11
ratbag debug: Loading resolutions data
ratbag raw: output report: 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 04 00 00 14 00 14 00 14 00 14 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading LEDs data
ratbag raw: output report: 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 03 00 00 00 04 a5 1d 2d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Switching to profile 1
ratbag raw: output report: 50 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  50 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading buttons data
ratbag raw: output report: 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 05 00 00 f0 01 f1 01 f2 01 1f 00 1e 00 e3 01 e8 01 e9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Unknown action code e3
ratbag debug: No mapping for button 8
ratbag debug: No mapping for button 9
ratbag debug: No mapping for button 10
ratbag debug: No mapping for button 11
ratbag debug: Loading resolutions data
ratbag raw: output report: 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 04 00 00 14 00 14 00 14 00 14 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading LEDs data
ratbag raw: output report: 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 03 00 00 00 64 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Switching to profile 2
ratbag raw: output report: 50 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  50 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading buttons data
[... cut ...]
ratbag debug: Switching back to initial profile 0
ratbag raw: output report: 50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: driver match found: ASUS
```

Red-M, 2022-11-23:
I do have keyboard keys set to the extra buttons that say `none`, I don't have any of them that are not mapped to something.

Red-M, 2022-11-23:
The 1050dpi profiles on the mouse are my own and the other 2 are factory.
My profiles have the entire mouse set to red for the LEDs.

kyokenn, 2022-11-23:
What action did you bind to your DPI button on your first profile? It's parsed as "e3".

Looks like the driver was able to read only 8 buttons. Maybe another extra command is needed to read other buttons. Windows VM with ROG Armoury and USB debugger could help to figure this out.

>I've able to get ratbagd 0.17 to recognise the mouse with a custom device file but it turns off all the LEDs and sets the DPI to the highest value when the mouse is plugged in (sometimes the LEDs for a very brief moment stay as what is set in the mouse then they turn off). 

This bug needs more testing. I'll suggest you to stop and disable ratbagd and play around with ratbag-command (with "-vvv" option for capturing the packets), try to change the color of leds, DPI.

Red-M, 2022-11-23:
I think the "not working and its massively buggy" issue is an old version of ratbagd, it seems to work better/normally now as I think the running binary was old (I installed from Debian testing source packages over stable packages).

That button is double click:
[image]

Red-M, 2022-11-23:
For reference, here is the side buttons:
[image]

Red-M, 2022-11-23:
This is profile 5:
[image]
[image]

kyokenn, 2022-11-23:
>That button is double click

Does it work on Linux? I tried to bind it too, but it doesn't work for me.

>I do have a windows PC which I've set the mouse up with but I'd rather move to Linux everywhere so I'd like to support this further.

It's not a problem. You can debug on bare-metal Windows installation or in the VM. Just run the busdog (https://github.com/djpnewton/busdog) alongside with Armoury, try to change some mouse settings and post the busdog logs here so I can implement the missing features.

Red-M, 2022-11-23:
[View_and_set_double_click.txt](https://github.com/kyokenn/rogdrv/files/10076288/View_and_set_double_click.txt)
[Sync_all_5_profiles.txt](https://github.com/kyokenn/rogdrv/files/10076289/Sync_all_5_profiles.txt)

Red-M, 2022-11-23:
set power indicator on dock
change all LEDs to blue then green then red
set lift off distance to high then back to low
Change idle time to 5 minutes then back to 3 minutes, twice
Enable and disable angle snapping
Set mouse wheel LED to blue static at 50% brightness
Set logo LED to green static at 50% brightness
Set side LED to orange reactive at 50% brightness
[bunch _of_stuff.txt](https://github.com/kyokenn/rogdrv/files/10076327/bunch._of_stuff.txt)

Red-M, 2022-11-23:
[set_keyboard_macro.txt](https://github.com/kyokenn/rogdrv/files/10076396/set_keyboard_macro.txt)

kyokenn, 2022-11-23:
Thanks. I also need the packets when the Armoury reads all profiles on startup. So you will need to restart the Armoury.

Red-M, 2022-11-23:
Done
[armoury_start_up.txt](https://github.com/kyokenn/rogdrv/files/10076517/armoury_start_up.txt)

kyokenn, 2022-11-23:
>Done
>[armoury_start_up.txt](https://github.com/kyokenn/rogdrv/files/10076517/armoury_start_up.txt)

Wierd. I still don't see the packets with button bindings. Maybe they will appear when you switch to the buttons tab or switch the profile. 

I'm looking for the request+response like this one:
```
ratbag raw: output report: 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 05 00 00 f0 01 f1 01 f2 01 e4 01 e5 01 e6 01 e8 01 e9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
The packets should start with "12 05". I hope to see some extra packets with extra buttons next to them.

Red-M, 2022-11-23:
I tried starting armoury again but I don't see those, I've switched to another profile which does make armoury wait and it seems to list out the key mappings again.

[change_to_another_profile.txt](https://github.com/kyokenn/rogdrv/files/10076672/change_to_another_profile.txt)

Red-M, 2022-11-23:
I've also killed a few asus services and started armoury up again, still the same, no 12 05 packets.

Red-M, 2022-11-23:
```
Id	Type	Time	Length	Hex	Ascii	
56	In  (USB URB Function: 9)	0.005804	64	12 00 00 00 22 00 01 00 05 07 02 03 14 29 91 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	...."........)..................................................	
```
I'm not 100% but this seems to be it reading the buttons out of the mouse.

kyokenn, 2022-11-23:
>I'm not 100% but this seems to be it reading the buttons out of the mouse.

No. This is firmware version, it's parsed as:
```
ratbag debug: Primary version 11.91.29
ratbag debug: Secondary version 01.00.22
```

The only packets that I see is "12 07" which just reads the battery settings, like timeout and alert levels. 
I also don't see any packets for reading performance settings "12 04" and led settings "12 03".

Red-M, 2022-11-23:
ah ok, I'm not sure how to trigger those as the UI starts off on buttons. I can try a bit more tomorrow.

kyokenn, 2022-11-24:
I have tried other debuggers. It looks like the Armoury has changed and become worse. It doesn't reads the mouse settings. It does writes the settings and hopes that they are not altered on other PC. So the current ratbag implementation is even better than Armoury.

I've still got the button codes and can implement button rebinding. 

>For giggles I checked 12 06 seems to produce

This doesn't looks like the action codes, 1c is KEY_Y.
Can you also dump 12 04? It have performance settings like angle snapping, maybe extra buttons have placed here too.

Red-M, 2022-11-24:
```
ratbag debug: Loading buttons data                                                                                                                                                                                                                                            
ratbag raw: output report: 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                                                    
ratbag raw: input report:  12 04 00 00 14 00 14 00 14 00 14 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Red-M, 2022-11-24:
I'm trying a few other HID codes out but without some doco or reaching into ratbag-python, I can't really spray this device enough to know what other codes are available.

kyokenn, 2022-11-24:
I have implemented extra button binding in the separate tree of my repo - https://github.com/kyokenn/libratbag/tree/spathax
You should able to rebind extra 6 side buttons.

>I'm trying a few other HID codes out but without some doco or reaching into ratbag-python, I can't really spray this device enough to know what other codes are available.

Do you prefer ratbag-python? I can push my WIP ASUS driver for it.

Red-M, 2022-11-24:
I don't prefer it but I was going to use it to debug the config settings.
The Spatha X doesn't seem to need any quirks set for the device file as well.

Red-M, 2022-11-24:
I'll offer a PR to merge in upstream's master and also fix the device file.

It seems to me that libratbag wants to move to ratbag-python as the project is concerned about maintainability of a C codebase so it is probably a good idea to port over the driver to that. I was thinking about doing that to dip my toes into the codebase.

kyokenn, 2022-11-24:
>I don't prefer it but I was going to use it to debug the config settings.

I have pushed ratbag-python fork with ASUS driver - https://github.com/kyokenn/ratbag-python/tree/asusv1

kyokenn, 2022-11-24:
>It seems to me that libratbag wants to move to ratbag-python as the project is concerned about maintainability of a C codebase so it is probably a good idea to port over the driver to that.

It could take a long time before ratbag-python replaces libratbag, because it will need all those drivers to be ported.

Red-M, 2022-11-24:
Yeah, I was going to do that because I have the Python knowledge and a bit of C from messing with Cython as a bit of fun.

My PR is here #1

Red-M, 2022-11-24:
```
./builddir/ratbag-command -vvv /dev/hidraw1 info
ratbag debug: New device: ASUSTeK ROG SPATHA X
ratbag debug: Using data directory '/root/git/libratbag/data/devices'
ratbag debug: device assigned driver asus
ratbag debug: hidraw info: bus 0x03 vendor 0xb05 product 0x1977
ratbag debug: ASUSTeK ROG SPATHA X is device '/dev/hidraw1'.
ratbag debug: Parsing HID report descriptor
ratbag debug: Parsing HID report descriptor
ratbag debug: Button 0 is mapped to 0xf0
ratbag debug: Button 1 is mapped to 0xf1
ratbag debug: Button 2 is mapped to 0xf2
ratbag debug: Button 3 is mapped to 0xe8
ratbag debug: Button 4 is mapped to 0xe9
ratbag debug: Button 5 is mapped to 0xe6
ratbag debug: Button 6 is mapped to 0xe4
ratbag debug: Button 7 is mapped to 0xe5
ratbag debug: Button 8 is mapped to 0xea
ratbag debug: Button 9 is mapped to 0xeb
ratbag debug: Button 10 is mapped to 0xec
ratbag debug: Button 11 is mapped to 0xed
ratbag debug: Button 12 is mapped to 0xee
ratbag debug: Button 13 is mapped to 0xef
ratbag raw: output report: 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 00 00 00 29 91 11 00 05 07 00 03 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Initial profile is 0
ratbag debug: Primary version 00.00.00
ratbag debug: Secondary version 11.91.29
ratbag debug: Loading buttons data
ratbag raw: output report: 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 05 00 00 f0 01 f1 01 f2 01 1f 00 1e 00 e3 01 e8 01 e9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Unknown action code e3
ratbag debug: Loading resolutions data
ratbag raw: output report: 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 04 00 00 14 00 14 00 14 00 14 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading LEDs data
ratbag raw: output report: 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 03 00 00 00 64 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Switching to profile 1
ratbag raw: output report: 50 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  50 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading buttons data
ratbag raw: output report: 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 05 00 00 f0 01 f1 01 f2 01 1f 00 1e 00 e3 01 e8 01 e9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Unknown action code e3
ratbag debug: Loading resolutions data
ratbag raw: output report: 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 04 00 00 14 00 14 00 14 00 14 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading LEDs data
ratbag raw: output report: 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  12 03 00 00 00 64 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Switching to profile 2
ratbag raw: output report: 50 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  50 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: Loading buttons data
[... cut ...]
ratbag raw: output report: 50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag raw: input report:  50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ratbag debug: driver match found: ASUS
```

1977 is the wired mode USB PID.
Looks like it read all of the buttons now.

kyokenn, 2022-11-24:
> Looks like it read all of the buttons now.

It doesn't reads them, they always will appear as UNKNOWN. But it should possible to rebind them. I think Piper will display the new action if you bind the button until you restart ratbagd.

Red-M, 2022-11-24:
Seems to work.
Kinda annoying that ASUS removed reading the settings off the mouse....

closed 2023-03-02 by kyokenn

ruvaldak, 2023-07-20:
is there any way to configure my Spatha X properly in piper or some other gui application? Attempting to use piper makes it impossibe to view the side button configuration, there's no mouse image, and trying to set a button to anything other than a mouse button causes it to fail to commit.

kyokenn, 2023-07-23:
>Attempting to use piper makes it impossibe to view the side button configuration, there's no mouse image (left with a 404 image)

Someone needs to draw an image.

>trying to set a button to anything other than a mouse button causes it to fail to commit

Which ratbag implementation are you using? libratbag or ratbag-python? I also would need a logs to figure out the problem.