kyokenn/rogdrv #51: Gladius III Wireless support

Phjorttos, 2022-12-05:
Linux n00b here asking support for ROG Gladius III Wireless.

Bus 005 Device 002: ID 0b05:197f ASUSTek Computer, Inc. ROG GLADIUS III WIRELESS

kyokenn, 2022-12-06:
I have added mouse to the driver. You can test it.

Phjorttos, 2022-12-13:
It's seems behave same as Gladius 3 wired. Read through issue #45. Should I try to get logs using armoury on windows installation and what exactly is needed?

kyokenn, 2022-12-13:
>Should I try to get logs using armoury on windows installation and what exactly is needed?

Yes. It will help a lot.

You can check the issue https://github.com/kyokenn/rogdrv/issues/50 where we discussed USB debugging process.

You will need to the USB logs for the all features that are not working in this driver.

kyokenn, 2023-03-02:
I have switched back-end to ratbag-python, so there is a chance that I could work now.