kyokenn/rogdrv #52: Support for [ASUSTeK ROG GLADIUS II CORE]

ycMia, 2022-12-22:
Hello, I've installed this software successfully but I got some problem to find my device [ASUSTeK ROG GLADIUS II CORE]
I supposed that it is as the same as Gladius II, however the result turns not so positive.
Btw, I run Rocky 8.7 as my OS and the environment is local, not a remote one.

```
$ rogdrv-config dpi --debug
[2022-12-23 01:50:28,504] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-12-23 01:50:28,506] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,509] [DEBUG] 0 devices found
[2022-12-23 01:50:28,509] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-12-23 01:50:28,510] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,510] [DEBUG] 0 devices found
[2022-12-23 01:50:28,510] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-12-23 01:50:28,510] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,510] [DEBUG] 0 devices found
[2022-12-23 01:50:28,510] [DEBUG] searching for device Gladius3 (VendorID: 0x0B05, ProductID: 0x197B)
[2022-12-23 01:50:28,511] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,511] [DEBUG] 0 devices found
[2022-12-23 01:50:28,511] [DEBUG] searching for device Gladius3Wireless (VendorID: 0x0B05, ProductID: 0x197F)
[2022-12-23 01:50:28,511] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,511] [DEBUG] 0 devices found
[2022-12-23 01:50:28,511] [DEBUG] searching for device Pugio (VendorID: 0x0B05, ProductID: 0x1846)
[2022-12-23 01:50:28,512] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,512] [DEBUG] 0 devices found
[2022-12-23 01:50:28,512] [DEBUG] searching for device PugioGladiusII (VendorID: 0x0B05, ProductID: 0x1851)
[2022-12-23 01:50:28,512] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,512] [DEBUG] 0 devices found
[2022-12-23 01:50:28,512] [DEBUG] searching for device StrixCarry (VendorID: 0x0B05, ProductID: 0x18B4)
[2022-12-23 01:50:28,513] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,513] [DEBUG] 0 devices found
[2022-12-23 01:50:28,513] [DEBUG] searching for device StrixImpact (VendorID: 0x0B05, ProductID: 0x1847)
[2022-12-23 01:50:28,513] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,513] [DEBUG] 0 devices found
[2022-12-23 01:50:28,513] [DEBUG] searching for device StrixImpactIIWireless (VendorID: 0x0B05, ProductID: 0x1949)
[2022-12-23 01:50:28,514] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,514] [DEBUG] 0 devices found
[2022-12-23 01:50:28,514] [DEBUG] searching for device StrixImpactIIWirelessWired (VendorID: 0x0B05, ProductID: 0x1947)
[2022-12-23 01:50:28,514] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,514] [DEBUG] 0 devices found
[2022-12-23 01:50:28,514] [DEBUG] searching for device StrixImpactIIElectroPunk (VendorID: 0x0B05, ProductID: 0x1956)
[2022-12-23 01:50:28,515] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,515] [DEBUG] 0 devices found
[2022-12-23 01:50:28,515] [DEBUG] searching for device Buzzard (VendorID: 0x0B05, ProductID: 0x1816)
[2022-12-23 01:50:28,515] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,515] [DEBUG] 0 devices found
[2022-12-23 01:50:28,515] [DEBUG] searching for device TUFM3 (VendorID: 0x0B05, ProductID: 0x1910)
[2022-12-23 01:50:28,516] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,516] [DEBUG] 0 devices found
[2022-12-23 01:50:28,516] [DEBUG] searching for device KerisWireless (VendorID: 0x0B05, ProductID: 0x1960)
[2022-12-23 01:50:28,516] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,516] [DEBUG] 0 devices found
[2022-12-23 01:50:28,517] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x195E)
[2022-12-23 01:50:28,517] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,517] [DEBUG] 0 devices found
[2022-12-23 01:50:28,517] [DEBUG] searching for device ChakramWireless (VendorID: 0x0B05, ProductID: 0x18E5)
[2022-12-23 01:50:28,518] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,518] [DEBUG] 0 devices found
[2022-12-23 01:50:28,518] [DEBUG] searching for device ChakramWirelessWired (VendorID: 0x0B05, ProductID: 0x18E3)
[2022-12-23 01:50:28,518] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,518] [DEBUG] 0 devices found
[2022-12-23 01:50:28,518] [DEBUG] searching for device ChakramX (VendorID: 0x0B05, ProductID: 0x1A1A)
[2022-12-23 01:50:28,519] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,519] [DEBUG] 0 devices found
[2022-12-23 01:50:28,519] [DEBUG] searching for device Pugio2 (VendorID: 0x0B05, ProductID: 0x1906)
[2022-12-23 01:50:28,519] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,519] [DEBUG] 0 devices found
[2022-12-23 01:50:28,519] [DEBUG] searching for device Pugio2Wired (VendorID: 0x0B05, ProductID: 0x1908)
[2022-12-23 01:50:28,520] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 01:50:28,520] [DEBUG] 0 devices found
Device not found
```
(running $ rogdrv)
```
$rogdrv
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib64/python3.6/threading.py", line 937, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/site-packages/rog/__main__.py", line 65, in run
    e = self._device.next_event()
AttributeError: 'NoneType' object has no attribute 'next_event'

Traceback (most recent call last):
  File "/usr/local/bin/rogdrv", line 8, in <module>
    sys.exit(rogdrv())
  File "/usr/local/lib/python3.6/site-packages/rog/__main__.py", line 393, in rogdrv
    gtk3_main(app.device)
  File "/usr/local/lib/python3.6/site-packages/rog/ui/__init__.py", line 49, in gtk3_main
    if not device.profiles:
AttributeError: 'NoneType' object has no attribute 'profiles'
```
To ensure that I've already pluged in that device
```
$ sudo dmesg | grep USB | grep ASUS
~~
[  318.521775] hid-generic 0003:0B05:18DD.0003: hiddev97,hidraw2: USB HID v1.11 Device [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1a.0-1.2/input0
[  318.522831] hid-generic 0003:0B05:18DD.0004: input,hidraw3: USB HID v1.11 Mouse [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1a.0-1.2/input1
[  318.575459] hid-generic 0003:0B05:18DD.0005: input,hiddev98,hidraw4: USB HID v1.11 Keyboard [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1a.0-1.2/input2
[ 1955.865389] hid-generic 0003:0B05:18DD.0006: hiddev97,hidraw2: USB HID v1.11 Device [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1d.0-1.7/input0
[ 1955.866416] hid-generic 0003:0B05:18DD.0007: input,hidraw3: USB HID v1.11 Mouse [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1d.0-1.7/input1
[ 1955.919786] hid-generic 0003:0B05:18DD.0008: input,hiddev98,hidraw4: USB HID v1.11 Keyboard [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1d.0-1.7/input2
[ 8929.188366] hid-generic 0003:0B05:18DD.0009: hiddev97,hidraw2: USB HID v1.11 Device [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1d.0-1.7/input0
[ 8929.189320] hid-generic 0003:0B05:18DD.000A: input,hidraw3: USB HID v1.11 Mouse [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1d.0-1.7/input1
[ 8929.242584] hid-generic 0003:0B05:18DD.000B: input,hiddev98,hidraw4: USB HID v1.11 Keyboard [ASUSTeK ROG GLADIUS II CORE] on usb-0000:00:1d.0-1.7/input2
[11082.997121] hid-generic 0003:0B05:18DD.000C: hiddev97,hidraw2: USB HID v1.11 Device [ASUSTeK ROG GLADIUS II CORE] on usb-0000:0e:00.0-1.2/input0
[11083.001456] hid-generic 0003:0B05:18DD.000D: input,hidraw3: USB HID v1.11 Mouse [ASUSTeK ROG GLADIUS II CORE] on usb-0000:0e:00.0-1.2/input1
[11083.057917] hid-generic 0003:0B05:18DD.000E: input,hiddev98,hidraw4: USB HID v1.11 Keyboard [ASUSTeK ROG GLADIUS II CORE] on usb-0000:0e:00.0-1.2/input2
[15499.043872] hid-generic 0003:0B05:18DD.000F: hiddev97,hidraw2: USB HID v1.11 Device [ASUSTeK ROG GLADIUS II CORE] on usb-0000:0e:00.0-1.2/input0
[15499.048212] hid-generic 0003:0B05:18DD.0010: input,hidraw3: USB HID v1.11 Mouse [ASUSTeK ROG GLADIUS II CORE] on usb-0000:0e:00.0-1.2/input1
[15499.104449] hid-generic 0003:0B05:18DD.0011: input,hiddev98,hidraw4: USB HID v1.11 Keyboard [ASUSTeK ROG GLADIUS II CORE] on usb-0000:0e:00.0-1.2/input2
~~
```
If there's anyone could help, my appreciate :)

ycMia, 2022-12-22:
P.s. this device's key bind and RGB led can be edit by 'ASUS armoury' when on a windows 10 OS. I've tried it before.

kyokenn, 2022-12-22:
I have added device to the driver. You can test again.

ycMia, 2022-12-22:
Thank you kyokenn for the immediate reply! 
The result turns semi-positive that I can detect this device via software now. The data transfer are good as well.
However it turns blank (no string shown on the terminal) after it started. And I can't get control on my cursor too.
I can't interrupt the program via keyboard (^C) so I just detached the mouse from usb ( you can see it in the end of this log)

well, in the debugging mode, the receives the data well ( I can see the difference between move the mouse straightly and move it like a circle)

(that's a bored one, please skip if you want)
```
rogdrv --console --debug > ~/log.txt
[2022-12-23 02:33:57,733] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-12-23 02:33:57,734] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 02:33:57,739] [DEBUG] 0 devices found
[2022-12-23 02:33:57,739] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-12-23 02:33:57,739] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 02:33:57,740] [DEBUG] 0 devices found
[2022-12-23 02:33:57,740] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-12-23 02:33:57,740] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 02:33:57,740] [DEBUG] 0 devices found
[2022-12-23 02:33:57,740] [DEBUG] searching for device Gladius2Core (VendorID: 0x0B05, ProductID: 0x18DD)
[2022-12-23 02:33:57,741] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-23 02:33:57,745] [DEBUG] found 3 subdevices:
[2022-12-23 02:33:57,745] [DEBUG] 0002:000f:00: ASUSTeK ROG GLADIUS II CORE interface 0
[2022-12-23 02:33:57,745] [DEBUG] 0002:000f:01: ASUSTeK ROG GLADIUS II CORE interface 1 [using as keyboard]
[2022-12-23 02:33:57,745] [DEBUG] 0002:000f:02: ASUSTeK ROG GLADIUS II CORE interface 2 [using as control]
[2022-12-23 02:33:57,745] [DEBUG] opening control subdevice
[2022-12-23 02:33:57,746] [DEBUG] opening device "0002:000f:02"
[2022-12-23 02:33:57,853] [DEBUG] opening keyboard subdevice
[2022-12-23 02:33:57,853] [DEBUG] opening device "0002:000f:01"
[2022-12-23 02:34:03,608] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,613] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,615] [DEBUG] < 00 00 01 00 00 00
[2022-12-23 02:34:03,619] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,621] [DEBUG] < FF FF 01 00 00 00
[2022-12-23 02:34:03,626] [DEBUG] < FF FF 01 00 00 00
[2022-12-23 02:34:03,629] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,632] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,633] [DEBUG] < 00 00 01 00 00 00
[2022-12-23 02:34:03,636] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,639] [DEBUG] < 00 00 01 00 00 00
[2022-12-23 02:34:03,640] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,648] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,649] [DEBUG] < 00 00 01 00 00 00
[2022-12-23 02:34:03,654] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,663] [DEBUG] < 00 00 01 00 00 00
[2022-12-23 02:34:03,666] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,673] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,735] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,738] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,744] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,747] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,747] [DEBUG] < 00 00 FF FF 00 00
[2022-12-23 02:34:03,748] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,753] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,757] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,760] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,762] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,767] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,770] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,772] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,777] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,780] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,783] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,787] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,792] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,808] [DEBUG] < FF FF 00 00 00 00
[2022-12-23 02:34:03,982] [DEBUG] < 01 00 00 00 00 00
[2022-12-23 02:34:04,676] [DEBUG] < 00 00 FF FF 00 00
[2022-12-23 02:34:04,680] [DEBUG] < 00 00 FF FF 00 00
[2022-12-23 02:34:04,684] [DEBUG] < 00 00 FF FF 00 00
[... 1242 lines cut ...]
    self._log(DEBUG, msg, args, **kwargs)
  File "/usr/lib64/python3.6/logging/__init__.py", line 1444, in _log
    self.handle(record)
  File "/usr/lib64/python3.6/logging/__init__.py", line 1454, in handle
    self.callHandlers(record)
  File "/usr/lib64/python3.6/logging/__init__.py", line 1516, in callHandlers
    hdlr.handle(record)
  File "/usr/lib64/python3.6/logging/__init__.py", line 865, in handle
    self.emit(record)
  File "/usr/lib64/python3.6/logging/__init__.py", line 997, in emit
    stream.write(self.terminator)
KeyboardInterrupt
```
Oh, I've tried `rogdrv-config profile -p 1` like issue #45 [030276](https://github.com/user030276) did
light and dpi turns well. but I can't move the cursor until detached it and re-plugin.

After plugin, these adjustments turns effective. I can use the profile 1 that I've set it once in winodws.

Well, actually, I haven't seen any GUI when running `rogdrv` simply. there's no any information shown on the terminal.
One useful information is that I couldn't move my cursor ... under that , too.

That's a mile stone. Just like magic that I've seen the information stream flow from my mouse to the screen via hex strings :D.
The question is: is that the normal thing? Or I faced a bug.

If you have any Idea please tell to me, thanks again!

kyokenn, 2022-12-23:
Can you please show the output of ```rogdrv-config profile -p 1 --debug``` ?

ycMia, 2022-12-24:
Hello here Kyokenn,
```
$ rogdrv-config profile -p 1 --debug
[2022-12-25 00:29:55,706] [DEBUG] searching for device Gladius2 (VendorID: 0x0B05, ProductID: 0x1845)
[2022-12-25 00:29:55,707] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-25 00:29:55,711] [DEBUG] 0 devices found
[2022-12-25 00:29:55,711] [DEBUG] searching for device Gladius2Origin (VendorID: 0x0B05, ProductID: 0x1877)
[2022-12-25 00:29:55,711] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-25 00:29:55,711] [DEBUG] 0 devices found
[2022-12-25 00:29:55,711] [DEBUG] searching for device Gladius2OriginPink (VendorID: 0x0B05, ProductID: 0x18CD)
[2022-12-25 00:29:55,712] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-25 00:29:55,712] [DEBUG] 0 devices found
[2022-12-25 00:29:55,712] [DEBUG] searching for device Gladius2Core (VendorID: 0x0B05, ProductID: 0x18DD)
[2022-12-25 00:29:55,712] [DEBUG] getting list of devices using "cython-hidapi"
[2022-12-25 00:29:55,718] [DEBUG] found 3 subdevices:
[2022-12-25 00:29:55,718] [DEBUG] 2-1.7:1.0: ASUSTeK ROG GLADIUS II CORE interface 0
[2022-12-25 00:29:55,719] [DEBUG] 2-1.7:1.1: ASUSTeK ROG GLADIUS II CORE interface 1 [using as keyboard]
[2022-12-25 00:29:55,719] [DEBUG] 2-1.7:1.2: ASUSTeK ROG GLADIUS II CORE interface 2 [using as control]
[2022-12-25 00:29:55,719] [DEBUG] opening control subdevice
[2022-12-25 00:29:55,719] [DEBUG] opening device "2-1.7:1.2"
[2022-12-25 00:29:55,814] [DEBUG] opening keyboard subdevice
[2022-12-25 00:29:55,814] [DEBUG] opening device "2-1.7:1.1"
[2022-12-25 00:29:55,848] [DEBUG] setting profile to 1
[2022-12-25 00:29:55,848] [DEBUG] > 50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Traceback (most recent call last):
  File "/usr/local/bin/rogdrv-config", line 8, in <module>
    sys.exit(rogdrv_config())
  File "/usr/local/lib/python3.6/site-packages/rog/__main__.py", line 402, in rogdrv_config
    app.run()
  File "/usr/local/lib/python3.6/site-packages/rog/__main__.py", line 93, in run
    method()
  File "/usr/local/lib/python3.6/site-packages/rog/__main__.py", line 162, in profile
    self._device.set_profile(args.profile)
  File "/usr/local/lib/python3.6/site-packages/rog/device/base.py", line 400, in set_profile
    self.query(bytes(request))
  File "/usr/local/lib/python3.6/site-packages/rog/device/base.py", line 199, in query
    response = self.read()
  File "/usr/local/lib/python3.6/site-packages/rog/device/base.py", line 169, in read
    data = self._ctl.read(64)
  File "/usr/local/lib/python3.6/site-packages/rog/hid.py", line 32, in read
    return self._device.read(count)
  File "hid.pyx", line 219, in hid.device.read
OSError: read error
$ exit
exit

Script done on 2022-12-25 00:30:23+08:00
```
These final error begins when I pulled my mouse out. These profiles are edited successfully after that (as same as I done on windows).

kyokenn, 2022-12-27:
The interface could be wrong. I have changed the interface. Can you try again?

ycMia, 2022-12-30:
All Right, I'll do it ... However , forgive me, I'm not there with my OS during the current month.
Could I close this Issue for a period untill I got there and have you informed? Thanks you for the help anlongside this with me, Kyokenn :)

ps-spark, 2023-03-20:
@kyokenn Thank you very much for your great work!