kyokenn/rogdrv #53: Support for Keris Aimpoint

ChristianCiach, 2023-01-28:
Hi, thanks for this neat project!

I just got my new Keris Aimpoint Wireless, and unfortunately I cannot get it to work with rogdrv.

I was able to figure out the product_ids (wired and wireless). There are also a few difference between the Aimpoint and the original Keris:
* Way higher DPI.
* Five profiles (I think the original had three profiles, but the Aimpoint is supposed to have five, according to Asus).
* No mouse wheel LED, only logo LED.

This is how far I got:

```py
class KerisAimpointWireless(BatteryV2Mixin, DoubleDPIMixin, BitmaskMixin, Device):
    """
    Keris Wireless in wireless mode.
    """
    product_id = 0x1A68
    profiles = 5
    buttons = 8
    leds = 1
    keyboard_interface = 2
    control_interface = 0
    wireless = True
    dpis = 4

class KerisAimpointWirelessWired(KerisAimpointWireless):
    """
    Keris Wireless in wired mode.
    """
    product_id = 0x1A66
    wireless = False
```

The product_id in bluetooth mode is `0x1A6A`, but I couldn't get this to work because all sub-device `interface_number` are shown as `-1` with python-hidapi.

`rogdrv` seems to work fine, but I didn't really test it. I am only interested in `rogdrv-config`. These are the issues:

**DPI settings don't seem to work:**

This is the output:

```
> sudo rogdrv-config dpi
DPI Preset 1 (red): 25600
DPI Preset 2 (purple): 25600
DPI Preset 3 (blue): 25600
DPI Preset 4 (green): 25600
```

Of course, these values are not true. Every preset should have a different value and setting a custom dpi value changes nothing.

**LED settings only turn off the LED:**

LED settings seem to *kinda* work. But when setting the brightness, the values 0 to 2 turn off the LED completely. `3` is barely noticable and `4`  is still pretty dim. 

There are probably more things that don't work correctly, but this is how far I got.

kyokenn, 2023-01-30:
Can you run all the DPI/LED commands with "--debug" option and post the output?

ChristianCiach, 2023-02-01:
Sorry, I will come back to you next weekend at the earliest. 

Good news is that I now also have a "Gladius Aimpoint", which I expect to be very similar to the Keris Aimpoint. Maybe the solution will be some kind of `AimpointMixin` class to support all the new Aimpoint devices.

ChristianCiach, 2023-02-01:
I just did some more tests with my Keris Aimpoint. After removing the `DoubleDPIMixin` from the Device, setting DPI seems to work fine! 

This means these issues are still to be solved:
* Reading DPI values
* Setting the LED brightness

By the way: Is there no command to switch the active DPI preset? I couldn't find one.

Here are some logs. All logs begin with these lines, so I've omitted them from the output of the individual commands:

```
[2023-02-01 17:55:15,018] [DEBUG] searching for device KerisWirelessWired (VendorID: 0x0B05, ProductID: 0x1A66)
[2023-02-01 17:55:15,018] [DEBUG] getting list of devices using "cython-hidapi"
[2023-02-01 17:55:15,021] [DEBUG] found 7 subdevices:
[2023-02-01 17:55:15,021] [DEBUG] /dev/hidraw9: ASUSTeK ROG KERIS WIRELESS AIMPOINT interface 0 [using as control]
[2023-02-01 17:55:15,021] [DEBUG] /dev/hidraw10: ASUSTeK ROG KERIS WIRELESS AIMPOINT interface 1
[2023-02-01 17:55:15,021] [DEBUG] /dev/hidraw10: ASUSTeK ROG KERIS WIRELESS AIMPOINT interface 1
[2023-02-01 17:55:15,021] [DEBUG] /dev/hidraw11: ASUSTeK ROG KERIS WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 17:55:15,021] [DEBUG] /dev/hidraw11: ASUSTeK ROG KERIS WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 17:55:15,021] [DEBUG] /dev/hidraw11: ASUSTeK ROG KERIS WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 17:55:15,021] [DEBUG] /dev/hidraw11: ASUSTeK ROG KERIS WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 17:55:15,021] [DEBUG] opening control subdevice
[2023-02-01 17:55:15,021] [DEBUG] opening device "/dev/hidraw9"
[2023-02-01 17:55:15,073] [DEBUG] opening keyboard subdevice
[2023-02-01 17:55:15,073] [DEBUG] opening device "/dev/hidraw11"
```

<details><summary><code>$ rogdrv-config led --debug -b 4 -m default -c ff0000</code></summary>
<p>

```
[2023-02-01 17:55:15,077] [DEBUG] setting LED all color to rgb(255,0,0) with mode default and brightness 4
[2023-02-01 17:55:15,077] [DEBUG] > 51 28 03 00 00 04 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:55:15,079] [DEBUG] < 51 28 03 00 00 04 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:55:15,079] [DEBUG] saving setting
[2023-02-01 17:55:15,079] [DEBUG] > 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:55:15,173] [DEBUG] < 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:55:15,173] [DEBUG] getting LED colors
[2023-02-01 17:55:15,173] [DEBUG] > 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:55:15,175] [DEBUG] < 12 03 00 00 00 04 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 LED       | MODE    | COLOR  | BRIGHTNESS
-----------|---------|--------|-----------
 1 logo    | default | FF0000 | 4
```

**This mostly works**, but the LED is way too dim. Setting the brightness to any other value turns off the LED completely.

</p>
</details>

<details><summary><code>$ rogdrv-config led --debug -b 4 -m battery</code></summary>
<p>

```
[2023-02-01 17:56:00,011] [DEBUG] setting LED all color to rgb(0,0,0) with mode battery and brightness 4
[2023-02-01 17:56:00,012] [DEBUG] > 51 28 03 00 06 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:56:00,014] [DEBUG] < 51 28 03 00 06 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:56:00,014] [DEBUG] saving setting
[2023-02-01 17:56:00,014] [DEBUG] > 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:56:00,108] [DEBUG] < 50 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:56:00,108] [DEBUG] getting LED colors
[2023-02-01 17:56:00,108] [DEBUG] > 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 17:56:00,110] [DEBUG] < 12 03 00 00 06 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 LED       | MODE    | COLOR  | BRIGHTNESS
-----------|---------|--------|-----------
 1 logo    | battery | 000000 | 4
```

[... cut ...]
[2023-02-01 18:29:58,795] [DEBUG] < 12 04 00 00 FF FF FF FF FF FF FF FF 03 00 02 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DPI Preset 1 (red): 12800
DPI Preset 2 (purple): 12800
DPI Preset 3 (blue): 12800
DPI Preset 4 (green): 12800

```

**Setting DPI works fine!** Bug the reported values are still wrong. 

</p>
</details>

Please click on the collapsed sections to see the logs and a short explanation.

The other commands, like `profile` and `rate` seem to work fine. I didn't test the rest.

ChristianCiach, 2023-02-01:
Here is `hid.enumerate()` output for the **Gladius III Aimpoint Wireless**:

**Wired:**
```
{'path': b'/dev/hidraw10', 'vendor_id': 2821, 'product_id': 6768, 'serial_number': 'DA825369A7BAÄ`', 'release_number': 518, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 65281, 'usage': 1, 'interface_number': 0}, 
{'path': b'/dev/hidraw11', 'vendor_id': 2821, 'product_id': 6768, 'serial_number': 'DA825369A7BA\x90\x9c', 'release_number': 518, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 2, 'interface_number': 1}, 
{'path': b'/dev/hidraw11', 'vendor_id': 2821, 'product_id': 6768, 'serial_number': 'DA825369A7BA\x90\x9c', 'release_number': 518, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 1, 'interface_number': 1}, 
{'path': b'/dev/hidraw12', 'vendor_id': 2821, 'product_id': 6768, 'serial_number': 'DA825369A7BAÄ`', 'release_number': 518, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 12, 'usage': 1, 'interface_number': 2}, 
{'path': b'/dev/hidraw12', 'vendor_id': 2821, 'product_id': 6768, 'serial_number': 'DA825369A7BAÄ`', 'release_number': 518, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 128, 'interface_number': 2}, 
{'path': b'/dev/hidraw12', 'vendor_id': 2821, 'product_id': 6768, 'serial_number': 'DA825369A7BAÄ`', 'release_number': 518, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 65473, 'usage': 1, 'interface_number': 2}, 
{'path': b'/dev/hidraw12', 'vendor_id': 2821, 'product_id': 6768, 'serial_number': 'DA825369A7BAÄ`', 'release_number': 518, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 6, 'interface_number': 2}
```

**Wireless (USB receiver):**
```
{'path': b'/dev/hidraw0', 'vendor_id': 2821, 'product_id': 6770, 'serial_number': 'DA825369A7BA', 'release_number': 772, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 65281, 'usage': 1, 'interface_number': 0},
{'path': b'/dev/hidraw1', 'vendor_id': 2821, 'product_id': 6770, 'serial_number': 'DA825369A7BA', 'release_number': 772, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 2, 'interface_number': 1},
{'path': b'/dev/hidraw1', 'vendor_id': 2821, 'product_id': 6770, 'serial_number': 'DA825369A7BA', 'release_number': 772, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 1, 'interface_number': 1},
{'path': b'/dev/hidraw2', 'vendor_id': 2821, 'product_id': 6770, 'serial_number': 'DA825369A7BA', 'release_number': 772, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 12, 'usage': 1, 'interface_number': 2}, 
{'path': b'/dev/hidraw2', 'vendor_id': 2821, 'product_id': 6770, 'serial_number': 'DA825369A7BA', 'release_number': 772, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 128, 'interface_number': 2}, 
{'path': b'/dev/hidraw2', 'vendor_id': 2821, 'product_id': 6770, 'serial_number': 'DA825369A7BA', 'release_number': 772, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 65473, 'usage': 1, 'interface_number': 2}, 
{'path': b'/dev/hidraw2', 'vendor_id': 2821, 'product_id': 6770, 'serial_number': 'DA825369A7BA', 'release_number': 772, 'manufacturer_string': 'ASUSTeK', 'product_string': 'ROG GIII WIRELESS AIMPOINT', 'usage_page': 1, 'usage': 6, 'interface_number': 2}
```

ChristianCiach, 2023-02-01:
I cannot get the Gladius III AimPoint to work. Using these device definitions:

```py
class Gladius3Wireless(Gladius3):
    """
    ROG Gladius III Wireless.
    """
    product_id = 0x197F
    profiles = 5
    dpis = 4

class Gladius3WirelessAimPoint(Device):
    product_id = 0x1A70
    wireless = True
    profiles = 5
    buttons = 8
    leds = 1
    keyboard_interface = 2
    control_interface = 0
```

`rogdrv` seems to work, but `rogdrv-config dpi --debug` gives me this:
```
[2023-02-01 19:10:20,014] [DEBUG] searching for device Gladius3WirelessAimPoint (VendorID: 0x0B05, ProductID: 0x1A70)
[2023-02-01 19:10:20,014] [DEBUG] getting list of devices using "cython-hidapi"
[2023-02-01 19:10:20,017] [DEBUG] found 7 subdevices:
[2023-02-01 19:10:20,017] [DEBUG] /dev/hidraw0: ASUSTeK ROG GIII WIRELESS AIMPOINT interface 0 [using as control]
[2023-02-01 19:10:20,017] [DEBUG] /dev/hidraw1: ASUSTeK ROG GIII WIRELESS AIMPOINT interface 1
[2023-02-01 19:10:20,017] [DEBUG] /dev/hidraw1: ASUSTeK ROG GIII WIRELESS AIMPOINT interface 1
[2023-02-01 19:10:20,017] [DEBUG] /dev/hidraw2: ASUSTeK ROG GIII WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 19:10:20,017] [DEBUG] /dev/hidraw2: ASUSTeK ROG GIII WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 19:10:20,017] [DEBUG] /dev/hidraw2: ASUSTeK ROG GIII WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 19:10:20,017] [DEBUG] /dev/hidraw2: ASUSTeK ROG GIII WIRELESS AIMPOINT interface 2 [using as keyboard]
[2023-02-01 19:10:20,017] [DEBUG] opening control subdevice
[2023-02-01 19:10:20,017] [DEBUG] opening device "/dev/hidraw0"
[2023-02-01 19:10:20,069] [DEBUG] opening keyboard subdevice
[2023-02-01 19:10:20,069] [DEBUG] opening device "/dev/hidraw2"
[2023-02-01 19:10:20,077] [DEBUG] getting DPI and polling rate
[2023-02-01 19:10:20,077] [DEBUG] number of dpi presets: 2
[2023-02-01 19:10:20,078] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 19:10:20,079] [DEBUG] < 12 04 00 00 FF FF FF FF FF FF FF FF 03 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Traceback (most recent call last):
  File "/usr/bin/rogdrv-config", line 8, in <module>
    sys.exit(rogdrv_config())
  File "/usr/lib/python3.10/site-packages/rog/__main__.py", line 402, in rogdrv_config
    app.run()
  File "/usr/lib/python3.10/site-packages/rog/__main__.py", line 93, in run
    method()
  File "/usr/lib/python3.10/site-packages/rog/__main__.py", line 252, in dpi
    dpis, rate, response, snapping = self._device.get_dpi_rate_response_snapping()
  File "/usr/lib/python3.10/site-packages/rog/device/base.py", line 433, in get_dpi_rate_response_snapping
    rate = defs.POLLING_RATES[response[i]]
KeyError: 255
```

I've tried all kinds of Mixins, but the error remains.

`rogdrv-config led` has the same issues as with the Keris AimPoint, meaning the LED is too dim, with exactly the same log output.

kyokenn, 2023-02-01:
>By the way: Is there no command to switch the active DPI preset? I couldn't find one.

It's not supported by the mouse. ASUS hasn't designed such feature at the first place because first mice had only the 2 DPI presets.

>This mostly works, but the LED is way too dim. Setting the brightness to any other value turns off the LED completely.

Can you comment those 2 lines: https://github.com/kyokenn/rogdrv/blob/master/rog/device/base.py#L346-L347
and try other brightness values up to 255?
Maybe your mouse has other brightness range where 0 is the lowest and 255 is the maximum brightness.

>Wrong reported values. They should be something like 800, 1000, 1200, 1400, I think.
```
[2023-02-01 18:29:58,793] [DEBUG] > 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-02-01 18:29:10,868] [DEBUG] < 12 04 00 00 FF FF FF FF FF FF FF FF 03 00 02 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DPI Preset 1 (red): 12800
DPI Preset 2 (purple): 12800
DPI Preset 3 (blue): 12800
DPI Preset 4 (green): 12800
```

That's pretty bad. Looks like they changed the command for getting DPI for the mice with high DPI and the current command just returns FF for each preset.
It's not possible to do reverse engineering with Armoury and USB debugger app anymore, because the current version of Armoury doesn't read the mouse settings and it's not possible to downgrade it AFAIK.
You can just keep it as is or try to send random commands to the mice.
Maybe I should write a bruteforce script which sends all the possible commands from "00 00" to "FF FF" and logs the output...

ChristianCiach, 2023-02-01:
> Can you comment those 2 lines: https://github.com/kyokenn/rogdrv/blob/master/rog/device/base.py#L346-L347
> and try other brightness values up to 255?

Thats exactly it! Removing these lines allows me to set the maximum brighness using `255` and `128` sets it to half brightness, as expected. Nice!

kyokenn, 2023-02-01:
>I cannot get the Gladius III AimPoint to work. Using these device definitions:

>[2023-02-01 19:10:20,077] [DEBUG] number of dpi presets: 2

I think it have 4 DPI presets.

>class Gladius3WirelessAimPoint(Device):

Maybe you are extending the wrong class or forgot to specify "dpis" property.

ChristianCiach, 2023-02-01:
> I think it have 4 DPI presets.
> 
> > class Gladius3WirelessAimPoint(Device):
> 
> Maybe you are extending the wrong class or forgot to specify "dpis" property.

Thanks, that was it! I didn't extend from `Gladius3` on purpose, because I need to exclude the `DoubleDPIMixin` that is included by all the other Gladius Devices. I just forgot to set the `dpis` attribute.

Using these device definitions makes the Gladius III AimPoint behave just the same as the Keris AimPoint:
```py
class Gladius3WirelessAimPoint(Device):
    product_id = 0x1A70
    wireless = True
    profiles = 5
    buttons = 8
    leds = 1
    dpis = 4
    keyboard_interface = 2
    control_interface = 0

class Gladius3WirelessWired(Gladius3WirelessAimPoint):
    product_id = 0x1A72
    wireless = False
```

I didn't bother with button mappings and I don't really understand why some of the other Gladius devices have been configured with `buttons = 9`. But I don't really care about this functionality.

ChristianCiach, 2023-02-01:
> > By the way: Is there no command to switch the active DPI preset? I couldn't find one.
> 
> It's not supported by the mouse. ASUS hasn't designed such feature at the first place because first mice had only the 2 DPI presets.

> > By the way: Is there no command to switch the active DPI preset? I couldn't find one.
> 
> It's not supported by the mouse. ASUS hasn't designed such feature at the first place because first mice had only the 2 DPI presets.

I think I can switch the DPI preset using the Armoury Crate abomination, but I am not completely sure about that. But, as you said, it may not be possible to reverse engineer that. And I don't have a clue how to do that in the first place :)

kyokenn, 2023-02-02:
>Edit: I can confirm that I can directly select the DPI preset when using Armoury Crate.

I see, so ASUS finally made it. Can you dump USB packets with busdog (https://github.com/djpnewton/busdog) ?

kyokenn, 2023-03-02:
I have idea to make DPI work, but you have to set "leds" option to "3" and test if you can save an read the missing LEDs.

More info of the same issue - https://github.com/libratbag/libratbag/issues/1415#issuecomment-1435638177

kyokenn, 2025-05-07:
Related issue for this mouse https://github.com/libratbag/libratbag/issues/1683