kyokenn/rogdrv #54: ROG Pugio II in bluetooth mode constantly disconnects

bam80, 2023-03-31:
It happens after several seconds of idling mouse.
After moving it again, it takes up to tens of seconds to recover:
```
мар 31 14:14:10 Feedme kernel: usb 2-4: new full-speed USB device number 83 using xhci_hcd
мар 31 14:14:10 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:10 Feedme org.kde.kdeconnect.daemon.desktop[5217]: "No carrier"
мар 31 14:14:10 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:11 Feedme kernel: usb 2-4: new full-speed USB device number 84 using xhci_hcd
мар 31 14:14:11 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:11 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:11 Feedme kernel: usb usb2-port4: attempt power cycle
мар 31 14:14:11 Feedme kernel: usb 2-4: new full-speed USB device number 85 using xhci_hcd
мар 31 14:14:11 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:12 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:12 Feedme kernel: usb 2-4: device not accepting address 85, error -71
мар 31 14:14:12 Feedme kernel: usb 2-4: new full-speed USB device number 86 using xhci_hcd
мар 31 14:14:12 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:12 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:12 Feedme kernel: usb 2-4: device not accepting address 86, error -71
мар 31 14:14:12 Feedme kernel: usb usb2-port4: unable to enumerate USB device
мар 31 14:14:16 Feedme kernel: usb 2-4: new full-speed USB device number 87 using xhci_hcd
мар 31 14:14:16 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:17 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:17 Feedme kernel: usb 2-4: new full-speed USB device number 88 using xhci_hcd
мар 31 14:14:17 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:17 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:17 Feedme kernel: usb usb2-port4: attempt power cycle
мар 31 14:14:18 Feedme kernel: usb 2-4: new full-speed USB device number 89 using xhci_hcd
мар 31 14:14:18 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:18 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:18 Feedme kernel: usb 2-4: device not accepting address 89, error -71
мар 31 14:14:18 Feedme kernel: usb 2-4: new full-speed USB device number 90 using xhci_hcd
мар 31 14:14:18 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:18 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:19 Feedme kernel: usb 2-4: device not accepting address 90, error -71
мар 31 14:14:19 Feedme kernel: usb usb2-port4: unable to enumerate USB device
мар 31 14:14:19 Feedme kernel: usb 2-4: new full-speed USB device number 91 using xhci_hcd
мар 31 14:14:19 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:20 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:20 Feedme kernel: usb 2-4: new full-speed USB device number 92 using xhci_hcd
мар 31 14:14:20 Feedme org.kde.kdeconnect.daemon.desktop[5217]: "No carrier"
мар 31 14:14:30 Feedme org.kde.kdeconnect.daemon.desktop[5217]: "No carrier"
мар 31 14:14:35 Feedme kernel: usb 2-4: device descriptor read/64, error -110
мар 31 14:14:36 Feedme kernel: usb 2-4: device descriptor read/64, error -71
мар 31 14:14:36 Feedme kernel: usb usb2-port4: attempt power cycle
мар 31 14:14:36 Feedme kernel: usb 2-4: new full-speed USB device number 93 using xhci_hcd
мар 31 14:14:36 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:36 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:37 Feedme kernel: usb 2-4: device not accepting address 93, error -71
мар 31 14:14:37 Feedme kernel: usb 2-4: new full-speed USB device number 94 using xhci_hcd
мар 31 14:14:37 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:37 Feedme kernel: usb 2-4: Device not responding to setup address.
мар 31 14:14:37 Feedme kernel: usb 2-4: device not accepting address 94, error -71
мар 31 14:14:37 Feedme kernel: usb usb2-port4: unable to enumerate USB device
мар 31 14:14:40 Feedme org.kde.kdeconnect.daemon.desktop[5217]: "No carrier"
```

I understand it probably has nothing to do with this driver but still hoping someone could lead me to the right direction to debug the problem.
Thanks.

kyokenn, 2023-03-31:
Looks like a problem with bluetooth adapter or usb hub/port. If it's an Wi-Fi + bluetooth m.2 combo card, then it could also be a firmware bug.

bam80, 2023-03-31:
Just found the reason - my phone was charging through USB port. When I disconnected it, things went way smoother!

Still I have somewhat not ideal behavior comparing with e.g Microsoft Arc Mouse:
with ROG, after several minutes of idling (the time seems wary) in disconnects ("Bluetooth" icon on Gnome panel disappears) and after that, it doesn't reconnect immediately. It still way better that when phone was charging, but Arc Mouse able reconnect almost immediately, and here I have delay of several seconds..

I wonder am I only one or others have the same experience?

bam80, 2023-03-31:
Update:
I tried to switch off USB autosuspend:
```
$ cat /sys/module/usbcore/parameters/autosuspend
2
$ sudo sh -c ' echo -1 >/sys/module/usbcore/parameters/autosuspend'
```
and now the Bluetooth icon doesn't seem to disappear by timeout. Need to test more.

> My mouse never disconnects from bluetooth. When it goes to sleep it just turns off the LEDs and instantly wakes up when touched.

And that is the other problem. It means the mouse eats a lot of battery while in a such "sleep".
Arc behaves differently. It really disconnects after stable period, thus goes to a true sleep.
But on touch, it reconnects almost instantly.
And I can say Arc preserves battery a way better indeed (many months), probably because of this reason.

kyokenn, 2023-03-31:
>And that is the other problem. It means the mouse eats a lot of battery while in a such "sleep".

It's a gaming mouse. In terms of battery life it works as expected, as described in tech specs.

bam80, 2023-03-31:
Do you mean it might be a "feature", to eliminate even smallest delay after wakeup?
Well, if so it's anti-feature for me, as I don't run games.
At least they could implement non-gaming mode where it sleeps properly..

bam80, 2023-04-01:
IIRC there was a setting in the setup app for Windows to control the sleep timeout.
Do you know if it works on Linux, and how can I make sure that?

bam80, 2023-04-10:
As initial problem is solved now, closing

closed 2023-04-10 by bam80