kyokenn/rogdrv #59: Asus ROG Strix Impact not working / installation issues

s-ki, 2023-06-24:
Hi, following the installation instructions from this site failed:

git clone --recurse-submodules https://github.com/kyokenn/rogdrv.git

Cloning into 'rogdrv'...
remote: Enumerating objects: 788, done.
remote: Counting objects: 100% (123/123), done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 788 (delta 67), reused 79 (delta 37), pack-reused 665
Receiving objects: 100% (788/788), 246.00 KiB | 2.51 MiB/s, done.
Resolving deltas: 100% (508/508), done.
Submodule 'ratbag-python' (git@github.com:kyokenn/ratbag-python.git) registered for path 'ratbag-python'
Cloning into '/home/foo/tmp/rogdrv/ratbag-python'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:kyokenn/ratbag-python.git' into submodule path '/home/foo/tmp/rogdrv/ratbag-python' failed
Failed to clone 'ratbag-python'. Retry scheduled
Cloning into '/home/foo/tmp/rogdrv/ratbag-python'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:kyokenn/ratbag-python.git' into submodule path '/home/foo/tmp/rogdrv/ratbag-python' failed
Failed to clone 'ratbag-python' a second time, aborting

It seems that despite using the https URL, git requires a valid SSH key linked to a GitHub account when resolving dependencies via --recurse-submodules.

So I installed both packages manually:
git clone https://github.com/kyokenn/rogdrv.git
cd rogdrv
git clone https://github.com/kyokenn/ratbag-python.git
(as root) pip3 install ./ratbag-python
(as root) pip3 install .
(as root) ./install_udev

The GUI tool rogdrv shows an icon in the tray bar. Sometimes clicking the icon doesn't have any effect, sometimes it shows a menu where I can select nothing but "profile 0". No further profiles, no settings like dpi, led color and so on.

Any rogdrv-config command (like rogdrv-config dpi or rogdrv-config dpi -d 1600) just shows:
Unable to find driver 'logitech_g300'
Unable to find driver 'steelseries'
Unable to find driver 'roccat-kone-pure'
Unable to find driver 'logitech_g600'
Unable to find driver 'sinowealth'
Unable to find driver 'roccat-kone-emp'
Unable to find driver 'gskill'
Unable to find driver 'etekcity'
Unable to find driver 'sinowealth_nubwo'

...but doesn't output or change the current mouse settings.
The debug output shows that the mouse is identified correctly (I assume "asus" is the correct driver for all Asus devices, including the Strix Impact?):
rogdrv-config dpi --debug -d 1600
[2023-06-24 15:57:10,589] [DEBUG] Loading drivers
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'logitech_g300'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'steelseries'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'roccat-kone-pure'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'logitech_g600'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'sinowealth'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'roccat-kone-emp'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'gskill'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'etekcity'
[2023-06-24 15:57:10,613] [ERROR] Unable to find driver 'sinowealth_nubwo'
[2023-06-24 15:57:10,621] [DEBUG] Using driver 'asus' for usb:0b05:1847 /dev/hidraw0
[2023-06-24 15:57:10,622] [DEBUG] Using driver 'asus' for usb:0b05:1847 /dev/hidraw1
[2023-06-24 15:57:10,622] [DEBUG] Using driver 'asus' for usb:0b05:1847 /dev/hidraw2
[2023-06-24 15:57:10,981] [DEBUG] Button 0 is mapped to 0xf0
[2023-06-24 15:57:10,981] [DEBUG] Button 1 is mapped to 0xf1
[2023-06-24 15:57:10,981] [DEBUG] Button 2 is mapped to 0xf2
[2023-06-24 15:57:10,981] [DEBUG] Button 3 is mapped to 0xe8
[2023-06-24 15:57:10,981] [DEBUG] Button 4 is mapped to 0xe9
[2023-06-24 15:57:10,981] [DEBUG] Button 5 is mapped to 0xe6
[2023-06-24 15:57:10,981] [DEBUG] Button 6 is mapped to 0xe4
[2023-06-24 15:57:10,981] [DEBUG] Button 7 is mapped to 0xe5
[2023-06-24 15:57:10,981] [DEBUG] fd TX (64): 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,985] [DEBUG] fd RX (64): 12 00 00 00 00 01 03 05 01 04 01 01 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,985] [DEBUG] Initial profile is 1
[2023-06-24 15:57:10,986] [DEBUG] Primary version 00.00.00
[2023-06-24 15:57:10,986] [DEBUG] Secondary version 03.01.00
[2023-06-24 15:57:10,986] [DEBUG] Switching to profile 0
[2023-06-24 15:57:10,986] [DEBUG] fd TX (64): 50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,990] [DEBUG] fd RX (64): ff aa 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,990] [DEBUG] Loading buttons data
[2023-06-24 15:57:10,990] [DEBUG] fd TX (64): 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,994] [DEBUG] fd RX (64): 12 05 00 00 f0 01 f1 01 f2 01 00 00 00 00 e6 01 00 00 e8 01 e9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,994] [DEBUG] Loading resolutions data
[2023-06-24 15:57:10,994] [DEBUG] fd TX (64): 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,998] [DEBUG] fd RX (64): 12 04 00 00 13 00 07 00 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:10,998] [DEBUG] Loading LEDs data
[2023-06-24 15:57:10,998] [DEBUG] fd TX (64): 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:11,002] [DEBUG] fd RX (64): 12 03 00 00 00 01 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:11,002] [DEBUG] Switching back to initial profile 1
[2023-06-24 15:57:11,002] [DEBUG] fd TX (64): 50 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:11,006] [DEBUG] fd RX (64): ff aa 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:11,006] [DEBUG] fd TX (64): 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:11,010] [DEBUG] fd RX (64): 12 00 00 00 00 01 03 05 01 04 01 01 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:11,010] [DEBUG] Initial profile is 1
[2023-06-24 15:57:11,010] [DEBUG] Switching back to initial profile 1
[2023-06-24 15:57:11,010] [DEBUG] fd TX (64): 50 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-24 15:57:11,014] [DEBUG] fd RX (64): ff aa 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

lsusb shows mouse as:
Bus 001 Device 002: ID 0b05:1847 ASUSTek Computer, Inc. ROG STRIX IMPACT

ls -l /dev/hidraw*
crw-rw----+ 1 root plugdev 248, 0 24. Jun 15:08 /dev/hidraw0
crw-rw----+ 1 root plugdev 248, 1 24. Jun 15:08 /dev/hidraw1
crw-rw----+ 1 root plugdev 248, 2 24. Jun 15:08 /dev/hidraw2

What am I missing?
OS is Debian 11 Bullseye.

kyokenn, 2023-06-24:
>It seems that despite using the https URL, git requires a valid SSH key linked to a GitHub account when resolving dependencies via --recurse-submodules.

I have fixed submodule url.

>...but doesn't output or change the current mouse settings.

Ported libratbag fix to ratbag-python for single profile mice which I made recently for libratbag.
Make sure that you have updated ratbag-python submodule and reinstalled it with pip.

s-ki, 2023-06-25:
That was quick, thanks!
Installation runs fine now. I can change dpi settings using the rogdrv-config tool.
In the graphical rogdrv tool, both dpi presets show up correcty, but are grayed out and cannot be modified.
LEDs behave weird in both tools. In the GUI tool, the color setting is grayed out and stuck to #000000. I can change brightness and switch mode between "cycle" (works) and "on"/"breathing" (both don't seem to work, LED stays black).
The rogdrv-config tool doesn't seem to have any effect on the LEDs. It shows the current setting configured with the graphical tool, but doesn't apply any changes itself. The rogdrv tool was quit before trying rogdrv-config:

rogdrv-config led -c 0040ff -m ON -b 255 --debug
[2023-06-25 15:51:15,793] [DEBUG] Loading drivers
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'logitech_g300'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'steelseries'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'roccat-kone-pure'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'logitech_g600'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'sinowealth'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'roccat-kone-emp'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'gskill'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'etekcity'
[2023-06-25 15:51:15,817] [ERROR] Unable to find driver 'sinowealth_nubwo'
[2023-06-25 15:51:15,825] [DEBUG] Using driver 'asus' for usb:0b05:1847 /dev/hidraw0
[2023-06-25 15:51:15,825] [DEBUG] Using driver 'asus' for usb:0b05:1847 /dev/hidraw1
[2023-06-25 15:51:15,825] [DEBUG] Using driver 'asus' for usb:0b05:1847 /dev/hidraw2
[2023-06-25 15:51:16,186] [DEBUG] Button 0 is mapped to 0xf0
[2023-06-25 15:51:16,186] [DEBUG] Button 1 is mapped to 0xf1
[2023-06-25 15:51:16,186] [DEBUG] Button 2 is mapped to 0xf2
[2023-06-25 15:51:16,186] [DEBUG] Button 3 is mapped to 0xe8
[2023-06-25 15:51:16,186] [DEBUG] Button 4 is mapped to 0xe9
[2023-06-25 15:51:16,186] [DEBUG] Button 5 is mapped to 0xe6
[2023-06-25 15:51:16,186] [DEBUG] Button 6 is mapped to 0xe4
[2023-06-25 15:51:16,186] [DEBUG] Button 7 is mapped to 0xe5
[2023-06-25 15:51:16,186] [DEBUG] fd TX (64): 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,189] [DEBUG] fd RX (64): 12 00 00 00 00 01 03 05 01 04 01 01 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,189] [DEBUG] Primary version 00.00.00
[2023-06-25 15:51:16,189] [DEBUG] Secondary version 03.01.00
[2023-06-25 15:51:16,189] [DEBUG] Switching to profile 0
[2023-06-25 15:51:16,189] [DEBUG] fd TX (64): 50 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,193] [DEBUG] fd RX (64): ff aa 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,193] [DEBUG] Loading buttons data
[2023-06-25 15:51:16,193] [DEBUG] fd TX (64): 12 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,197] [DEBUG] fd RX (64): 12 05 00 00 f0 01 f1 01 f2 01 00 00 00 00 e6 01 00 00 e8 01 e9 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,198] [DEBUG] Loading resolutions data
[2023-06-25 15:51:16,198] [DEBUG] fd TX (64): 12 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,201] [DEBUG] fd RX (64): 12 04 00 00 17 00 07 00 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,202] [DEBUG] Loading LEDs data
[2023-06-25 15:51:16,202] [DEBUG] fd TX (64): 12 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[2023-06-25 15:51:16,205] [DEBUG] fd RX (64): 12 03 00 00 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0: CYCLE #000000 brightness=64

Thanks again for your great work.