kyokenn/rogdrv #61: ASUS ROG Chakram Core support (0b05:1958)

pipes80-collab, 2023-11-18:
Could the ROG Chakram core model be supported in the future?
0b05:1958 ASUSTek Computer, Inc. ROG CHAKRAM CORE

Thanks for work

kyokenn, 2023-11-23:
I have added it to the ratbag-python. You can test, if it works for you.

pipes80-collab, 2023-11-26:
it doesn't work, it deconfigures the device and makes it slow and unusable

kyokenn, 2024-01-02:
You may need to run the ratbagd with sudo (```sudo ratbagd```) or change the device files permissions ```chmod 666 /dev/hidraw*``` (just once, for tests, it will be reseted on the next reboot)

pipes80-collab, 2024-02-27:
when i run sudo ratbagd i take this message 
[image]

kyokenn, 2024-03-04:
You need to copy d-bus config:
```sudo cp -fv ./ratbag-python/dbus/org.freedesktop.ratbag1.conf /etc/dbus-1/system.d/```

pipes80-collab, 2024-04-16:
it works but when rogdrv is launched the mouse takes on a configuration that makes the device unusable and you have to enter Windows to configure it, it is certainly not fully compatible with the mouse I use

kyokenn, 2024-04-30:
Which linux distro is that? You need to figure out which folders you have in /etc/dbus-1

> it deconfigures the device and makes it slow and unusable

This is expected. It's not possible to read DPI settings from the mouse, so they will be reseted to lowest on the first run.

pipes80-collab, 2024-05-05:
I use archlinux, the problem is that the keys are also changed because it is not even possible to click or do other actions